f4exb/sdrangel #1063: .deb qml dependencies

srcejon, 2021-11-28:
I just tried installing the release .deb for the first time on a clean Ubuntu 20.04 Desktop VM.

The maps don't work, with the error:

qrc:/map/map/map_5_12.qml:2:1: module "QtQuick.Window" is not installed
qrc:/map/map/map_5_12.qml:4:1: module "QtLocation" is not installed
qrc:/map/map/map_5_12.qml:5:1: module "QtPositioning" is not installed
qrc:/map/map/map_5_12.qml:3:1: module "QtQuick.Controls" is not installed

In the log.

This can be fixed with:

sudo apt-get install qml-module-qtlocation  qml-module-qtlocation qml-module-qtpositioning qml-module-qtquick-window2 qml-module-qtquick-dialogs qml-module-qtquick-controls qml-module-qtquick-controls2 qml-module-qtquick-layouts

But I guess it would be better if these were in the package dependencies so they are automatically installed along with the .deb.

srcejon, 2021-11-29:
Well, all of those modules are in debian/control and sdrangel_6.17.4-1_amd64.buildinfo, so not sure why they weren't installed.

srcejon, 2021-11-29:
They don't appear here though:

dpkg -I ./sdrangel_6.17.4-1_amd64.deb 
 new Debian package, version 2.0.
 size 24952112 bytes: control archive=4964 bytes.
    1948 bytes,    23 lines      control              
   10345 bytes,   140 lines      md5sums              
     219 bytes,    10 lines      shlibs               
      73 bytes,     2 lines      triggers             
 Package: sdrangel
 Version: 6.17.4-1
 Architecture: amd64
 Maintainer: Edouard Griffiths, F4EXB <f4exb06@gmail.com>
 Installed-Size: 64768
 Depends: libairspy0 (>= 1.0.9~), libavcodec58 (>= 7:4.2), libavformat58 (>= 7:4.1), libavutil56 (>= 7:4.0), libc6 (>= 2.29), libfaad2 (>= 2.7), libfftw3-single3 (>= 3.3.5), libgcc-s1 (>= 4.0), libhackrf0, libopencv-core4.2 (>= 4.2.0+dfsg), libopencv-imgcodecs4.2 (>= 4.2.0+dfsg), libopencv-imgproc4.2 (>= 4.2.0+dfsg), libopencv-videoio4.2 (>= 4.2.0+dfsg), libopus0 (>= 1.1), libqt5charts5 (>= 5.9.0~rc), libqt5core5a (>= 5.12.2), libqt5gui5 (>= 5.12.2) | libqt5gui5-gles (>= 5.12.2), libqt5location5 (>= 5.6.0), libqt5multimedia5 (>= 5.6.0~beta), libqt5network5 (>= 5.8.0), libqt5opengl5 (>= 5.0.2), libqt5positioning5 (>= 5.6.0), libqt5qml5 (>= 5.0.2), libqt5quickwidgets5 (>= 5.3.0), libqt5serialport5 (>= 5.6.0), libqt5texttospeech5 (>= 5.8.0~alpha), libqt5websockets5 (>= 5.6.0), libqt5widgets5 (>= 5.2.0), libstdc++6 (>= 9), libswresample3 (>= 7:4.0), libswscale5 (>= 7:4.0), libuhd3.15.0 (>= 3.15.0.0), libusb-1.0-0 (>= 2:1.0.22), libxml2 (>= 2.7.4), libasound2, libgl1-mesa-glx, libqt5multimedia5-plugins, pulseaudio, ffmpeg
 Section: hamradio
 Priority: optional
 Homepage: https://github.com/f4exb/sdrangel
 Description: SDR/Analyzer/Generator front-end for various hardware
  SDR/Analyzer/Generator front-end for Airspy, BladeRF, HackRF,
  RTL-SDR, FunCube, LimeSDR, PlutoSDR, RSP, USRP.
  Also File source and sink for I/Q samples, network I/Q sources with
  remote instance.
  Based on Qt5 framework and OpenGL for the spectrum and scope rendering.
  Builds on Linux, Windows and Mac O/S
  Reception modes supported:
    Analog: AM, ATV, NFM, WFM, SSB, broadcast FM, APT
    Digital: D-Star, Yaesu SF, DMR, dPMR, FreeDV, DAB, DVB-S, LoRa, ADS-B, Packet (AX.25/APRS), AIS
    Analyzer: Generic channel
  Transmission modes supported:
    Analog: AM, ATV, NFM, SSB, WFM
    Digital: DVB-S, Packet (AX.25), AIS, 802.15.4

in source/control those modules are under Build-Depends: do they need to be under Depends: instead?

srcejon, 2021-11-29:
E.g: 

```
-Depends: ${shlibs:Depends}, ${misc:Depends}, libasound2, libgl1-mesa-glx, libqt5multimedia5-plugins, pulseaudio, ffmpeg

+Depends: ${shlibs:Depends},
 ${misc:Depends},
 libasound2,
 libgl1-mesa-glx,
 libqt5multimedia5-plugins,
 pulseaudio,
 ffmpeg,
 qml-module-qtlocation,
 qml-module-qtpositioning,
 qml-module-qtquick-window2,
 qml-module-qtquick-dialogs,
 qml-module-qtquick-controls,
 qml-module-qtquick-controls2,
 qml-module-qtquick-layouts,
 qml-module-qtgraphicaleffects
```

closed 2021-12-01 by f4exb

f4exb, 2021-12-01:
Sounds reasonable. If this does not work please re-open.

f4exb, 2021-12-01:
Artifact is here: https://ci.appveyor.com/project/f4exb/sdrangel/builds/41736131/job/ohhx42rianotkj8o/artifacts

Looks better:
```
dpkg -I ./sdrangel_6.17.4-1_amd64.deb 
 new Debian package, version 2.0.
 size 24961676 bytes: control archive=5028 bytes.
    2171 bytes,    23 lines      control              
   10345 bytes,   140 lines      md5sums              
     219 bytes,    10 lines      shlibs               
      73 bytes,     2 lines      triggers             
 Package: sdrangel
 Version: 6.17.4-1
 Architecture: amd64
 Maintainer: Edouard Griffiths, F4EXB <f4exb06@gmail.com>
 Installed-Size: 64812
 Depends: libairspy0 (>= 1.0.9~), libavcodec58 (>= 7:4.2), libavformat58 (>= 7:4.1), libavutil56 (>= 7:4.0), libc6 (>= 2.29), libfaad2 (>= 2.7), libfftw3-single3 (>= 3.3.5), libgcc-s1 (>= 4.0), libhackrf0, libopencv-core4.2 (>= 4.2.0+dfsg), libopencv-imgcodecs4.2 (>= 4.2.0+dfsg), libopencv-imgproc4.2 (>= 4.2.0+dfsg), libopencv-videoio4.2 (>= 4.2.0+dfsg), libopus0 (>= 1.1), libqt5charts5 (>= 5.9.0~rc), libqt5core5a (>= 5.12.2), libqt5gui5 (>= 5.12.2) | libqt5gui5-gles (>= 5.12.2), libqt5location5 (>= 5.6.0), libqt5multimedia5 (>= 5.6.0~beta), libqt5network5 (>= 5.8.0), libqt5opengl5 (>= 5.0.2), libqt5positioning5 (>= 5.6.0), libqt5qml5 (>= 5.0.2), libqt5quickwidgets5 (>= 5.3.0), libqt5serialport5 (>= 5.6.0), libqt5texttospeech5 (>= 5.8.0~alpha), libqt5websockets5 (>= 5.6.0), libqt5widgets5 (>= 5.2.0), libstdc++6 (>= 9), libswresample3 (>= 7:4.0), libswscale5 (>= 7:4.0), libuhd3.15.0 (>= 3.15.0.0), libusb-1.0-0 (>= 2:1.0.22), libxml2 (>= 2.7.4), libasound2, libgl1-mesa-glx, libqt5multimedia5-plugins, pulseaudio, ffmpeg, qml-module-qtlocation, qml-module-qtpositioning, qml-module-qtquick-window2, qml-module-qtquick-dialogs, qml-module-qtquick-controls, qml-module-qtquick-controls2, qml-module-qtquick-layouts, qml-module-qtgraphicaleffects
 Section: hamradio
 Priority: optional
 Homepage: https://github.com/f4exb/sdrangel
 Description: SDR/Analyzer/Generator front-end for various hardware
  SDR/Analyzer/Generator front-end for Airspy, BladeRF, HackRF,
  RTL-SDR, FunCube, LimeSDR, PlutoSDR, RSP, USRP.
  Also File source and sink for I/Q samples, network I/Q sources with
  remote instance.                                                                                                                                                                                                                          
  Based on Qt5 framework and OpenGL for the spectrum and scope rendering.                                                                                                                                                                   
  Builds on Linux, Windows and Mac O/S                                                                                                                                                                                                      
  Reception modes supported:                                                                                                                                                                                                                
    Analog: AM, ATV, NFM, WFM, SSB, broadcast FM, APT                                                                                                                                                                                       
    Digital: D-Star, Yaesu SF, DMR, dPMR, FreeDV, DAB, DVB-S, LoRa, ADS-B, Packet (AX.25/APRS), AIS                                                                                                                                         
    Analyzer: Generic channel                                                                                                                                                                                                               
  Transmission modes supported:                                                                                                                                                                                                             
    Analog: AM, ATV, NFM, SSB, WFM                                                                                                                                                                                                          
    Digital: DVB-S, Packet (AX.25), AIS, 802.15.4         
```

commit 3d0a5d70 (2021-12-01): Ubuntu build: added qm-module dependencies in the Depend section. Fixes #1063