f4exb/sdrangel #1064: Crash when after source in windows

marcelrv, 2021-11-28:
After selecting source (hackrf one), hard crash in windows 11 (note, it is not instant, but after a second or so).
Using alternative source (testsource) there is no crash.
Using latest version, release windows build from github

```
2021-11-28 13:14:38.640 (I) SDRangel 6.17.4 Qt 5.15.2 64b x86_64 Windows 10 Version 2009 DSP Rx:24b Tx:16b PID 8860
2021-11-28 13:14:38.642 (D) MainWindow::MainWindow: load plugins...
2021-11-28 13:14:38.642 (D) PluginManager::loadPluginsPart folder C:/Program Files/SDRangel/../lib/sdrangel/plugins is empty
2021-11-28 13:14:38.642 (D) PluginManager::loadPluginsPart folder C:/Program Files/SDRangel/lib/plugins is empty
2021-11-28 13:14:38.643 (D) PluginManager::loadPluginsDir: fileName: antennatools.dll
2021-11-28 13:14:38.644 (I) PluginManager::loadPluginsDir: loaded plugin antennatools.dll
2021-11-28 13:14:38.644 (D) PluginManager::loadPluginsDir: fileName: aprs.dll
2021-11-28 13:14:38.646 (I) PluginManager::loadPluginsDir: loaded plugin aprs.dll
2021-11-28 13:14:38.646 (D) PluginManager::loadPluginsDir: fileName: beamsteeringcwmod.dll
2021-11-28 13:14:38.647 (I) PluginManager::loadPluginsDir: loaded plugin beamsteeringcwmod.dll
2021-11-28 13:14:38.647 (D) PluginManager::loadPluginsDir: fileName: chanalyzer.dll
2021-11-28 13:14:38.648 (I) PluginManager::loadPluginsDir: loaded plugin chanalyzer.dll
2021-11-28 13:14:38.648 (D) PluginManager::loadPluginsDir: fileName: demodadsb.dll
2021-11-28 13:14:38.651 (I) PluginManager::loadPluginsDir: loaded plugin demodadsb.dll
2021-11-28 13:14:38.651 (D) PluginManager::loadPluginsDir: fileName: demodais.dll
2021-11-28 13:14:38.652 (I) PluginManager::loadPluginsDir: loaded plugin demodais.dll
2021-11-28 13:14:38.652 (D) PluginManager::loadPluginsDir: fileName: demodam.dll
2021-11-28 13:14:38.653 (I) PluginManager::loadPluginsDir: loaded plugin demodam.dll
2021-11-28 13:14:38.653 (D) PluginManager::loadPluginsDir: fileName: demodapt.dll
2021-11-28 13:14:38.655 (I) PluginManager::loadPluginsDir: loaded plugin demodapt.dll
2021-11-28 13:14:38.655 (D) PluginManager::loadPluginsDir: fileName: demodatv.dll
2021-11-28 13:14:38.656 (I) PluginManager::loadPluginsDir: loaded plugin demodatv.dll
2021-11-28 13:14:38.656 (D) PluginManager::loadPluginsDir: fileName: demodbfm.dll
2021-11-28 13:14:38.657 (I) PluginManager::loadPluginsDir: loaded plugin demodbfm.dll
2021-11-28 13:14:38.657 (D) PluginManager::loadPluginsDir: fileName: demodchirpchat.dll
2021-11-28 13:14:38.658 (I) PluginManager::loadPluginsDir: loaded plugin demodchirpchat.dll
2021-11-28 13:14:38.658 (D) PluginManager::loadPluginsDir: fileName: demoddab.dll
2021-11-28 13:14:38.666 (I) PluginManager::loadPluginsDir: loaded plugin demoddab.dll
2021-11-28 13:14:38.666 (D) PluginManager::loadPluginsDir: fileName: demoddatv.dll
2021-11-28 13:14:38.671 (I) PluginManager::loadPluginsDir: loaded plugin demoddatv.dll
2021-11-28 13:14:38.671 (D) PluginManager::loadPluginsDir: fileName: demoddsd.dll
2021-11-28 13:14:38.672 (I) PluginManager::loadPluginsDir: loaded plugin demoddsd.dll
2021-11-28 13:14:38.672 (D) PluginManager::loadPluginsDir: fileName: demodfreedv.dll
2021-11-28 13:14:38.673 (I) PluginManager::loadPluginsDir: loaded plugin demodfreedv.dll
2021-11-28 13:14:38.673 (D) PluginManager::loadPluginsDir: fileName: demodnfm.dll
2021-11-28 13:14:38.677 (I) PluginManager::loadPluginsDir: loaded plugin demodnfm.dll
2021-11-28 13:14:38.678 (D) PluginManager::loadPluginsDir: fileName: demodpacket.dll
2021-11-28 13:14:38.684 (I) PluginManager::loadPluginsDir: loaded plugin demodpacket.dll
2021-11-28 13:14:38.684 (D) PluginManager::loadPluginsDir: fileName: demodpager.dll
2021-11-28 13:14:38.689 (I) PluginManager::loadPluginsDir: loaded plugin demodpager.dll
2021-11-28 13:14:38.690 (D) PluginManager::loadPluginsDir: fileName: demodssb.dll
2021-11-28 13:14:38.692 (I) PluginManager::loadPluginsDir: loaded plugin demodssb.dll
2021-11-28 13:14:38.692 (D) PluginManager::loadPluginsDir: fileName: demodvor.dll
2021-11-28 13:14:38.694 (I) PluginManager::loadPluginsDir: loaded plugin demodvor.dll
2021-11-28 13:14:38.694 (D) PluginManager::loadPluginsDir: fileName: demodvorsc.dll
2021-11-28 13:14:38.695 (I) PluginManager::loadPluginsDir: loaded plugin demodvorsc.dll
2021-11-28 13:14:38.695 (D) PluginManager::loadPluginsDir: fileName: demodwfm.dll
2021-11-28 13:14:38.697 (I) PluginManager::loadPluginsDir: loaded plugin demodwfm.dll
2021-11-28 13:14:38.697 (D) PluginManager::loadPluginsDir: fileName: featureafc.dll
2021-11-28 13:14:38.698 (I) PluginManager::loadPluginsDir: loaded plugin featureafc.dll
2021-11-28 13:14:38.698 (D) PluginManager::loadPluginsDir: fileName: featureais.dll
2021-11-28 13:14:38.699 (I) PluginManager::loadPluginsDir: loaded plugin featureais.dll
2021-11-28 13:14:38.700 (D) PluginManager::loadPluginsDir: fileName: featuredemodanalyzer.dll
2021-11-28 13:14:38.701 (I) PluginManager::loadPluginsDir: loaded plugin featuredemodanalyzer.dll
2021-11-28 13:14:38.701 (D) PluginManager::loadPluginsDir: fileName: featuregs232controller.dll
2021-11-28 13:14:38.702 (I) PluginManager::loadPluginsDir: loaded plugin featuregs232controller.dll
2021-11-28 13:14:38.702 (D) PluginManager::loadPluginsDir: fileName: featurepertester.dll
2021-11-28 13:14:38.704 (I) PluginManager::loadPluginsDir: loaded plugin featurepertester.dll
2021-11-28 13:14:38.704 (D) PluginManager::loadPluginsDir: fileName: featurerigctlserver.dll
2021-11-28 13:14:38.705 (I) PluginManager::loadPluginsDir: loaded plugin featurerigctlserver.dll
2021-11-28 13:14:38.705 (D) PluginManager::loadPluginsDir: fileName: featuresatellitetracker.dll
2021-11-28 13:14:38.706 (I) PluginManager::loadPluginsDir: loaded plugin featuresatellitetracker.dll
2021-11-28 13:14:38.706 (D) PluginManager::loadPluginsDir: fileName: featuresimpleptt.dll
2021-11-28 13:14:38.707 (I) PluginManager::loadPluginsDir: loaded plugin featuresimpleptt.dll
2021-11-28 13:14:38.707 (D) PluginManager::loadPluginsDir: fileName: featurestartracker.dll
[... 344 lines cut ...]
2021-11-28 13:14:42.460 (D) DeviceAPI::loadSamplingDeviceSettings: Loading Rx preset [default | no name]
2021-11-28 13:14:42.460 (D) Preset::findBestDeviceConfig: serial matched (exact): id: sdrangel.samplesource.fileinput ser: 
2021-11-28 13:14:42.460 (D) DeviceAPI::loadSamplingDeviceSettings: source center frequency: 0 Hz
2021-11-28 13:14:42.460 (D) DeviceAPI::loadSamplingDeviceSettings: deserializing source sdrangel.samplesource.fileinput[0]: 
2021-11-28 13:14:42.460 (D) DeviceUISet::loadRxChannelSettings: Loading preset [default | no name]
2021-11-28 13:14:42.460 (D) DeviceUISet::loadRxChannelSettings: 0 channel(s) in preset
2021-11-28 13:14:42.463 (D) MainWindow::loadFeatureSetPresetSettings: preset [default | no name]
2021-11-28 13:14:42.463 (D) MainWindow::loadFeatureSetPresetSettings: m_apiAdapter: 0x1ed51ef6980
2021-11-28 13:14:42.463 (D) FeatureUISet::loadFeatureSetSettings: Loading preset [default | no name]
2021-11-28 13:14:42.463 (D) FeatureUISet::loadFeatureSetSettings: 0 feature(s) in preset
2021-11-28 13:14:42.465 (D) MainWindow::MainWindow: update preset controls...
2021-11-28 13:14:42.466 (W) MainWindow::MainWindow: could not register resource file C:/Program Files/SDRangel/sdrbase.rcc
2021-11-28 13:14:42.466 (D) StaticFileController: docroot=:/webapi, encoding=UTF-8, maxAge=60000
2021-11-28 13:14:42.466 (D) StaticFileController: cache timeout=60000, size=1000000
2021-11-28 13:14:42.467 (D) HttpListener: Listening on port 8091
2021-11-28 13:14:42.467 (I) WebAPIServer::start: starting web API server at http://:8091
2021-11-28 13:14:42.469 (D) MainWindow::MainWindow: end
2021-11-28 13:14:42.493 (D) GLSpectrum::initializeGL: context:  major:  4  minor:  6  ES:  no
2021-11-28 13:14:42.493 (D) GLShaderTextured::initializeGL: m_useImmutableStorage:  true
2021-11-28 13:14:42.494 (D) GLShaderTextured::initializeGL: m_useImmutableStorage:  true
2021-11-28 13:14:42.494 (D) GLShaderTextured::initializeGL: m_useImmutableStorage:  true
2021-11-28 13:14:42.495 (D) GLShaderTextured::initializeGL: m_useImmutableStorage:  true
2021-11-28 13:14:42.495 (D) GLShaderTextured::initializeGL: m_useImmutableStorage:  true
2021-11-28 13:14:42.495 (D) GLShaderTextured::initializeGL: m_useImmutableStorage:  true
```

srcejon, 2021-11-28:
Is it related to https://github.com/f4exb/sdrangel/issues/718

Might try removing SoapySDR.dll

f4exb, 2021-12-17:
Confirmed SoapySDR.dll fixes HackRF issue. I am now wondering why SoapySDR should be supported at all?

closed 2022-03-24 by github-actions[bot]