f4exb/sdrangel #1066: SDRangel restores LimeSDR[0:0] on startup rather than LimeSDR[0:1]

srcejon, 2021-11-29:
SDRangel typically restores the last DeviceSet that was being used when re-started.

One example of where this doesn't quite happen, is for LimeSDR and USRP that have multiple channels. It seems [0:0] is always restored, even if [0:1] was last used.

f4exb, 2021-11-29:
How comes you get the second Rx ([0:1]) on the first tab R0? This is normally impossible. When you change to LimeSDR in R0 it always get the first Rx ([0,0]) to open the second you need to do it in another tab and as you know SDRangel only restores the first tab (R0) at re-start.

srcejon, 2021-11-29:
Just press the select sampling device then select [0:1]

f4exb, 2021-11-29:
Ah yes... each Rx is seen as a different device in the list of devices. However this is not reflected in the settings as in fact it has not to be there because you cannot change it dynamically (it is part of the device object initialization data). But what is saved or restored is the so called "working preset" and is indeed a preset (from Preset class) which is based only on settings information.

So it works as designed and hence the issue is a fundamental design issue.

f4exb, 2021-12-05:
The Rx (or Tx) number is what is called the `deviceItemIndex` in the `SamplingDevice` structure that represents the "device" in the device set: https://github.com/f4exb/sdrangel/blob/master/sdrbase/plugin/plugininterface.h#L67

However this information is not retained in the `DeviceConfig` structure of a `Preset`: https://github.com/f4exb/sdrangel/blob/master/sdrbase/settings/preset.h#L41

The save/restore mechanism uses a Preset to save and restore the information and therefore the `deviceItemIndex` is not retained. When restoring it will take the first index (0).

f4exb, 2021-12-05:
In fact this happens here: https://github.com/f4exb/sdrangel/blob/master/sdrgui/mainwindow.cpp#L233 and then the settings for the device once selected are applied from the working preset. `getRxSamplingDeviceIndex` does the job of finding the right device.

commit fcdeffa5 (2021-12-05): GUI flavor: save and restore the device item index for the initial set up (R0). Fixes #1066
closed 2021-12-05 by f4exb