f4exb/sdrangel #1067: No "satellite Map" on the Ubuntu 21 10 version

choupette-change, 2021-11-29:
There's no "Satellite Map" on the Ubuntu version, ok Windows version.
[image]

srcejon, 2021-11-29:
It's working for me:

[image]

Can you send a screenshot of your Map settings window?

Note that the list of maps is dynamic - so if there was a temporary problem connecting to the satellite map provider, it may not appear in the drop down. Try starting SDRangel again and see if it appears.

Also check log file for any Map/SSL messages.

choupette-change, 2021-11-29:
I restarted SdrAngel with the last version it's ok but the display is "capricious", I have to close the feature map and reopen it to show the "satellite map" tab when I restart sdrAngel. For the map plugin it's always ok .... :(
[image]

closed 2022-03-06 by github-actions[bot]

protektwar, 2022-03-13:
I have a white screen where the map should be.... 
[image]

srcejon, 2022-03-13:
Can you post your log file? Quite possibly a missing dependency.

protektwar, 2022-03-13:
Where do I find the log file?

srcejon, 2022-03-13:
Preferences > Logging menu.

protektwar, 2022-03-13:
Done, attached...
[sdrangelLog.log.zip](https://github.com/f4exb/sdrangel/files/8239306/sdrangelLog.log.zip)

srcejon, 2022-03-13:
2022-03-13 10:59:49.444 (W) qrc:/map/map.qml:5:1: module "QtPositioning" is not installed
2022-03-13 10:59:49.444 (W) qrc:/map/map.qml:4:1: module "QtLocation" is not installed

See: https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux for packages to install

protektwar, 2022-03-13:
[image]

srcejon, 2022-03-13:
Do you have qtlocation5 as well as qtlocation5-dev?

protektwar, 2022-03-13:
I have copy/paste -d the long package list: the only new installed packages have been these, I think the mission was qml qtlocation and qtpositioning, now it's working.
The following NEW packages will be installed:
  doxygen libhidapi-dev libusb-dev libxml2-dev qml-module-qtlocation qml-module-qtpositioning

Thanks!