f4exb/sdrangel #1068: Improvement proposal

choupette-change, 2021-12-01:
Can we provide data output in .kml format for use with Google Earth for AIS and ADSB decoders so as to have the trajectories in 3D .. ?
[image]

f4exb, 2021-12-01:
AIS is for ships on the sea so is 3D really useful?

choupette-change, 2021-12-01:
Of course not, my 3d request is mainly for aircraft trajectories but I am also interested in outputting AIS data in .kml format if possible :)

srcejon, 2021-12-03:
I guess the best place for this, would be in the Map feature, so that all the different objects the map supports can be exported, and only needs to be implemented in one plugin.

srcejon, 2022-02-06:
The SDRangel Map feature now supports 3D:

[image]

[image]

Do you really want .kml as well, or will this do?

choupette-change, 2022-02-06:
Hello, I haven't tested the 3D maps yet where can we find them and the
V6.19? If possible i want .kml output.
Greetings

f4exb, 2022-02-06:
v 6.19 is yet to be released but the 3D map feature is already in the master branch

choupette-change, 2022-02-07:
error when installing the latest master:

make[2]: ***
[pluginssrv/channelrx/demodapt/CMakeFiles/demodaptsrv.dir/build.make:147 :
pluginssrv/channelrx/demodapt/CMakeFiles/demodaptsrv.dir/aptdemodimageworker.cpp.o]
Erreur 1
make[2]: *** Attente des tâches non terminées....
[ 41%] Building CXX object
pluginssrv/channelrx/noisefigure/CMakeFiles/noisefiguresrv.dir/noisefiguresettings.cpp.o
[ 42%] Building CXX object
pluginssrv/channelrx/noisefigure/CMakeFiles/noisefiguresrv.dir/noisefigurebaseband.cpp.o
make[1]: *** [CMakeFiles/Makefile2:12640 :
pluginssrv/channelrx/demodapt/CMakeFiles/demodaptsrv.dir/all] Erreur 2
make[1]: *** Attente des tâches non terminées....
[ 42%] Building CXX object
pluginssrv/channelrx/noisefigure/CMakeFiles/noisefiguresrv.dir/noisefiguresink.cpp.o
[ 42%] Building CXX object
pluginssrv/channelrx/sigmffilesink/CMakeFiles/sigmffilesinksrv.dir/sigmffilesinkbaseband.cpp.o
[ 42%] Building CXX object
pluginssrv/channelrx/noisefigure/CMakeFiles/noisefiguresrv.dir/noisefigureplugin.cpp.o
[ 42%] Building CXX object
pluginssrv/channelrx/noisefigure/CMakeFiles/noisefiguresrv.dir/noisefigurewebapiadapter.cpp.o
[ 42%] Building CXX object
pluginssrv/channelrx/sigmffilesink/CMakeFiles/sigmffilesinksrv.dir/sigmffilesinksink.cpp.o
[ 42%] Building CXX object
pluginssrv/channelrx/sigmffilesink/CMakeFiles/sigmffilesinksrv.dir/sigmffilesinksettings.cpp.o
[ 42%] Building CXX object
pluginssrv/channelrx/sigmffilesink/CMakeFiles/sigmffilesinksrv.dir/sigmffilesinkmessages.cpp.o
[ 42%] Building CXX object
pluginssrv/channelrx/sigmffilesink/CMakeFiles/sigmffilesinksrv.dir/sigmffilesinkwebapiadapter.cpp.o
[ 42%] Building CXX object
pluginssrv/channelrx/sigmffilesink/CMakeFiles/sigmffilesinksrv.dir/sigmffilesinkplugin.cpp.o
[ 42%] Linking CXX shared library
../../../lib/pluginssrv/libnoisefiguresrv.so
[ 42%] Linking CXX shared library
../../../lib/pluginssrv/libsigmffilesinksrv.so
[ 42%] Built target noisefiguresrv
[ 42%] Built target sigmffilesinksrv
make: *** [Makefile:171 : all] Erreur 2
***@***.***:~/SDRangel/sdrangel/build$

srcejon, 2022-02-07:
You don't appear to have pasted the actual error.

However, please make sure that you've git pulled and rebuilt / installed aptdec.

choupette-change, 2022-02-07:
thanks for the tip, i reloaded aptdec and it works. I will test the 3d cards :)

srcejon, 2024-04-14:
Implemented in 7.20

closed 2024-04-14 by srcejon