f4exb/sdrangel #1074: Add remote control to Remote Input device

srcejon, 2021-12-09:
I think it would be useful if the Remote Input device supported some basic remote control. 

That is to say, can it be enhanced to allow setting of the center frequency on the remote device? It already has a (read only) widget to display the center frequency, and it already has settings for the API on the remote device, so it seems like it should be relatively straight forward to allow the center frequency to be set in the Remote Input device, by having it make an API call to the remote device setting centerFrequency.

Common device settings like sample rate and decimation could also be supported in a similar way.

A DeviceSet parameter would probably need to be added.

f4exb, 2021-12-10:
There is no provision to know which device and which remote sink channel is actually sampling at the other end (Airspy, RTL-SDR, LimeSDR ...) so it is not possible to derive any controls. I think it should still be device agnostic. For now I do not see any practical way to do this. You can have full control of the distance instance by other means using the API (Python script, SDRangelCli).

f4exb, 2021-12-10:
It is more or less the equivalent of giving each device plugin the capability to control either a device actually connect to the system or a distant device. This seems excessively complex when you have other means of control.

f4exb, 2021-12-10:
The center frequency of the received stream depends on two things: 
  - The center frequency of the device
  - The shift frequency of the Remote Sink channel
 
There is no simple way to control this from a single frequency value.

srcejon, 2021-12-10:
Yes, this would need a Device Set parameter to support more than one device (much like we currently have in some of the Features).  

I don't think you'd need to know the Sink channel index, if you just want to change the Device center frequency / sample rate as it doesn't depend on the channel (assuming Sink decimation is 1). If you wanted to be able to set the Sink decimation and shift frequency, then you would need a Channel Index as well. In that case, you could add widgets for both of those settings in the Remote Input GUI as well, giving a user convenient control over what is being streamed all from the client.

For most devices, I don't think you'd need to know which specific SDR is being used just to set the centreFrequency - as they mostly appear to use the same setting name in the API (there may be an exception as I haven't checked them all). Unfortunately, this doesn't appear to be the case for device sample rate, as some appear to use indexes rather than actual sample rate values. That could probably be dealt with by looking at the deviceHwType string, but not ideal. Making the API have consistent settings for all devices, where possible, would be nicer. However, supporting sample rate changes isn't as important as the centreFrequency, in my opinion,

Yes, you can change the frequency via Python, but that isn't very usable in some use cases - E.g. Just imagine you are interactively scanning through some range of frequencies looking to see what is out there (Looking for DAB multiplexes for example). You don't want to have to switch to an editor, change the frequency in a script, run it, then switch back to SDRangel. Much nicer just to scroll on the frequency widget. With this change, using a remote device would be a more similar user experience to using a local device.

f4exb, 2021-12-10:
To use the API you need to know which device you are talking to since it is in the device hardware ID field. From there you can also derive anything specific beyond center frequency and sample rate. I think you need either full control or no control from the GUI. Therefore you also need to control all aspects of the remote sink channel. Passing host, port, device and channel index in the UDP meta data is not a problem (there is still plenty of room) and from there you can construct any API endpoint URL.

So I see the new interface divided in 3 parts:

  - the "local" part: anything controlled locally like DC block and IQ balance so what it is today in fact
  - the remote device part with controls that depend on the type of device. Specialized sub-widgets may be used that are more or less clones of the existing device plugin UIs
  - the remote sink channel part controls that are fixed

You do not necessarily need to edit and run a Python script to change parameters. The web app [SDRangleCli](https://github.com/f4exb/sdrangelcli) lets you control all aspects of the remote instance interactively. In the end by implementing this you are just re-writing parts of SDRangelCli and it is probably much more work than dealing with a web app.

Also using the frequency dial to display the center frequency let the user think he/she may be able to control the frequency remotely. This is obviously a lie and I may at least change this in the future replacing it by a simple label maybe with a larger font. I Already had to twist the display to show sub kHz values since with the possibility to decimate and move the remote sink channel in the baseband the exact center frequency may not always fall on an integer kHz value.

f4exb, 2021-12-22:
Let's be more minimalistic. As requested it is enough to be able to control the frequency and possibly the sample rate (hence the bandwidth). It would be possible and device type agnostic to route the device frequency setting through the Remote Sink channel. Moreover the center frequency (and sample rate) of the stream is not governed only by the device center frequency (and same for sample rate). The Remote Sink channel can use decimation and specific placement in bandwidth (with the half decimators chain) resulting in the actual center frequency and sample rate of the stream. Therefore it makes sense that control goes through the Remote Sink channel.

As developed earlier some information can be routed in the meta data of the stream. In particular the device set and channel index is required to adress the Remote Sink channel via API. Also for practicality the device center frequency and sample rate (that could be different of the stream center frequency and sample rate) can be included in the meta data.

The device center frequency and sample rate may only be accessible via API since if running with the GUI there are direct means to set these parameters (this would just bring confusion).

Thus the controls from the Remote Input could be:
  - Remote device frequency
  - Remote device sample rate
  - Remote sink channel decimation
  - Remote sink channel placement

These controls would be added to the GUI and it will still show the resulting stream center frequency and sample rate.

f4exb, 2021-12-22:
Note that similar controls can be put in place in the Remote Output device.

srcejon, 2021-12-22:
Sounds great. Thanks.

f4exb, 2021-12-26:
Implemented in v6.17.6

closed 2021-12-26 by f4exb