f4exb/sdrangel #1077: Afedri sdr

klikis, 2021-12-11:
Is it possible to add the Afedri device? Specifically, the possibility of connecting to my Afedri sdr in LAN.

Thank you.

klikis, 2021-12-11:
p.s. for Windows

f4exb, 2021-12-11:
Not sure it is possible to find the support library

klikis, 2021-12-12:
Thank you for your answer, I don't understand much about programming if you go to this page of the producer there are all the files I don't know if they can be of any use to you. Thank you again.              https://www.afedri-sdr.com/index.php/downloads

f4exb, 2021-12-12:
Thanks. I also found this post in the Afedri group that seems to contain a bit more description along wit the links: https://groups.io/g/afedrisdr/topic/soapysdr_support/81409255?p=,,,20,0,0,0::recentpostdate/sticky,,,20,2,0,81409255,previd=1639251350139351019,nextid=1622994048760494357&previd=1639251350139351019&nextid=1622994048760494357 Note that I have the same feelings as Alex regarding SoapySDR so I will not make any effort to support Afedri via Soapy.

klikis, 2021-12-12:
Thank you for your thorough reply. Thank you.

ATEECS, 2021-12-25:
Hello all,
All AFEDRI SDR-Net protocols are open source , there is lot of information and C sources that shows how to use correctly the AFEDRI protocol.
Despite Metis protocol is supported - I do not recommend to use it, because it was built for different hardware , so RF gains controls are limited, maximal bandwidth also limited to 768kHz.
I recommend to use AFEDRI SDR-Net network protocol, that is modification/extension of SDR-IP (NetSDR) protocols.
This protocol allows to have full control over SDR hardware, without no limitations.
For example:
[sdr_commander -  source code](http://afedri-sdr.com/index.php/downloads/category/15-sdr-commander-command-shell-utility?download=296:sdr-commander-v-1-28-command-shell-utility)  gives examples how to control SDR.
s[dr-split utility source code](http://afedri-sdr.com/index.php/downloads/category/24-source-code-directory?download=257:sdr-split-utility-v1-19), shows how to process UDP datastream.
[SDR_network_Control_box](http://afedri-sdr.com/index.php/downloads/category/21-sdr-control-source-code?download=243:sdr-network-control-box-x4-v5-21h-source-code-2) - has both functions inside implemented (in poor programmer's way) control and data stream processing
So please forget about AFEDRI and Metis (it will work, but will be very limited), Metis has poor designed protocol (that may be good for FPGA based hardware)
73!
Alex

OK1UNL, 2021-12-27:
I am also user of Afedri LAN IQ  excellent hardware. Personally me invite Afedri support.
Afedri receivers seems are widely used over the world for more than 10years. 
Thanks to both F4EXB and Alex 4Z5LV. 
73! Ladislav