f4exb/sdrangel #1078: No reception at all with ADS_B

m1ctk, 2021-12-12:
I usually use RTL1090 with virtual radar with my flight aware dongle but I am getting nothing at all with the ADS-B plugin. Do I need to also add a mode demodulator like WFM or AM etc ?

srcejon, 2021-12-12:
No you don't need to add WFM or AM.

Make sure you have the sample rate (SR) set to 2,400,000 and decimation (Dec) set to 1 and the gain turned up.

f4exb, 2021-12-12:
We would have appreciated a more detailed problem description as suggested in the template you obviously did not use or follow. You can also check this post: https://groups.io/g/sdrangel/topic/windows/87672761?p=,,,20,0,0,0::recentpostdate/sticky,,,20,2,0,87672761,previd=1639298618313767692,nextid=1634287381471779344&previd=1639298618313767692&nextid=1634287381471779344 where you will find some screenshots.

m1ctk, 2021-12-12:
> No you don't need to add WFM or AM.
> 
> Make sure you have the sample rate (SR) set to 2,400,000 and decimation (Dec) set to 1 and the gain turned up.

OK thanks for info but still nothing at all

closed 2021-12-12 by f4exb