f4exb/sdrangel #1080: No SDRAngel waterfall on Nvidia Nano

alphafox02, 2021-12-12:
I believe this is to be expected maybe due to OpenGL but not for sure. I compiled sdrangel per latest manual Linux directions and installed into /opt.

When sdrangel starts there’s no middle graphical waterfall. Is there a way I should compile it when using it on a Nano with the custom Nvidia driver?

Thank you.

srcejon, 2021-12-13:
Look in the log and see what you get for:

GLSpectrum::initializeGL: context: major: 2 minor: 0 ES: yes

GLSpectrum::initializeGL: context: major: 3 minor: 0 ES: no

It needs to be major version 3.

On windows, you can set QT_OPENGL=software to get unaccelerated v 3 support - not sure if this will work on your platform though.

alphafox02, 2021-12-13:
Thank you, I’ll check again this evening and get that info and check to see if the Windows based workaround works.

I should’ve made note that the operating system on the Nano is upgraded to 20.04 from the official 18.04 release. I believe qt is at 5.12 and (per a friend helping me check it out as well) Qtdiag reports : 

QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled.

alphafox02, 2021-12-13:
Doesn’t like like either, so maybe some otjrr problem I need to address. 

https://pastebin.com/xrxaEmfz

alphafox02, 2021-12-13:
Closing as I don’t think this has anything to do with SDRAngel itself. I see other applications having issues here. However, I’ll keep looking for a way to run sdrangel. 

https://forums.developer.nvidia.com/t/teamviewer-crash-cannot-find-opengl/83600/10

closed 2021-12-13 by alphafox02

alphafox02, 2022-01-02:
@srcejon looking at this again, I’ve built a qt version on the nano which I think would now support SDRAngel use of OpenGL, but I wanted to ask does sdrangel support usr of no opengl? Meaning is there a fall back for the spectrum that would work without QTs use of OpenGL?

srcejon, 2022-01-02:
As far as I know (and I didn't write the OpenGL waterfall code) OpenGL is required.

The latest version of SDRangel should now display the Open GL version in the status bar (at the bottom) - so let us know what you get for that.

srcejon, 2022-01-02:
Also - see some notes on Qt for Embedded Linux here: https://doc.qt.io/qt-5/embedded-linux.html

alphafox02, 2022-01-02:
Thank you. I’m rebuilding later today. Also, I’ve built qt from source including the configs mentioned in the Nano forum that should enable “EGL” I think it’s called. This or maybe one of the other settings mentioned should create a solution to run OpenGL with SDRAngel. I just have to build SDRAngel from source against this install, but what’s unclear to me is if all the other required qt plug-ins exist in this alternate install. Sorry if that all sounds confusing, I’ll test and just respond if it works. :) 

Thank you for the links.

alphafox02, 2022-01-04:
@srcejon one more thought on this, have you known anyone that’s created an app image out of SDRAngel? I realize it may be a little bigger with all the dependencies included but thought maybe perhaps I could build a special aarch64 version with the proper config  for qt that would support OpenGL on the Jetson nano.

srcejon, 2022-01-05:
I'm afraid not.

This might provide some hints:

https://www.qt.io/blog/2017/01/18/opengl-implementation-qt-quick-app-using-today

SDRangel will use Qt Quick for the Map feature, so perhaps try setting QSG_INFO=1 and then opening the Map feature.

Also perhaps try LIBGL_ALWAYS_SOFTWARE=1 or 0.

If you've built Qt - perhaps you could try some of the Qt Open GL examples (Or Qt Quick) examples - and see what happens with those first.

Also, assuming you are using the latest version of SDRangel,  what Open GL version is displayed in the status bar (at the bottom of the window - perhaps send a screen shot)

alphafox02, 2022-01-05:
I’ll try again, this is only unique to the Jetson nano and basically this is seen in the SDRAngel (console output) so I’m almost positive the normal packaged version of QT in 20.04 for aarch64 is built in such a way that doesn’t support the Jetson’s Nano use of QT/OpenGL.

In fact, the same error occurs with the latest SigDigger (also qt based) when tryi mg to use OpenGL for the waterfall. In SigDiggers case OpenGL is optional so there’s a fall back. 

Following several guide to include the one mentioned as a solution here hasn’t worked out yet, but maybe I’ll eventually get it. I just thought possibly a portal appaimage with special built Qt for this situation may be handy. I’ll document all details and results and share ASAP.

https://forums.developer.nvidia.com/t/solved-qt-opengl-eglfs-egl-xcb-x11-wayland-jetson-board-tk1/36332