f4exb/sdrangel #1081: RadioSonde plug in

m1ctk, 2021-12-13:
I think this would be hugely popular with integrated map and the ability to track more than one balloon simultaneously etc much like the Pi image SQ6KXY - Quite a challenge but it would be great

srcejon, 2021-12-14:
Can you capture some IQ files of the signals they transmit?

cygeus, 2022-01-02:
Decoders for several common Radiosonde types can be found @ https://github.com/rs1729/RS/tree/master/demod/mod

I don't have any IQ recordings readily available.
A bit more info can be found @

https://www.sigidwiki.com/wiki/Vaisala_RS41-SG_Weather_Balloon_(Radiosonde)
https://www.sigidwiki.com/wiki/Vaisala_RS92-SGP_Weather_Balloon_(Radiosonde)
https://www.sigidwiki.com/wiki/Lockheed_Martin_LMS-6_(Radiosonde)
https://www.sigidwiki.com/wiki/Meteomodem_M10_Weather_Balloon_(Radiosonde)
https://www.sigidwiki.com/wiki/IMet-1_RS_Radiosonde
https://www.sigidwiki.com/wiki/Graw_DFM-06_Weather_Balloon_(Radiosonde)
https://www.sigidwiki.com/wiki/GRAW_DFM-09_Weather_Balloon_(Radiosonde)
https://www.sigidwiki.com/wiki/Graw_DFM-17_(Radiosonde)

srcejon, 2022-01-02:
I captured an rs41 signal, so should be able to get that working when I have finished the map updates, but other IQ files would still be helpful.

cygeus, 2022-01-08:
There are IQ recordings of a couple different radiosondes available at https://rfhead.net/sondes/sonde_samples.tar.gz (courtesy of the https://github.com/projecthorus/radiosonde_auto_rx project)

    rs41_96k_float.bin - Vaisala RS41, Serial Number N3920808, 120 packets
    rs92_96k_float.bin - Vaisala RS92, Serial Number M2513116, 120 packets
    dfm09_96k_float.bin - Graw DFM09, Serial Number 637797, 96 Packets
    m10_96k_float.bin - Meteomodem M10, 120 packets
    imet4_96k_float.bin - iMet-4, Serial Number 15236, 119 packets
    lms6-400_96k_float.bin - Lockheed Martin LMS6 (400 MHz variant), Serial number 8097164, 120 packets.
    imet54_96k_float.bin - iMet-54, Serial number 55064062, 240 packets.
    rsngp_96k_float.bin - Vaisala RS92-NGP (1680 MHz), Serial number P3213708, 120 packets
    mrz_96k_float.bin - Meteo-Radiy MRZ, ID MRZ-5667-39155, ~105 packets.

f4exb, 2022-03-05:
Implemented in v6.20.0

closed 2022-03-05 by f4exb