f4exb/sdrangel #1084: Crash in libusb_free_transfer

gvanem, 2021-12-16:
Downloaded latest release v
running with RTLSDR and the AM/WFM demodulators are okay.
Clicking the refresh-device (?) button:
[image]

to reselect a sampling-device, Sdrangel.exe crashes:
```
(3194.3720): Access violation - code c0000005 (first chance)
First chance exceptions are reported before any exception handling.
This exception may be expected and handled.
*** ERROR: Symbol file could not be found.  Defaulted to export symbols for f:\gv\dx-radio\SDRAngel\official-build\libusb-1.0.dll -
libusb_1_0!libusb_free_transfer+0x196:
00007ffd`0116cb1e 8b4128          mov     eax,dword ptr [rcx+28h] ds:feeefeee`feeeff0e=????????
```
Address `feeefeee'feeeff0e` seems completely bogus.
The call-stack:
```
0:000> kp
Child-SP          RetAddr           Call Site
000000c2`db3ca9e0 00007ffd`0116cc7b libusb_1_0!libusb_free_transfer+0x196
000000c2`db3caa20 00007ffd`0116e61c libusb_1_0!libusb_submit_transfer+0x8b
000000c2`db3caa60 00007ffd`05a81d30 libusb_1_0!libusb_control_transfer+0x158
000000c2`db3caad0 00007ffd`05a84acf rtlsdr!rtlsdr_get_device_name(unsigned int index = 0)+0x130
000000c2`db3cab20 00007ffc`f1882255 rtlsdr!e4k_compute_pll_params(struct e4k_pll_params *
   oscp = 0x00000000`476c3e50, unsigned int fosc = 0, unsigned int intended_flo = 0xdb3cfa80)+0x11f
000000c2`db3cab70 00007ffc`f1882104 inputrtlsdr!qt_plugin_instance+0x6c5
000000c2`db3cabc0 00007ffd`12d78313 inputrtlsdr!qt_plugin_instance+0x574
000000c2`db3cabf0 00007ffd`12d781fd sdrgui!MainWindow::sampleSourceChanged+0xd3
000000c2`db3caca0 00007ffc`de351da7 sdrgui!MainWindow::samplingDeviceChanged+0x5d
000000c2`db3cad00 00007ffd`12dd85c0 Qt5Core!QObject::qt_static_metacall+0x1467
000000c2`db3cae50 00007ffc`de351d49 sdrgui!SamplingDeviceDialog::reject+0xbd0
```

srcejon, 2021-12-16:
I just tried a few times and can't reproduce it. Does it happen every repeatedly for you?

gvanem, 2021-12-17:
It happens every time.

srcejon, 2021-12-17:
I see rtlsdr!e4k_compute_pll_param in your stack trace - my RTL SDR are  R820T based rather than E4000 - so could explain the difference.

gvanem, 2021-12-17:
FYI, the call-stack was from running inside the debugger: `cdb -c g SDRangel.exe`

Now, I've rebuilt SDRangel for `MSVC/x86` with this [RTL-SDR](https://github.com/old-dab/rtlsdr.git) library and the latest [libusb](https://github.com/libusb/libusb.git) from master. But not without a lot of patches to fix the `DEVICE_API` and other classes exports etc.  The `exports/export.h` file is rather in-complete.

And see no such crashes. So it has to be an issue with either `x64` and/or the original (?) OsmoSDR version SDRangel is using.
Or an old libusb? You tell me.

srcejon, 2021-12-17:
You could try using a newer libusb and your other rtlsdr library with the x64 build to see if it makes any difference.

closed 2022-03-24 by github-actions[bot]