f4exb/sdrangel #1085: SDRangel Frizes up startup

dagengstrom, 2021-12-17:
### Subject or type of issue
* Describe your issue here.
* SDRangel Frizes up startup. It was working for sevral day with out any notice it just stop at startup 
* * The problem started at versjon v6.17.4 and continue in v6.17.5
* Is this a runtime or compilation issue? YES

[image]
Dag Engstrøm
Norway

srcejon, 2021-12-17:
Anything relevant in the log file? (Typically  in C:\Users\yourusername\sdrangel.log)

Also, you could possibly try deleting current and current-featureset from the registry in HKEY_CURRENT_USER\SOFTWARE\f4exb\SDRangel so it starts with no plugins.

dagengstrom, 2021-12-17:
Hi,
Deleting HKEY_CURRENT_USER\SOFTWARE\f4exb\SDRangel did the trick but witch
of the plugins causes issues?
Dag

srcejon, 2021-12-17:
Well, it looks like the only one you have open is DAB. But I haven't seen a similar problem with that.

dagengstrom, 2021-12-17:
Do you have an over view witch of the keys belongs to witch plugin so I can
kill it and see if it solve the problem and let you know

srcejon, 2021-12-17:
I think all the channels are in current and features in current-feature as binary values, so not sure you can break it down in to individual plugins easily.

dagengstrom, 2021-12-17:
OK the DAB plugin now thank you for the help :-)

dagengstrom, 2021-12-17:
Hi
Is there a way to activate auto tune for DAB? Let it run thru all the
channels?
Dag

srcejon, 2021-12-17:
Not currently.

dagengstrom, 2021-12-17:
OK thanks
Dag

closed 2022-03-24 by github-actions[bot]