f4exb/sdrangel #1087: noisefigure/noisefigure.cpp:183:26: error: ‘barycentric_rational’ is not a member of ‘boost::math’

blshkv, 2021-12-24:
I found an old bug report (https://github.com/f4exb/sdrangel/issues/943) but the problem seems back.

https://github.com/pentoo/pentoo-overlay/issues/1054

the error: https://gist.github.com/asarubbo/28c0b567f564f17db2026f1e9b74eccd
the build log: https://gist.github.com/asarubbo/38dc24e1b44ac4c96b5a5aa42eccc574

dev-libs/boost-1.77.0-r3

A workaround was proposed here:
https://stackoverflow.com/questions/13955627/boost-math-special-functions-compilation-error-when-using-boost-for-tr1

```
We were able to work around this by replacing

#include<boost/random.hpp>
with the more specific to our needs:

#include<boost/random/mersenne_twister.hpp>
#include<boost/random/uniform_int.hpp>
#include<ctime>
thus leaving out the problematic headers.
```

f4exb, 2021-12-28:
On most systems the package installation uses earlier versions of Boost that do not exhibit the issue. Ex: 1.74 with Ubuntu 21.10. Now Boost 1.78 is out. I would be curious to know if this has been fixed or not in this version.

f4exb, 2021-12-28:
This seems related to very old versions of gcc.

blshkv, 2021-12-28:
So again, you can find all related details in the log:
sys-devel/gcc:            11.2.0::gentoo
sys-devel/gcc-config:     2.5-r1::gentoo

srcejon, 2021-12-28:
#943 is a different issue. The constructor used wasn't available in the earlier version of boost that the user had.

I've just tried boost 1.77 and 1.78 on Windows, and I do see a compilation failure. I don't think the proposed workaround is relevant though, as we don't actually include boost/random.hpp.

It appears that in boost 1.77 boost::math::barycentric_rational has moved to boost::math::interpolators::barycentric_rational

https://www.boost.org/doc/libs/1_76_0/libs/math/doc/html/math_toolkit/barycentric.html
https://www.boost.org/doc/libs/1_77_0/libs/math/doc/html/math_toolkit/barycentric.html

f4exb, 2022-01-03:
Implemented in v6.17.7

closed 2022-01-03 by f4exb