f4exb/sdrangel #1088: Contour shuttleXpress and V.2

OK1UNL, 2021-12-25:
Please is a possible add support for Contour shuttleXpress and v.2 too?? 
This "unit" is a commercial IT component widely available and used as a "replace" VFO tuning knob and some others SDR SW works and support the "unit" too.
This "VFO" knob allows fast changing frequency, mode, frequency step etc....  
Thank you for support 
VY 73! Ladislav OK1UNL

f4exb, 2021-12-25:
I doubt very much this could be supported. It is Windows and Mac only and closed source. In any case this would be via an external program (SDRangel can be controlled via HTTP API) or a feature plugin.

OK1UNL, 2021-12-25:
Thanks for fast replay.
Here exist plug-in
https://sdrplugins.com/downloads
another one solution is here
https://codiweb.be/hdsdr/hdsdr.html
Maybe this help you. 
VY 73!

f4exb, 2021-12-25:
These links are specific to certain software so not of much help here. 

If the Contour jog dials would trigger Qt events then it would be possible to tie these events to some actions. This seems to be the case according to this post: https://forum.qt.io/topic/60914/how-to-not-allow-qt-to-handle-jog-dial-shuttle-input-device Then it would be possible to create a feature plugin where any Qt event could be tied to this or that control of this or that device or channel. It would be up to the user to make the connections in the plugin. 

Thus rather than being specific to Contour devices I'd rather see a generic jog dial support feature plugin. You may then use any joystick or game pad the same way as long as it is able to produce Qt events.

OK1UNL, 2021-12-27:
I must agree with you, probably  this mean " run on long distance" So sorry, I am not programmer, only active user with Linux distro Raspbian any experince. Aka "lama" 
Thank you for support.
Ladislav OK1UNL

f4exb, 2022-01-03:
It appears that the "shuttles" can work by associating their keys to specific keystrokes. Qt can very well handle key events this is what is already done in the commands shortcuts: https://github.com/f4exb/sdrangel/blob/master/sdrgui/readme.md#4b63-edit-command-or-command-group So some keystrokes could be associated to some actions in a generic keystroke control feature.

f4exb, 2022-01-03:
On Linux one can use this program to do the key bindings: https://github.com/nanosyzygy/ShuttlePRO
Works also for ShuttleXpress (keys are K5 to K9) from left to right around the jog dial.

f4exb, 2022-01-09:
Implemented in v6.18.0

closed 2022-01-09 by f4exb