f4exb/sdrangel #1089: adding extra modules for pager decing

djflip, 2021-12-26:
Would like to see as a future feature, the pager decoder module updated to support other protocols besides POCSAG, as many of the pager systems in my area (Australia) besides being POCSAG, also run on FLEX 1600 as well as F1 and F2, would be good to see them  added so there is no need to also be running PDW, wile also having a SDR receiver running in the background

srcejon, 2021-12-26:
Can you capture some IQ files using the File Sink?

f4exb, 2022-04-18:
Back as "enhancement"

bmn001, 2022-06-13:
I'd use this as well.  FLEX 3200 is very common in my area (Los Angeles).  POCSAG not so much.  Thanks!

robbietee, 2025-03-15:
Could we please have a FLEX Pager decoder added please? IQ file attached.

[Flex Paging.2025-03-15T20_41_37_902.zip](https://github.com/user-attachments/files/19266262/Flex.Paging.2025-03-15T20_41_37_902.zip)

robbietee, 2026-09-15:
Still waiting patiently for the pager decoder to be enhanced. I did try and do it myself using AI to edit the code for the decoder, but after nearly getting it right AI started to do its own thing and broke the plugin completely. I don't know enough about writing code to see where it was going wrong.

srcejon, 2026-10-27:
FLEX added in v8.

closed 2026-10-27 by srcejon