f4exb/sdrangel #1091: Full install script.

ve3elb, 2021-12-27:
Does anyone know if there is a full SDRangel install script raspberry pi/Linux. 
To install it manually take a lot of time because you have to use single commands for every plugin and dependencies. Or should there just be a linux Deb full install package.

f4exb, 2021-12-28:
For Raspberry you have a Docker image in Docker Hub: https://hub.docker.com/r/f4exb06/sdrangelsrv
This is the server variant. I do not recommend the GUI variant on the Pi anyway although some people had some success with it.
More documentation and helper scripts to start the image or a compose stack with the web UI and spectrun UI is available at https://github.com/f4exb/sdrangel-docker. The Docker files producing the image are there also if you want to build it yourself (with patience).

There is no "script" as such. Somebody suggested in another issue to use Ansible playbooks which would probably be the way to go.

hiperiondev, 2021-12-28:
I keep the native rpm for Fedora. Maybe someone could do the same for Raspbian .

https://copr.fedorainfracloud.org/coprs/lu3vea/sdrangel/

toxa1983poloskov, 2022-03-19:
Qt5WebEngineConfig.cmake
    qt5webengine-config.cmake

debian 11 bullseye (Raspberry Pi OS) how to install it on ?
 it is not in the official repository

closed 2022-05-24 by github-actions[bot]