f4exb/sdrangel #1093: cross platform

Observerist, 2021-12-30:
Presented as being cross platform windows, linux & macOS... but nowhere to find binaries for macOS ?

srcejon, 2021-12-30:
You can build for Mac using macports

https://github.com/f4exb/sdrangel/wiki/Build-for-Mac-using-Macports

f4exb, 2021-12-30:
And this is documented in the Wiki: https://github.com/f4exb/sdrangel/wiki/Build-for-Mac-using-Macports

Observerist, 2021-12-30:
Yeah, saw that on the wiki, thanks. Wished there was a binary instead of having to build with macports.
Tried that anyway on my Mojave run Mac, only to get another "failed" message.
Macports is updated, that part won't be the culprit. 
Not geeky enough to be comfortable enough using Macports, Qt or Homebrew. 

If you could evaluate this fail, would be really kind of you. Here the Terminal returns :

Warning: The macOS 10.14 SDK does not appear to be installed. Ports may not build correctly.
Warning: You can install it as part of the Xcode Command Line Tools package by running `xcode-select --install'.
--->  Computing dependencies for libgcc
The following dependencies will be installed:  libgcc11
Continue? [Y/n]: y
--->  Activating libgcc11 @11.2.0_3
Error: Failed to activate libgcc11: Image error: /opt/local/include/gcc/c++/algorithm is being used by the active libgcc port.  Please deactivate this port first, or use 'port -f activate libgcc11' to force the activation.
Error: See /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_release_tarballs_ports_lang_gcc11/libgcc11/main.log for details.
Error: Unable to execute port: upgrade fftw-3-single failed

srcejon, 2021-12-30:
You could try:

xcode-select --install

sudo port selfupdate
sudo port upgrade outdated

Observerist, 2022-01-04:
Thank you for this info.

My Xcode version for Mojave is updated to ver. 11.3 .... and is already bloated up to the 20 GB mark.

The error message is highly annoying 

---> Activating libgcc11 @11.2.0_3
Error: Failed to activate libgcc11: Image error: /opt/local/include/gcc/c++/algorithm is being used by the active libgcc port. Please deactivate this port first, or use 'port -f activate libgcc11' to force the activation.

What the heck could be an "active libgcc port" on my Mac ?

But I'll give it another try. 
Even though I'm NOT a programmer or developer, nor do I like to use macports, homebrew & Co.
I'm not a Mac user for nothing  :-D
Spending time trying to understand a language from another world is time lost on me.
I don't have the intelligence of you guys   :-(

So thank you for helping me out on this task.

Cheers

srcejon, 2022-01-04:
A similar issue is documented here: https://trac.macports.org/ticket/58746

That suggests to run:

sudo port -f uninstall libgcc
sudo port install libgcc
sudo port sync

bmn001, 2022-01-05:
I'm also confused how to get this running in Linux.

It says it's cross platform, but only a Windows version is available for download here on Github?

I just saw there's a Linux snap available from the author, and I installed it and it seems to work fine.  However it's version 5.9.0, which is from August 18th, 2020 - over a year ago.

Is there an official way to run the latest version of SDRangel in Linux?

srcejon, 2022-01-05:
There's usually a tar file on the release page (https://github.com/f4exb/sdrangel/releases) that contains .deb in the assets section.

I noticed it's missing for v6.17.7, but it's available for v6.17.6: https://github.com/f4exb/sdrangel/releases/download/v6.17.6/sdrangel-1220-master.tar.gz

bmn001, 2022-01-05:
Ah ha!  That's what was throwing me.  The latest release was missing the .deb.  I didn't think to check previous releases.

I'm now trying to figure out the dependencies to get this working in Ubuntu 21.10.

Thanks very much for the pointer!

> There's usually a tar file on the release page (https://github.com/f4exb/sdrangel/releases) that contains .deb in the assets section.
> 
> I noticed it's missing for v6.17.7, but it's available for v6.17.6: https://github.com/f4exb/sdrangel/releases/download/v6.17.6/sdrangel-1220-master.tar.gz

f4exb, 2022-01-06:
My bad... the upload of the .deb files are manual and I forgot to do it. It is there now.

bmn001, 2022-01-06:
I'm really struggling to install this on Ubuntu 21.10...
```
sudo dpkg -i sdrangel_6.17.6-1_amd64.deb 
Selecting previously unselected package sdrangel.
(Reading database ... 262635 files and directories currently installed.)
Preparing to unpack sdrangel_6.17.6-1_amd64.deb ...
Unpacking sdrangel (6.17.6-1) ...
dpkg: dependency problems prevent configuration of sdrangel:
 sdrangel depends on libopencv-core4.2 (>= 4.2.0+dfsg); however:
  Package libopencv-core4.2 is not installed.
 sdrangel depends on libopencv-imgcodecs4.2 (>= 4.2.0+dfsg); however:
  Package libopencv-imgcodecs4.2 is not installed.
 sdrangel depends on libopencv-imgproc4.2 (>= 4.2.0+dfsg); however:
  Package libopencv-imgproc4.2 is not installed.
 sdrangel depends on libopencv-videoio4.2 (>= 4.2.0+dfsg); however:
  Package libopencv-videoio4.2 is not installed.

dpkg: error processing package sdrangel (--install):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.34-0ubuntu3) ...
Processing triggers for mailcap (3.69ubuntu1) ...
Processing triggers for gnome-menus (3.36.0-1ubuntu1) ...
Processing triggers for desktop-file-utils (0.26-1ubuntu2) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Errors were encountered while processing:
 sdrangel
```
Then once I try to install the dependencies...apt just removes sdrangel altogether:
```
sudo apt install -f
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Correcting dependencies... Done
The following packages will be REMOVED:
  sdrangel
0 upgraded, 0 newly installed, 1 to remove and 1 not upgraded.
1 not fully installed or removed.
After this operation, 66.5 MB disk space will be freed.
Do you want to continue? [Y/n] n
Abort.
```
If I try to install the dependencies by name, I get this:
```
sudo apt install libopencv-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libopencv-dev is already the newest version (4.5.3+dfsg-1ubuntu1).
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 sdrangel : Depends: libopencv-core4.2 (>= 4.2.0+dfsg) but it is not installable
            Depends: libopencv-imgcodecs4.2 (>= 4.2.0+dfsg) but it is not installable
            Depends: libopencv-imgproc4.2 (>= 4.2.0+dfsg) but it is not installable
            Depends: libopencv-videoio4.2 (>= 4.2.0+dfsg) but it is not installable
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```
I just installed whatever version of libopencv that apt pulled down.

Trying ```sudo apt --fix-broken install``` as prompted will just remove sdrangel again.

I'm not sure how to fix this.

f4exb, 2022-01-06:
The .deb file is for Ubuntu 20.04

bmn001, 2022-01-07:
Ah that would explain it.  Thanks for the clarification.

I just pasted all the lines from https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux into a single bash script, ran it, and that did the trick.  I'm good in 21.10 now.

toxa1983poloskov, 2022-02-23:
Raspberry Pi OS ( debian 11 Bullseye)
there is no file in the repository qt5webengine.    
the build requests this dependency.  

where to download and how to install this dependency ?

N5BRG, 2022-02-23:
if you do a search on the web for:

raspberry pi install qt5webengine

you will get directions on how to install qt5webengine.

toxa1983poloskov, 2022-02-23:
Couldn't find the package qt5webengine   and  qt5webengine-dev

toxa1983poloskov, 2022-02-23:
I'm afraid of breaking the system.
give a link to detailed instructions on exactly what you need to do to build QT5

srcejon, 2022-05-14:
A MacOS test build dmg can be downloaded from: 

https://www.sdrangel.org/downloads/sdrangel-6.20.3-1550-gcbacc162b_11.6.5_x86_64.dmg

Would be useful to get some feedback on which versions of MacOS it works and which SDRs.

Observerist, 2022-05-14:
Not working under Mojave 10.14.6. 

If only this app had been build using an older OS than Big Sur, with the X-Code Tools that came with Mojave, before Big Sur.
In the Ham/SWL world, older laptops are usually repurposed for distinct duties, at least that's true in the Windows World.
Having to use the very latest macOS version for an app like SDRangel is an idea somehow hard to swallow.
Using a VM like Parallels & Co. is not a graceful solution.

Who then managed to get the Angel to fly ??

srcejon, 2022-05-17:
Here's today's build of V7 compiled on Mojave:

https://www.sdrangel.org/downloads/sdrangel-7.0.0-1688-g58116df00_10.14.6_x86_64.dmg

srcejon, 2022-05-23:
MacOS DMG now on releases page.

closed 2022-05-23 by srcejon

loewal, 2022-05-23:
This is heaven Jon!
Thank you very much. 
I also gave it a try on a spare Catalina-machine. Got an illegal hardware call. Nothing to worry, because I use Macports on that and that compiles v6 fine. 

Thanks again.

srcejon, 2022-05-23:
It was compiled on a machine with AVX2 - I guess I could use -DARCH_OPT=nehalem for the next build as appears to be used for ubuntu.