f4exb/sdrangel #1097: ADS-B  HW support

OK1UNL, 2022-01-09:
Please is a possible check support of Flightaware Prostick Plus??
This very good inexpensive piece of HW is based on RTL2832U + R820T2.
I tried connect it to last version SDRangel.  ( Win10Pro 64bit) 
SDRangel cannot found (recognize) this one device. 
On other hand I tried this one HW with Avare apk. on android smartphone and works well.
Also this "blue dongle" works with others ADS-B SW well.  
Thank you for effort!
VY 73! Ladislav OK1UNL

srcejon, 2022-01-10:
Does it work with any other SDR software on the same Windows machine? 

Have you installed Zadig etc? Eg. first step here: https://discussions.flightaware.com/t/how-to-install-pro-stick-dvb-t-on-windows/25070

OK1UNL, 2022-01-10:
Thanks for question, my answer YESS. On this one Windows machine me using  multiple RTL-SDR based hardware and SW too.
Under SDR# running original RTL-SDR v.3 dongle, Airspy R2 too. Zadig driver must run on this one PC machine.
At SDR# Flightaware Prostick Plus  dongle does not work. Reason is clear. Author and his team made "clean code" of SDR#
Me using also VRS or ADSBscope 2.7 dinosaurus together with RTL1090 ver.1 and 2 too. 
I am participate at Flightaware project 24/7/365 and my RX there Flightaware Prostick Plus on RPi 3B+. This is based on linux distro Raspbian. 
Generally I have not problem with this one cheap but powerfull HW. ( in reality RTL-SDR v.3 optimized for 1090MHz)  
SDRangel does not detect this one "blue dongle" Why?? Maybe PID, VID problem ??? 

Thanks for reading and help for community. 
VY 73! Ladislav OK1UNL

closed 2022-04-06 by github-actions[bot]