f4exb/sdrangel #1098: OGN/FLARM support

OK1UNL, 2022-01-09:
Please is a possible integrate support of OGN/FLARM 868MHz to SDRangel??
http://wiki.glidernet.org/ogn-receiver-hardware-and-software#preamp
This one link maybe help you. 

OGN/FLARM are fully compatible. This system is an "like ADS-B" but with really small power od transmitters.
Receivers are built on RTL-SDR technology.  It is for gliders, private helicopters etc.....
Thank you VY 73! Ladislav OK1UNL

srcejon, 2022-01-09:
I believe FLARM is encrypted, so probably not.

OK1UNL, 2022-01-10:
I am not sure see here:
This one must be open protocol fully compatible Open Glider Network with Flarm. 
https://www.youtube.com/watch?v=HUi3sv-0r2g
Ladislav

srcejon, 2022-01-10:
I don't speak German unfortunately, but according to this: https://github.com/glidernet/ogn-rf/blob/master/INSTALL ogn-decode is closed source.

There has been some reverse engineering of FLARM - Version 6 details are here: https://pastebin.com/YK2f8bfm but it seems Version 7 is now in use - which presumably uses a different encryption method.

closed 2022-04-06 by github-actions[bot]