f4exb/sdrangel #1102: Decoding multiple

srs4511351, 2022-01-14:
I have a SDRplay RSP1A with API 3.07 running on a Raspberry Pi 4 B.

I am having trouble decoding more than one APRS signal at the same time.
I can open two Packet Demodulator channels, tune them to two different APRS frequencies and separately decode the packets.
When I add two APRS features, they both display data from R0:0 PacketDemod although one APRS has R0:1 PacketDemod selected. I expect each APRS feature to decode packets from the selected Packet Demodulator.

I set up APRS :
In Channels, I opened two Packet Demodulators
Tuned one to 144.39 MHz and clicked play.
Tuned one to 144.93 MHz and clicked play.

In Features, opened two APRS features and one Map.
In the first APRS, I selected R0:0 PacketDemod and in the second APRS I selected R0:1 PacketDemod.

In map settings, clicked Show settings dialog and chose a map.

The APRS locations appeared on the map as they were received.

Am I doing this correctly?

How do you tell the difference between the two Packet Demodulators other than the order that they appear in the window?

----Steve

srcejon, 2022-01-14:
Sounds like you are doing it correctly, so could be a bug.

srcejon, 2022-01-14:
(previous response deleted as incorrect)

The Sources combo box shows all sources that the APRS feature is receiving packets from - it doesn't allow you select packets from only one demodulator.

The order in the window is how you tell them apart: RM:N means it's the N'th demod in the M'th Device Set.

srs4511351, 2022-01-14:
I was confused by the ability to choose one of the packet demods in the Sources combo box pulldown menu. As you indicate, it does nothing but list the demods in use.

Running it with only the station that was previously in R0:1 PacketDemod, I see that the APRS feature is not getting any data from it.

The Packet Demodulator is receiving data, including From and To, etc., but the Data (ASCII) sometimes has non-printable characters. It also has some of the same messages as the APRS station on the standard 144.39 MHz frequency.
It mentions Winlink, which must not be compatible with APRS.

So, with that in mind, it appears that the APRS feature is working correctly as I do get APRS data from one station.

As for identifying the N'th demod, I don't see anything in the demods indicating which number they are. Since you cannot move them around, we must assume that they are always in order as displayed in the window.

----Steve

srcejon, 2022-01-15:
Feel free to post up a screenshot showing the data.

srs4511351, 2022-01-16:
Here are two screen shots. The first image with no data in the APRS feature has Winlink. I suspect that the apparent scrambled data is due to compression.

The second image is from another station. There are many stations that appear in the APRS feature.

[image]
[image]

----Steve

srcejon, 2022-01-16:
APRS packets have PID f0, so the ones with cf are a different protocol.

In the first screenshot,  the data in the packets with PID f0 isnt in APRS format (it doesn't start with : ! Or @). 

So looks like it's working as it should.

srs4511351, 2022-01-16:
Now, I just need a program for each format.

Thanks for the help.

----Steve

closed 2022-01-16 by srs4511351