f4exb/sdrangel #1103: ADS-B broken flight tracks

MightyPork, 2022-01-14:
My ADS-B decoder often adds a fixed horizontal offset to plane position, resulting in broken zigzag paths.
Good chance I set something wrong, but I can't figure out what.

SDRangel 6.17.7, receiving with NooElec NeSDR on Arch Linux with a ground-plane antenna.

Logfile: 
[sdrangel.txt](https://github.com/f4exb/sdrangel/files/7868475/sdrangel.txt)

Setup and the bug:
[image]
[image]
[image]
[image]

My antenna:
[image]

I'm quite sure the reception itself is good, dump1090 does not have this problem:
[image]

srcejon, 2022-01-14:
Have you set your position correctly  under Preferences > My position

MightyPork, 2022-01-14:
You're right, I did not. I didn't even know that option exists! It works better with the setting.

Out of curiosity, what do you think was causing the glitch? Do ADS-B frames send the position with ambiguity that must be resolved this way?

srcejon, 2022-01-14:
Yep, they don't always broadcast full coordinates - see: https://mode-s.org/decode/content/ads-b/3-airborne-position.html

There may be other ways to handle it - but best to set your position, as it's used for various things in SDRangel.

MightyPork, 2022-01-14:
Okay, I'm closing this then. Perhaps the readme could be more explicit about the necessity to set the position.

From friefly reading the source, it looks like dump1090 waits for two messages and uses the following algorithm for disambiguation http://www.lll.lu/~edward/edward/adsb/DecodingADSBposition.html

closed 2022-01-14 by MightyPork

srcejon, 2022-01-29:
I'll update the readme. I'll also see if I can display a message if My Position is set to the default values.

The plugin does use the global decoding that requires two messages, but it also does a local decode that can work on just one position message + position of the receiver.