f4exb/sdrangel #1104: Compiling DSP/symsyncpp

gvanem, 2022-01-14:
I see that `sdrbase/dsp/symsync.cpp` is not meant to be used; it's not listed in `CMakeLists.txt`.

But trying to compile it with MSVC/clang-cl anyway, there are several errors. Like:
```
DSP/symsync.cpp(48,85): error: invalid operands to binary expression ('float' and '_Fcomplex' (aka '_C_float_complex'))
    liquid_float_complex y = (s.m_real / SDR_RX_SCALEF) + (s.m_imag / SDR_RX_SCALEF)*I;
                                                          ~~~~~~~~~~~~~~~~~~~~~~~~~~^~
```

AFAICS, a  `liquid_float_complex` is always a structure in C++ and adding `-D_LIBCPP_COMPLEX` generates
many warnings. But with this patch, it compiles:
```diff
--- a/dsp/symsync.cpp 2020-11-14 18:16:15
+++ b/dsp/symsync.cpp 2022-01-14 15:56:17
@@ -45,7 +45,7 @@
 {
     unsigned int nn;
     Real v = -1.0f;
-    liquid_float_complex y = (s.m_real / SDR_RX_SCALEF) + (s.m_imag / SDR_RX_SCALEF)*I;
+    liquid_float_complex y = { s.m_real / SDR_RX_SCALEF, s.m_imag / SDR_RX_SCALEF };
     symsync_crcf_execute(m_sync, &y, 1, m_z, &nn);

     for (unsigned int i = 0; i < nn; i++)
@@ -72,7 +72,7 @@
 liquid_float_complex SymbolSynchronizer::runZ(const Sample& s)
 {
     unsigned int nn;
-    liquid_float_complex y = (s.m_real / SDR_RX_SCALEF) + (s.m_imag / SDR_RX_SCALEF)*I;
+    liquid_float_complex y = { s.m_real / SDR_RX_SCALEF, s.m_imag / SDR_RX_SCALEF };
     symsync_crcf_execute(m_sync, &y, 1, m_z, &nn);

     for (unsigned int i = 0; i < nn; i++)
```

But I fail to understand why this file is here for.

f4exb, 2022-01-14:
Let's not keep things that are obviously unused. I will double check that.

f4exb, 2022-01-14:
Not used outside symsunc.h symsync.cpp thus will be removed.

commit 68edf32c (2022-01-15): Removed unused symbol synchronization object and irrelevant references to liquidsdr. Fixes #1104
closed 2022-01-15 by f4exb