f4exb/sdrangel #1105: Crash loading preset or during startup

DBMandrake, 2022-01-17:
Hi All,

I'm running SDR Angel 6.18.0 on Windows 10 64 bit, with an RTL-SDR.COM v3 USB adaptor. This adaptor is stable on the other SDR software I use.

The issue I'm seeing is frequent crashes when trying to load presets, and typically once a crash occurs loading a preset the program will no longer load past the splash screen without crashing as it keeps trying to automatically load the preset which caused the previous crash. The only way I can get out of this situation is to delete the current-featureset registry key as described elsewhere here.

The crash I'm seeing looks similar to https://github.com/f4exb/sdrangel/issues/845 however I'm running on Windows not Linux so I wasn't sure if it was just superficially similar or not.

The preset I'm trying to create is fairly simple - an AM demodulator and two POCSAG decoders. Here is a screenshot of the configuration before saving the preset:

[image]

When I try to load the preset the program will crash. The windows event log will show an error similar to this:

```
Faulting application name: sdrangel.exe, version: 0.0.0.0, time stamp: 0x61dabdf4
Faulting module name: Qt5Core.dll, version: 5.15.2.0, time stamp: 0x5fa4dd3b
Exception code: 0xc0000005
Fault offset: 0x00000000000ba4ae
Faulting process ID: 0x3650
Faulting application start time: 0x01d80b3343e79fd4
Faulting application path: C:\Program Files\SDRangel\sdrangel.exe
Faulting module path: C:\Program Files\SDRangel\Qt5Core.dll
Report ID: 2b49107f-5e0d-4cab-a47e-fbbcae54b242
Faulting package full name: 
Faulting package-relative application ID: 
```

The following is a debug log at the time of the crash:

[SDRAngel-crash-log.log](https://github.com/f4exb/sdrangel/files/7878223/SDRAngel-crash-log.log)

Here is a zipped copy of the registry keys including the preset etc:

[SDRAngel-crash-reg-keys.zip](https://github.com/f4exb/sdrangel/files/7878226/SDRAngel-crash-reg-keys.zip)

The strange thing is that I have previously been able to save and reload a preset for this basic POCSAG configuration without crashes, and the only difference between now and when it wasn't crashing was different settings for sample rate, bandwidth etc in the R0 window in the top left, however I don't remember what the exact settings were that were not crashing, as I've tweaked many of them, and I've been unable to find a combination again which does not crash.

srcejon, 2022-01-25:
I've tried loading your preset about 10 times with the same version of SDRangel, and it doesn't crash for me.

DBMandrake, 2022-01-25:
What SDR hardware are you testing with ?

Does it crash if you change the frequency, sample rate and other device related settings, start the SDR device to apply the settings, then try to reload the preset ? It seems to be related to the preset changing SDR device settings.

It's crashing so frequently for me that I've had to create a batch file to delete the two registry keys so that it doesn't keep loading the previous presets and crashing on startup. :(

srcejon, 2022-01-25:
With a RTL SDR Blog v3, as you said you were.

I tried loading the preset with the SDR running and stopped. Changing frequency and sample rate made no difference.

You might want to see this issue too: https://github.com/f4exb/sdrangel/issues/1039 - it seemed to help using a different version of the rtl sdr dll.

Other than that, all I can really suggest is that you install a debugger (Visual Studio Community for Free - https://visualstudio.microsoft.com/vs/community/ - install with Desktop development for C++ and MFC) and then I send you a debug build to try.

DBMandrake, 2022-01-25:
Hi,

I tried exchanging the version of rtlsdr.dll with the one I compiled from https://github.com/winterrace/librtlsdr for use with rtl_433, which as far as I know is the latest most maintained version of the library unfortunately it still crashes.

I already have Visual Studio 2022 installed so I can certainly give this a try, although I may need a little guidance on how to capture the crash data you're looking for as I've only done basic debugging in VS before.

srcejon, 2022-01-25:
I believe the other problem was solved by using an earlier build of the DLL.

To debug, download and install this version of SDRangel: https://www.sdrangel.org/builds/sdrangel-6.18.1-274-gef0377cf4-win64.exe

Run a Visual Studio native x64 native tools command prompt (Should be under Start Menu)
cd "C:\Program Files\SDRangel"
devenv /DebugExe .\SDRangel.exe
(Press ignore if it asks if you want to restart the application with elevated permissions)
Press Start button (Or press F5)
When it crashes, send a screenshot showing the Call Stack

DBMandrake, 2022-01-25:
Ok this is interesting, I've installed the above beta version in a different directory so I can switch back and forth between the two versions, however they presumably share all their preferences settings, saved pre-sets etc as they will be in the same location in the registry. (It seems to be the case anyway)

I thought I'd test this beta version on it's own before even firing up Visual Studio and so far I have been unable to crash it. In fact when 6.18.0 crashes on startup (after showing the splash dialogue but before the main window appears) due to it trying to auto-load the last pre-set / configuration (which would normally require me to delete the two registry keys) the 6.18.1-274 beta launches and works fine, including restoring the "problematic" pre-sets.

Also no amount of changing of RTL-SDR settings or reloading of pre-sets causes it to crash.

So apparently whatever this bug is is already fixed in the beta version! When you were trying to reproduce the crash were you using 6.18.0 or this latest beta build ?

srcejon, 2022-01-25:
There are some changes, but nothing that should intentionally have any effect on the problem you reported. Might be worth doing a diff to see which files have changed.

srcejon, 2022-01-25:
Well, the versions of the code should be the same - but they have been compiled differently and with different options and device support.

Couple of things you can try:

You could copy rtlsdr.dll and libusb-1.0.dll from the working version to the non-working version and see if it now works. Or the reverse and see if it stops working.

Another thing to try might just be to delete some of the other devices I haven't included in the build from your nonworking version, so: bladeRF.dll hackrf.dll libiio.dll inputairspy.dll inputbladerf*.dll inputhackrf.dll inputplutosdr.dll mimobladerf*.dll outputbladerf*.dllo outputhackrf.dll outputplutosdr.dll plutodevice.dll - They shouldn't cause any problems, but who knows...

DBMandrake, 2022-01-25:
> Well, the versions of the device libraries should be the same - but they have been compiled differently and with different options and device support.

Interesting.

> Couple of things you can try:
> 
> You could copy rtlsdr.dll and libusb-1.0.dll from the working version to the non-working version and see if it now works. Or the reverse and see if it stops working.

Copying these two files from the debug build to the normal build does not help unfortunately.

> Another thing to try might just be to delete some of the other devices I haven't included in the build from your nonworking version, so: bladeRF.dll hackrf.dll libiio.dll inputairspy.dll [...]

Tried deleting all these dll's on the non-working version but it still crashes.

I went a step further and copied sdrangel.exe and sdrangel.pdb from the debug installation to the normal installation - this was enough to give me the background debug window and as I suspected, the log file is not getting flushed when the crash occurs, with many lines missing from the log which are show in a screenshot of the debug window just before it disappears.

Here is the logging window a fraction of a second before the crash:

[image]

And here is the full log file of the full debug installation running which includes log entries beyond the point at which the normal version is crashing:

[SDRAngel-nocrash.log](https://github.com/f4exb/sdrangel/files/7937027/SDRAngel-nocrash.log)

And here is the version of the log when it crashes, but you'll note that some of the end of the log is missing vs the screen capture presumably due to the process crashing before the buffer is flushed to disk:

[SDRAngel-crash.log](https://github.com/f4exb/sdrangel/files/7937042/SDRAngel-crash.log)

Not sure if the log file is in a deterministic order - if not the above may not be of much use.

Not sure what to try next give that the debug build doesn't crash, and my attempts to mix and match dll's and debugging symbols between versions has not been successful. (Complaints about missing entry points etc)

srcejon, 2022-01-26:
Can you please try this build:

 https://www.sdrangel.org/builds/sdrangel-6.18.1-274-gef0377cf4-win64_rel.exe

If doesn't include debug symbols, but please can you try to debug it as outlined in my earlier post.

DBMandrake, 2022-01-26:
I'll give it a try tonight. I can only debug it if it doesn't crash of course. :)

srcejon, 2022-01-26:
It's the same source as the previous build I sent, except is built using the Release config rather than the RelWithDebugInfo config.

This should be nearly the same as 6.18.0. There are changes to some of the plugins, but not the ones you are using. If this crashes, then it suggests it may be something to do with the build config.

DBMandrake, 2022-01-29:
Sorry, I didn't get a chance to look at this further until now, however there seems to be some confusion over versions.

Three days before you posted a debug build for me to try the official version 6.18.1 was released, and I hadn't noticed or check out this version. When I test that official 6.18.1 build just now the easily reproducible crash when I load pre-sets does not occur.

So the problem seems to be fixed already in the 6.18.1 official build. I originally posted this issue when 6.18.0 was the current version.

I have seen the official 6.18.1 crash on me once when changing settings but I have been unable to reproduce it, but I think the issue I opened this ticket about may have already been fixed by 6.18.1.

DBMandrake, 2022-01-29:
Aha. I can reproduce the other crash I mentioned which is separate to the pre-set loading one that seems to be fixed now.

I recently (since I first opened this ticket) bought an Airspy Mini, so I now have both an rtl-sdr v3 and Airspy Mini plugged into the same PC.

If I start acquisition using the rtl-sdr v3, stop acquisition, then attempt to switch to the Airspy Mini in the sampling device list, it crashes every time. If I simply switch acquisition source immediately after starting the program without having started and stopped acquisition it does not crash.

Should I close this ticket and create a new one or continue here ? Is it worth trying to debug this with VS without debug symbols or do I need a new build with debug symbols ? (Because the original build with debug symbols you compiled for me omitted support for the Airspy Mini...)

srcejon, 2022-01-29:
Try this for Airspy support with debug:

https://www.sdrangel.org/builds/sdrangel-6.18.1-274-gef0377cf4-win64_airspy.exe

DBMandrake, 2022-01-30:
Unfortunately I can't get this debug build of yours to crash when switching sampling sources...

However I can get the 6.18.1 official release to crash when switching sources while being debugged, not sure whether the output is of any use without debug symbols but here it is anyway....

[image]

[image]

The steps I'm taking to trigger the crash:

1) Select RTL device, start acquisition with Pager demodulator running (not sure if the latter is important)
2) Stop acquisition.
3) Select Airspy from select sampling dialogue and press OK
4) Crash. (Sometimes)

The fact that it only crashes sometimes and doesn't seem to crash on a (slower) debug build makes me wonder if it's some kind of race condition.

srcejon, 2022-01-30:
Thanks. The stack trace suggests it's a crash inside libusb/rtlsdr.dll. There's actually a couple of other issues open with similar. Some people find an older version of the rtlsdr dll works for them. I don't know why. There's also a newer version libusb you should be able to try.

It's possible it's related to different USB controllers being used. You might try using a hub, or switching between USB 3 and 2 to see if that makes any difference. But it's a bit of guess work at this stage, unfortunately.

closed 2022-04-07 by github-actions[bot]