f4exb/sdrangel #1106: Feature request - export Satellite Data table from SatelliteTracker via API request

Valstee, 2022-01-17:
Hi @f4exb , Hi @srcejon, 
Is it possible to export with the API  this table : [Satellite Data](https://github.com/f4exb/sdrangel/tree/master/plugins/feature/satellitetracker#satellite-data) from Satellite Tracker ?
I want to get the list of the pass that my station has planned to receive during the day.
Because everything happens with the server version on a raspberry so I don't really have visibility except with the logs.
I can recreate the table with the library used by satellitetracker but a request via the API would be perfect.
thank you,
Valentin

srcejon, 2022-01-17:
Not currently,  but could be added to the feature report.

srcejon, 2022-02-06:
Above patch adds most of the data in the table and a full pass list. Let me know if you need anything else.

f4exb, 2022-02-08:
Deployed in v6.19.0

closed 2022-02-08 by f4exb