f4exb/sdrangel #1110: Difficulty compiling in Linux

QtheTech, 2022-01-23:
Good day,

I've run into some problems with compiling SDRAngel.  I've followed the instructions here, installed only the plugins for the hardware that I want to use (RTL-SDR and HackRF).  Prior to that, I installed all the recommended packages and non-hardware dependencies.

I'm running Linux Mint 20.2.  My Linux knowledge is relatively new, I've only been working in Linux for a little under two years, and I'm still learning, so I apologize in advance if I seem slow to understand some questions.  Hopefully I'll learn some more working through this issue!

The results of my cmake and make commands are pasted below the tear line.  I'm hoping I just missed something simple in the procedure, but I've gone over it and I just can't seem to see what it is I missed.  I'd appreciate any insights anyone has into what I could try to do a successful install.  

Thanks in advance!

Regards,
John

------------------------------------------------------------------------------------------------------------------------------------------
john@Acer-LM:/opt/build/sdrangel/build$ cmake -Wno-dev -DDEBUG_OUTPUT=ON -DRX_SAMPLE_24BIT=ON \
> -DBUILD_SERVER=OFF \
> -DCMAKE_BUILD_TYPE=RelWithDebInfo \
> -DMIRISDR_DIR=/opt/install/libmirisdr \
> -DHACKRF_DIR=/opt/install/libhackrf \
> -DRTLSDR_DIR=/opt/install/librtlsdr \
> -DSOAPYSDR_DIR=/opt/install/SoapySDR \
> -DAPT_DIR=/opt/install/aptdec \
> -DCM256CC_DIR=/opt/install/cm256cc \
> -DDSDCC_DIR=/opt/install/dsdcc \
> -DSERIALDV_DIR=/opt/install/serialdv \
> -DMBE_DIR=/opt/install/mbelib \
> -DCODEC2_DIR=/opt/install/codec2 \
> -DSGP4_DIR=/opt/install/sgp4 \
> -DLIBSIGMF_DIR=/opt/install/libsigmf \
> -DDAB_DIR=/opt/install/libdab \
> -DBUILD_SERVER=OFF \
> -DCMAKE_INSTALL_PREFIX=/opt/install/sdrangel ..
-- SDRangel Version: 6.18.0-32-g85184e2b4
-- Build type set to: RelWithDebInfo
-- Compiling for 24 bit Rx DSP chain
-- Target architecture: x86_64-native
-- Looking for __AVX__ - forced found
-- Looking for __SSE4_2__ - forced found
-- Looking for __SSE4_1__ - forced found
-- Looking for __SSSE3__ - forced found
-- Looking for __SSE3__ - forced found
-- Looking for __SSE2__ - forced found
-- Interprocedural optimization enabled
-- Found FFTW3F: /usr/include, /usr/lib/x86_64-linux-gnu/libfftw3f.so, /usr/lib/x86_64-linux-gnu/libfftw3f_threads.so
-- Checking for module 'faad2'
--   No package 'faad2' found
-- FAAD LIBRARY /usr/lib/x86_64-linux-gnu/libfaad.so
-- FAAD INCLUDE DIRS /usr/include
-- Checking for module 'sgp4'
--   No package 'sgp4' found
-- SGP4 LIBRARIES /opt/install/sgp4/lib/libsgp4s.so
-- SGP4 INCLUDE DIRS /opt/install/sgp4/include/SGP4
-- Checking for module 'apt'
--   No package 'apt' found
-- APT LIBRARIES /opt/install/aptdec/lib/libapt.so
-- APT INCLUDE DIRS /opt/install/aptdec/include/apt
-- Checking for module 'libairspy'
--   No package 'libairspy' found
-- libairspy not found.
-- Checking for module 'libairspyhf'
--   No package 'libairspyhf' found
-- libairspyhf not found.
-- Checking for module 'libbladeRF>=2.0'
--   No package 'libbladeRF' found
-- libbladerf not found.
-- Could not find Lime Suite
-- Checking for module 'libiio>=0.7'
--   No package 'libiio' found
-- libiio not found.
-- Checking for module 'libmirisdr'
[... 340 lines cut ...]
/usr/include/x86_64-linux-gnu/qt5/QtGui/qcolor.h:278:38: note:   conversion of argument 2 would be ill-formed:
/opt/build/sdrangel/sdrbase/dsp/spectrumsettings.cpp:227:18: error: cannot bind non-const lvalue reference of type ‘QColor&’ to an rvalue of type ‘QColor’
  227 |     in >> marker.m_show;
      |           ~~~~~~~^~~~~~
In file included from /usr/include/x86_64-linux-gnu/qt5/QtGui/QColor:1,
                 from /opt/build/sdrangel/sdrbase/dsp/spectrummarkers.h:23,
                 from /opt/build/sdrangel/sdrbase/dsp/spectrumsettings.h:27,
                 from /opt/build/sdrangel/sdrbase/dsp/spectrumsettings.cpp:24:
/usr/include/x86_64-linux-gnu/qt5/QtGui/qcolor.h:73:5: note:   after user-defined conversion: ‘QColor::QColor(QRgb)’
   73 |     QColor(QRgb rgb) Q_DECL_NOTHROW;
      |     ^~~~~~
/opt/build/sdrangel/sdrbase/dsp/spectrumsettings.cpp:222:14: note: candidate: ‘QDataStream& operator>>(QDataStream&, SpectrumAnnotationMarker&)’
  222 | QDataStream& operator>>(QDataStream& in, SpectrumAnnotationMarker& marker)
      |              ^~~~~~~~
/opt/build/sdrangel/sdrbase/dsp/spectrumsettings.cpp:222:68: note:   no known conversion for argument 2 from ‘SpectrumAnnotationMarker::ShowState’ to ‘SpectrumAnnotationMarker&’
  222 | QDataStream& operator>>(QDataStream& in, SpectrumAnnotationMarker& marker)
      |                                          ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
[ 24%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/util/visa.cpp.o
[ 24%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/util/weather.cpp.o
[ 24%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/plugin/plugininterface.cpp.o
[ 24%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/plugin/pluginapi.cpp.o
make[2]: *** [sdrbase/CMakeFiles/sdrbase.dir/build.make:965: sdrbase/CMakeFiles/sdrbase.dir/dsp/spectrumsettings.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:2602: sdrbase/FesqJohnACMakeFiles/sdrbase.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

f4exb, 2022-01-23:
Same failure in the Ubuntu 20.04 CI here: https://ci.appveyor.com/project/f4exb/sdrangel/builds/42296834/job/wrv7agql6a584x8t
It works in later releases of Ubuntu or Debian Bullseye. It is possibly due to the version of Qt not implementing `operator>>`in `QDataStream`for enums.

f4exb, 2022-01-23:
It looks like the problem is fixed: https://ci.appveyor.com/project/f4exb/sdrangel/build/job/i87fisf5oore11rs

closed 2022-01-23 by f4exb

QtheTech, 2022-01-23:
Your change got it working...successfully installed, using SDRAngel with my RTL-SDR now!  Thank you for figuring out the issue, and for putting out the new version.  Looking forward to exploring its features!

John