f4exb/sdrangel #1111: Resolve Airspy HF+ compatibility on 20.04 without kernel changes

alphafox02, 2022-01-23:
There’s a currently closed ticket #600 and I’m not sure if comments on it are noticed by anyone so this ticket is just to ensure the discussion is noticed. Delete or merge as anyone sees fit.

Recently I received the Airspy Discovery HF+ for testing in DragonOS. I find that it functions with all other applications without a kernel or libusb change, except for with SDRAngel.

There might be an issue in the plugin/Airspy code that with a kernel change may be getting masked or is basically a workaround - but there’s hope a kernel change is not needed. I’m hoping this may generate some thoughts on what to change because I’ve been unable to make headway besides learning how to format cdecl for gcc. 

The Airspy dev suggested initially this in your Airspyworker.h file

static int __cdecl rx_callback(airspyhf_transfer_t* transfer);

Here’s more info from the Airspy dev.

“maybe it's the ABI of the compiler used with the kernel

the ABI is only stable when using explicit calling conventions

it is the case with the library (check ADDCALL define)”

alphafox02, 2022-01-24:
SDRAngel compiled from source on aarch64 20.04 with a 5.4 kernel and ran on the Pi4 allows the Airspy HF+ Discovery to work. This is with the same version libusb as available on the X64 based 20.04 as well as a 5.4 kernel. As to what the differences are within the kernel or gcc, that I’m not sure.

srcejon, 2022-01-26:
I've just booted up the DragonOS R20 ISO in a VM (on a Windows host) - and my Airspy Discovery HF+ works in SDRangel (as was the case for #600).

alphafox02, 2022-01-26:
That’s so strange, I’m going to try it again. Like you actually get output in the spectrum window on the top area?

srcejon, 2022-01-26:
Yep.

There's definitely a problem there - as other people have reported it - but presumably it is specific to some combination of h/w and s/w.

Perhaps try a different USB port or USB hub (maybe USB 2 instead of 3) and see if that makes any difference.

srcejon, 2022-01-26:
So VM Workstation can set USB compatibility to USB 2.0 or USB 3.1.

It works when set to USB 2.
It doesn't work when set to USB 3.1.

alphafox02, 2022-01-26:
Same exact results for me! Here’s some additional info, the same laptop booted natively with Linux (DragonOS) resulted in the Airspy HF not working, meaning no rx even with a usb 2.0 hub used and plugged into a 3.0 port. 

Once the same laptop was booted into windows, VMware was used to conduct the same tests as you with the same hub. I switched from usb 3.0 to 2.0 in VMware and the Airspy HF Discovery worked on the same exact DragonOS with no kernel or libusb change.

alphafox02, 2022-01-27:
This is shown in dmesg on the laptop that's running Linux native with a USB 2.0 hub attached and Airspy HF Discovery plugged in. I start and stop SDRAngel to make this output occur.

[  153.350891] usb 3-6.6: EP not empty, refuse reset
[  153.520284] xhci_hcd 0000:00:14.0: Can't enqueue URB while manually clearing toggle
[  153.520291] usb 3-6.6: usbfs: usb_submit_urb returned -22
[  172.274706] usb 3-6.6: Process 7288 (DSPDeviceSource) called USBDEVFS_CLEAR_HALT for active endpoint 0x81

I see that even with the hub it's using the xhci_hcd driver. 

After running this on my laptop
lspci -nn | grep USB

I found the xHCI controller number out of this

00:14.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB xHCI [8086:8c31] (rev 04)
00:1a.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #2 [8086:8c2d] (rev 04)
00:1d.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #1 [8086:8c26] (rev 04)

Then ran this to force the USB 3 to use operate as USB 2.

sudo setpci -H1 -d 8086:8c31 d0.l=0

Unplugged the Airspy HF Discovery and plugged it back in, checked dmesg and could see it was using the ehci driver. Guess what, SDRAngel then worked with the Airspy HF.

However, I do notice this in dmesg now after stopping SDRAngel.

[  740.132396] CPU: 3 PID: 9622 Comm: DSPDeviceSource Tainted: P           OE     5.4.0-47-generic #51-Ubuntu
[  740.132404] RIP: 0010:ehci_endpoint_reset+0xfa/0x110
[  740.132408] Code: 80 be 84 00 00 00 01 75 a0 e8 32 e6 ff ff eb 99 e8 6b fb ff ff eb 92 48 c7 c7 98 ce 60 b8 c6 05 b6 f3 fe 00 01 e8 b1 2c 88 ff <0f> 0b e9 78 ff ff ff 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 40 00
[  740.132410] RSP: 0018:ffffb02283b87d20 EFLAGS: 00010082
[  740.132413] RAX: 0000000000000000 RBX: ffff9ddd08ac6d80 RCX: 0000000000000006
[  740.132415] RDX: 0000000000000007 RSI: 0000000000000092 RDI: ffff9dddce8d78c0
[  740.132417] RBP: ffffb02283b87d50 R08: 0000000000000565 R09: 0000000000000004
[  740.132418] R10: 0000000000000000 R11: 0000000000000001 R12: ffff9dddb7e10800
[  740.132420] R13: 0000000000000002 R14: 0000000000000297 R15: ffff9dddb7e10b34
[  740.132423] FS:  00007f5f8ffff700(0000) GS:ffff9dddce8c0000(0000) knlGS:0000000000000000
[  740.132425] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  740.132427] CR2: 00007f5f9bdfee20 CR3: 0000000741d04002 CR4: 00000000001606e0
[  740.132429] Call Trace:
[  740.132439]  usb_hcd_reset_endpoint+0x28/0x70
[  740.132444]  usb_clear_halt+0xb0/0xc0
[  740.132449]  usbdev_do_ioctl+0x970/0x1010
[  740.132455]  ? _copy_from_user+0x3e/0x60
[  740.132459]  usbdev_ioctl+0xe/0x20
[  740.132464]  do_vfs_ioctl+0x407/0x670
[  740.132468]  ksys_ioctl+0x67/0x90
[  740.132472]  __x64_sys_ioctl+0x1a/0x20
[  740.132480]  do_syscall_64+0x57/0x190
[  740.132488]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  740.132492] RIP: 0033:0x7f5ffb25a50b
[  740.132496] Code: 0f 1e fa 48 8b 05 85 39 0d 00 64 c7 00 26 00 00 00 48 c7 c0 ff ff ff ff c3 66 0f 1f 44 00 00 f3 0f 1e fa b8 10 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 55 39 0d 00 f7 d8 64 89 01 48
[  740.132497] RSP: 002b:00007f5f8fffdd38 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[  740.132500] RAX: ffffffffffffffda RBX: 0000557d830c5aa0 RCX: 00007f5ffb25a50b
[  740.132502] RDX: 00007f5f8fffdd44 RSI: 0000000080045515 RDI: 000000000000003d
[  740.132504] RBP: 0000557d83135940 R08: 00007f5f8fffdc70 R09: 00007f5fc033725e
[  740.132505] R10: 00007f5f8fffdc70 R11: 0000000000000246 R12: 00007f5f8fffddd0
[  740.132507] R13: 0000000000000001 R14: 00007f5f8fffddc8 R15: 00007f5f8fffddc8
[  740.132511] ---[ end trace 4905286fcb17c54a ]---
[  959.578018] usb 1-1.2: Process 10004 (DSPDeviceSource) called USBDEVFS_CLEAR_HALT for active endpoint 0x81
dragon@dragon:~$

alphafox02, 2022-02-03:
Does this information help shed light on the issue? At the very least I believe it shows the airspyhf + can be used with the standard kernel and libusb present in 20.04. 

What’s not clear is to why it’s necessary to force USB ports to 2.0 when other SDR applications on the same computer, same OS, can use the HF+ without forcing to usb 2.0.

alphafox02, 2022-02-09:
Would it be feasible that somehow the code that interfaces with the Airspy Hf+ force it’s USB port to 2.0 while in use or be modified to account for whatever is unique to SDRAngel on 20.04 with stock libusb and 5.4 kernel? I simply cannot replicate this issue with any other piece of software running the Hf+ on the same computer. Something so isolated as this would seem maybe something within SDRAngel could be tweaked opposed to needing to change the kernel or libusb as mentioned on the mailing list. 

Let me know what else I can provide to help with this.

closed 2022-04-17 by github-actions[bot]