f4exb/sdrangel #1113: Recent Release flagged as Malicious/Trojan | Verified Malicious with HybridAnalysis

EndlessEden, 2022-01-24:
(https://hybrid-analysis.com/sample/04649f422cbbddb11d0167273fc0ced4ffbced6f5ee1fb9b9d7cb633a6b2ec42)

Recent release (sdrangel-6.18.1-win64.exe) was immediately flagged by my Antivirus. The installer interacts with a infected vcredist in some suspicious ways. See report above.

-- This may still be a false-positive however hybrid-analysis confirms it is out of the ordinary. Please advise.

f4exb, 2022-01-24:
OK I am going to suspend all binary releases in the next versions.

srcejon, 2022-01-24:
No problems reported by Windows Defender.

The report you've linked to says:

1/54 Antivirus vendors marked sample as malicious (1% detection rate)

So I'd imagine it's a false positive.

EndlessEden, 2022-01-25:
@f4exb - I wasnt suggesting to lol. - I know full well how much work goes into this and other like projects. | Simply warning and sharing supplied information from research.
--
@srcejon - (https://www.virustotal.com/gui/file/04649f422cbbddb11d0167273fc0ced4ffbced6f5ee1fb9b9d7cb633a6b2ec42/detection) You are correct, it is false-positive. However ^^^ as i pointed out in the hybrid-analysis link, the vcredist package bundled in the installer acts like its /infected/ thus suggesting take a look if this is a normal process.

HA does sandboxing for malware analysis and while "Trojan.PSW.Python.fd" is almost always a false-positive, however the specific trojan specified often is often bundled within vcredist from infected mirrors... so i made a choice to bring it up here.

As i said, i was quite certain it was a false positive, i just wanted the community to take a look and give a final say and/or submit a request to have the false-positive removed.

srcejon, 2022-01-25:
It looks like vc_redist.x64.exe is picked up from here: https://github.com/f4exb/sdrangel-windows-libraries/tree/master/vcredist 

It is signed by Microsoft with a SHA256 signature, according to Windows Explorer.

It is 3 years old though, so we could probably pick up the latest version from https://aka.ms/vs/17/release/vc_redist.x64.exe (which is the latest permalink for it from https://docs.microsoft.com/en-US/cpp/windows/latest-supported-vc-redist?view=msvc-170), which also includes support for Visual Studio 2022.

srcejon, 2022-01-25:
PR with latest VC redist here: https://github.com/f4exb/sdrangel-windows-libraries/pull/10

Have run through building an installer and installing it, and it seems to run OK.

f4exb, 2022-01-25:
I have merged and updated the submodule reference. The latest build can be found at the bottom of this page: https://github.com/f4exb/sdrangel/actions/runs/1746380392 I think you need to be logged in to Github to be able to download.

Please let us know if this fixes the issue.

EndlessEden, 2022-01-25:
> 
> Please let us know if this fixes the issue.

(https://www.virustotal.com/gui/file/bce0656693752fd7c93650c5143424c462a0e1608fb81084af151f3f55cc743c) - Still comes back **false positive**, but confirmed atleast (see below) vcredist is clean. so recommend having the False Positive removed from their detection routines. | cant run HA analysis given file size limitations.

Just to confirm (incase anyone else comes looking) via VT/HA that both vcredist are clean 
VC_redist.x86.exe: (https://www.virustotal.com/gui/file/f02dea65ea65633d1718e6c5e5eee7d2df640d1fff332e4669dea530b8c4f0e7|https://hybrid-analysis.com/sample/f02dea65ea65633d1718e6c5e5eee7d2df640d1fff332e4669dea530b8c4f0e7)
`this can be confirmed and compared against: https://download.visualstudio.microsoft.com/download/pr/571ad766-28d1-4028-9063-0fa32401e78f/F02DEA65EA65633D1718E6C5E5EEE7D2DF640D1FFF332E4669DEA530B8C4F0E7/VC_redist.x86.exe`

VC_redist.x64.exe: (https://www.virustotal.com/gui/file/5d3d8c6779750f92f3726c70e92f0f8bf92d3ae2abd43ba28c6306466de8a144|https://hybrid-analysis.com/sample/5d3d8c6779750f92f3726c70e92f0f8bf92d3ae2abd43ba28c6306466de8a144) `this can be confirmed and compared against: https://download.visualstudio.microsoft.com/download/pr/571ad766-28d1-4028-9063-0fa32401e78f/5D3D8C6779750F92F3726C70E92F0F8BF92D3AE2ABD43BA28C6306466DE8A144/VC_redist.x64.exe`

Most of the "malicious" warnings are surrounding the WIX platform libraries bundled inside vcredist. But given these are signed installers that i personally confirmed are identical to the ones that MS distributed. The only conclusion is its 100% false-positive.

As a result i will close. (Bold is for TL;DR, if some one comes looking out of concern afterwards.)

closed 2022-01-25 by EndlessEden

EndlessEden, 2022-01-27:
Just a updated: "fftwf-wisdom.exe" is what is being flagged as a trojan. (Which i believe is just fftw3)  - you may want to check your packaged version of fftw is clean.