f4exb/sdrangel #1114: Markers are way to complicated to use

Kabuse, 2022-01-25:
I was very happy, when the markers finally has been introduced, but now that hey are here, I got  frustrated with it.

First of all, I was not able to use the GUI for creating all my markers. Somehow I was able to create two markers and export them to a .csv-file (not to the SDRangle directory, because it is owned by root). After that, I was able to edit the file and load it by the markers dialog (shouldn't be the save and load icons are vice versa?).

Then, I was a bit shocked that I had to click -> "Full" -> "Anno" -> "All" in order to make them visible.

When I noticed that I have to do this evey time, I change to another preset, I was totally devestated.

Why can there not just be a checkox "Show Markers", which takes all the information from the presets, including bandwith?

Honestly, I love to use SDRangel a lot, but this is just usability from hell.

f4exb, 2022-01-25:
> not to the SDRangle directory, because it is owned by root

I don't know how you managed this. This is a bad idea.

> shouldn't be the save and load icons are vice versa?

Export (not save) is the arrow outwards and import (not load) the arrow inwards. I guess this makes sense.

> I was a bit shocked that I had to click -> "Full" -> "Anno" -> "All" in order to make them visible.

At least you have a way to make them all visible in full all at once. Depending on many factors: the text length, the marker span, the spectrum width it is not always desirable to see all markers expanded encumbering the screen preventing sorting one from another.

> When I noticed that I have to do this evey time,

You don't. The visibility level (hidden, only the top or full) is persistent in the preset including the default "working" preset. The marker type display (no markers, spectrum markers, annotation markers) is also persistent. So I do not see what you are complaining about. The markers will not be displayed if nothing is displayed so you still need to make the spectrum display active once.

> Why can there not just be a checkox "Show Markers", which takes all the information from the presets, including bandwith?

This complaint is irrelevant all these parameters are persistent in the preset.

> Honestly, I love to use SDRangel a lot, but this is just usability from hell.

Then use a different software if you don't like SDRangel anymore. I do not see the point to change anything regarding points you mention that are irrelevant.

f4exb, 2022-01-25:
> not to the SDRangle directory, because it is owned by root

OK... maybe this one. Qt provides a default writable directory which should work whether you compile the code in your own build directory or you use a packaged binary.

commit 60a1955d (2022-01-25): Spectrum Markers dialog: export/import annotations .csv file in standard writable location. Fixes #1114
closed 2022-01-25 by f4exb