f4exb/sdrangel #1117: This coprocessor register access is deprecated in ARMv8

orlanmon, 2022-01-26:
# SDRangel Issue Template
Please let us know how you use the software, and what you enjoy, or don't enjoy the most.  
We are interested to find out how other people make use of this software.

#### From the Project Wiki:  
   *I ask you (as a User or Collaborator) to take the role of first level support.   
   It means you have to help me and not the opposite: you have to help me  
   understand the problem or what you think is a problem*

### Subject or type of issue
*  Compilation Error Raspbian - Raspberry PI4 - This coprocessor register access is deprecated in ARMv8 
* Is this a runtime or compilation issue?  Compilation
* If compilation, where in the instructions are you stuck? Linking libmodchirpchatsrv.so

### SDR
* We always want to first rule out that there is no issue with your hardware
* Which SDR are you using?
* Which version of Driver do you have loaded for the SDR?
* If applicable, Can you prove SDR functionality with a loopback test?
* Have you tried power cycling the SDR or the machine connecting to the SDR? Sometimes a reset is all you need.
 * Perhaps this is the issue, that the issue causes you to need to reset.

### Development environment
* Compilation instructions are given in terms of Ubuntu 21.04, have you tried to recreate the issue using this environment? No
* CPU and RAM of your machine if applicable

### Actual behavior
* Tell us what happens. -  Compilation Error Raspbian - Raspberry PI4 - This coprocessor register access is deprecated in ARMv8 during Linking libmodchirpchatsrv.so

### Expected behavior
* Tell us what should happen
* Helps us ensure that you understand what you are trying to do so that we can rule out any user error.

### Steps to reproduce
* Tell us how to reproduce this problem. Include screenshots, where applicable
* If radio related, have you tried to reproduce with a different radio, if possible?
* Please list Center Frequency, bandwidth and any other applicable RF settings that are pertinent to the issue.

### Log Files
* Please read any pertinant log files!
* Attach any relevant information you find here.

f4exb, 2022-01-26:
Usage of Raspbian is discouraged. As expected it assumes an armv7 environment and ignores the newer armv8 architecture that has better performance.

Please use [Ubuntu server](https://ubuntu.com/download/raspberry-pi) compiling the server variant (`-DBUILD_GUI=OFF`)

orlanmon, 2022-01-27:
Thank you for that information and I switching to a Minis Forum Z83-F with Ubuntu Server.   By any chance do you know if this would possibly work on a Linux Mint?  Much appreciated!

closed 2022-02-03 by f4exb