f4exb/sdrangel #1118: sdrangel-6.18.1: segmentation fault during each exit

yarda, 2022-01-27:
### Subject or type of issue
It segfaults during each shutdown. It's minor problem, but it's annoying.

### SDR
* We always want to first rule out that there is no issue with your hardware
* Which SDR are you using?
USRP1
* Which version of Driver do you have loaded for the SDR?
uhd-4.1.0.5
* If applicable, Can you prove SDR functionality with a loopback test?
Works
* Have you tried power cycling the SDR or the machine connecting to the SDR? Sometimes a reset is all you need.
Yes
 * Perhaps this is the issue, that the issue causes you to need to reset.

### Development environment
Fedora 34, x86_64, distro package, not official yet (I am trying to get it into the Fedora repo and became downstream maintainer).

### Actual behavior
* Tell us what happens.
It crashes during each shutdown.
* Be specific here, It is important that we be able to recreate the issue
A bit complicated to install, because the package is not yet approved and included into the official Fedora release:
- Deploy Fedora 34
- `dnf install mock`
- `gpasswd -a USER mock`
- `su - USER`
- `wget 'https://jskarvad.fedorapeople.org/serialdv/serialdv-1.1.4-1.fc34.src.rpm'`
- `mock -r fedora-34-x86_64 ./serialdv-1.1.4-1.fc34.src.rpm`
- `mock -r fedora-34-x86_64 --no-clean install /var/lib/mock/fedora-34-x86_64/result/serialdv-1.1.4-1.fc34.x86_64.rpm`
- `mock -r fedora-34-x86_64 --no-clean install /var/lib/mock/fedora-34-x86_64/result/serialdv-devel-1.1.4-1.fc34.x86_64.rpm`
- `wget 'https://jskarvad.fedorapeople.org/sdrangel/sdrangel-6.18.1-1.fc34.src.rpm'`
- `mock -r fedora-34-x86_64 --no-clean ./sdrangel-6.18.1-1.fc34.src.rpm`
- `dnf install /var/lib/mock/fedora-34-x86_64/result/sdrangel-6.18.1-1.fc34.x86_64.rpm`
- `sdrangel`
- Wait until it loads then exit it the normal way

* Please list any relevant information leading up to the issue

### Expected behavior
* Tell us what should happen
No crash
* Helps us ensure that you understand what you are trying to do so that we can rule out any user error.

### Steps to reproduce
* Tell us how to reproduce this problem. Include screenshots, where applicable
* If radio related, have you tried to reproduce with a different radio, if possible?
Yes, it's reproducible also without radio.
* Please list Center Frequency, bandwidth and any other applicable RF settings that are pertinent to the issue.

### Log Files
* Please read any pertinant log files!
* Attach any relevant information you find here.

Log (we do not have all plugins in the Fedora due to SW patents and because some of the support libs are not yet packaged into Fedora):
```
022-01-27 21:54:50.562 (I) SDRangel 6.18.1 Qt 5.15.2 64b x86_64 Fedora 34 (Workstation Edition) DSP Rx:24b Tx:16b PID 29616
2022-01-27 21:54:50.562 (I) MainSettings::MainSettings: settings file: format: 0 location: /home/yarda/.config/f4exb/SDRangel.conf
2022-01-27 21:54:51.235 (W) QMetaObject::connectSlotsByName: No matching signal for on_presetTree_currentItemChanged(QTreeWidgetItem*,QTreeWidgetItem*)
2022-01-27 21:54:51.235 (W) QMetaObject::connectSlotsByName: No matching signal for on_presetTree_itemActivated(QTreeWidgetItem*,int)
2022-01-27 21:54:52.813 (I) PluginManager::loadPluginsDir: loaded plugin libantennatools.so
2022-01-27 21:54:52.831 (I) PluginManager::loadPluginsDir: loaded plugin libaprs.so
2022-01-27 21:54:52.833 (I) PluginManager::loadPluginsDir: loaded plugin libbeamsteeringcwmod.so
2022-01-27 21:54:52.835 (I) PluginManager::loadPluginsDir: loaded plugin libchanalyzer.so
2022-01-27 21:54:52.879 (I) PluginManager::loadPluginsDir: loaded plugin libdemodadsb.so
2022-01-27 21:54:52.882 (I) PluginManager::loadPluginsDir: loaded plugin libdemodais.so
2022-01-27 21:54:52.883 (I) PluginManager::loadPluginsDir: loaded plugin libdemodam.so
2022-01-27 21:54:52.885 (I) PluginManager::loadPluginsDir: loaded plugin libdemodatv.so
2022-01-27 21:54:52.888 (I) PluginManager::loadPluginsDir: loaded plugin libdemodbfm.so
2022-01-27 21:54:52.890 (I) PluginManager::loadPluginsDir: loaded plugin libdemodchirpchat.so
[... 90 lines cut ...]
#11 0x00007f28f8afecae in QTabWidget::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) () at /lib64/libQt5Widgets.so.5
#12 0x00007f28f7e7f430 in void doActivate<false>(QObject*, int, void**) () at /lib64/libQt5Core.so.5
#13 0x00007f28f8ad8ab5 in QStackedWidget::widgetRemoved(int) () at /lib64/libQt5Widgets.so.5
#14 0x00007f28f7e7f430 in void doActivate<false>(QObject*, int, void**) () at /lib64/libQt5Core.so.5
#15 0x00007f28f89436f2 in QStackedLayout::widgetRemoved(int) () at /lib64/libQt5Widgets.so.5
#16 0x00007f28f8944399 in QStackedLayout::takeAt(int) () at /lib64/libQt5Widgets.so.5
#17 0x00007f28f893e0c0 in removeWidgetRecursively(QLayoutItem*, QObject*) () at /lib64/libQt5Widgets.so.5
#18 0x00007f28f891d427 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib64/libQt5Widgets.so.5
#19 0x00007f28f7e4b7d8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib64/libQt5Core.so.5
#20 0x00007f28f7e7bba7 in QObjectPrivate::setParent_helper(QObject*) () at /lib64/libQt5Core.so.5
#21 0x00007f28f7e7c0f9 in QObject::~QObject() () at /lib64/libQt5Core.so.5
#22 0x00007f28f9857c15 in FeatureWindow::~FeatureWindow() (this=0x55d155af7fd0, this=<optimized out>)
    at /usr/src/debug/sdrangel-6.18.1-1.fc34.x86_64/x86_64-redhat-linux-gnu/sdrgui/sdrgui_autogen/DMHXEJ42XS/../../../../sdrgui/gui/featurewindow.h:29
#23 FeatureWindow::~FeatureWindow() (this=0x55d155af7fd0, this=<optimized out>)
    at /usr/src/debug/sdrangel-6.18.1-1.fc34.x86_64/x86_64-redhat-linux-gnu/sdrgui/sdrgui_autogen/DMHXEJ42XS/../../../../sdrgui/gui/featurewindow.h:29
#24 0x00007f28f98fefda in FeatureUISet::~FeatureUISet() (this=0x55d15536d570, this=<optimized out>)
    at /usr/src/debug/sdrangel-6.18.1-1.fc34.x86_64/sdrgui/feature/featureuiset.cpp:38
#25 0x00007f28f98644fd in FeatureUISet::~FeatureUISet() (this=0x55d15536d570, this=<optimized out>)
--Type <RET> for more, q to quit, c to continue without paging--c
    at /usr/src/debug/sdrangel-6.18.1-1.fc34.x86_64/sdrgui/feature/featureuiset.cpp:35
#26 MainWindow::removeAllFeatureSets() (this=this@entry=0x7ffc819ee7e0) at /usr/src/debug/sdrangel-6.18.1-1.fc34.x86_64/sdrgui/mainwindow.cpp:695
#27 0x00007f28f98658a5 in MainWindow::~MainWindow() (this=0x7ffc819ee7e0, this=<optimized out>) at /usr/src/debug/sdrangel-6.18.1-1.fc34.x86_64/sdrgui/mainwindow.cpp:321
#28 0x000055d153c3f0cb in runQtApplication(int, char**, qtwebapp::LoggerWithFile*) (argc=<optimized out>, argv=<optimized out>, logger=0x55d154b9ec10) at /usr/src/debug/sdrangel-6.18.1-1.fc34.x86_64/app/main.cpp:132
#29 0x000055d153c3e6db in main(int, char**) (argc=1, argv=0x7ffc819eeac8) at /usr/src/debug/sdrangel-6.18.1-1.fc34.x86_64/app/main.cpp:138
```

f4exb, 2022-01-27:
It seems different systems behave differently. In Ubuntu 21.10 there is no segfault however adding some debug messages at strategic places there is indeed an issue:
```
2022-01-28 00:09:26.625 (D) MainWindow::removeAllFeatureSets: 1
2022-01-28 00:09:26.625 (D) MainWindow::tabFeaturesIndexChanged: -1
2022-01-28 00:09:26.625 (D) MainWindow::~MainWindow: end
```
The `removeAllFeatureSets` method is called only in MainWindow destructor. But at this time there is still the connection: 
```
connect(ui->tabFeatures, SIGNAL(currentChanged(int)), this, SLOT(tabFeaturesIndexChanged()));
```
Obviously `tabFeaturesIndexChanged` is called even when there is no more items in the feature tabs. Therefore at that time `ui->tabFeatures->currentIndex()` returns -1 and it tries to use it as index to `m_featureUIs`. 

In fact `tabFeaturesIndexChanged` should be disconnected before all tabs are removed.

closed 2022-01-27 by f4exb

yarda, 2022-01-28:
Thanks, I can confirm it fixed the problem.

commit 26b8619b (2022-01-27): MainWindow: fixed removal of all feature sets so that the tab index changed slot is disconnected before. Fixes #1118