f4exb/sdrangel #1119: sdrangel-6.18.1: USRP1 on the TX/RX switch the receiving antenna sometimes changes from the TX/RX to the RX2

yarda, 2022-01-27:
### Subject or type of issue
* Describe your issue here.
I have USRP1 with the SBX and WBX daughterboards, SBX is the primary daughertboard, uhd-4.1.0.5 driver and firmware are used. If I add source and sink both with the antenna set to the TX/RX antenna, sometimes (in more than the 50 % cases) the receiving antenna after the transmit changes to the RX2 (i.e. the second antenna on the daughterboard). It seems it cannot be switched back to the primary TX/RX antenna (in the UI there is still written TX/RX antenna and nothing happens if I try to switch it there). It can be switched back by doing few TX/RX cycles and then it is usually correctly set.

* Is this a runtime or compilation issue?
Runtime
* If compilation, where in the instructions are you stuck?

### SDR
* We always want to first rule out that there is no issue with your hardware
* Which SDR are you using?
Ettus USRP1 with SBX and WBX daughterboards.
* Which version of Driver do you have loaded for the SDR?
uhd-4.1.0.5
* If applicable, Can you prove SDR functionality with a loopback test?
it seems to work
* Have you tried power cycling the SDR or the machine connecting to the SDR? Sometimes a reset is all you need.
Yes, but the problem is still reproducible. Also restarting the sdrangel switched the antenna to the correct one.
* Perhaps this is the issue, that the issue causes you to need to reset.

### Development environment
* Compilation instructions are given in terms of Ubuntu 21.04, have you tried to recreate the issue using this environment?
The same envirnment and build as described in the #1118
* CPU and RAM of your machine if applicable
Intel(R) Core(TM) i7-4600U CPU @ 2.10GHz,  8 GB RAM, but it doesn't seem to be related to the problem.

### Actual behavior
* Tell us what happens.
I have sink and source, both set to the primary TX/RX antenna. Sometimes (in cca. 50 % of cases) the receiving antenna after the transmit mysteriously changes to the RX2 (i.e. the second antenna on the daughterboard). It cannot be switched back to the primary TX/RX antenna (in the UI there is still written TX/RX antenna and nothing happens if I try to switch it there). It seem there are multiple ways how to switch it back:
- do few TX/RX cycles until it is correctly set.
- restart sdrangel
- remove the source and re-add it back (just clicking the reload button doesn't help)
This can be also firmware issue, but I haven't encountered it yet with other software (e.g. gnuradio) and taking into account that the problem can be cured by removal and re-addition of the source (R0) I think there could be at least workaround for it in the sdrangel. 
* Be specific here, It is important that we be able to recreate the issue
* Please list any relevant information leading up to the issue

### Expected behavior
* Tell us what should happen
The receiving antenna shouldn't change without user intervention, it should correspond to what is set in the UI. If the receiver is set to the TX/RX antenna it should stick with the TX/RX antenna even after the TX.
* Helps us ensure that you understand what you are trying to do so that we can rule out any user error.

### Steps to reproduce
* Tell us how to reproduce this problem. Include screenshots, where applicable
* If radio related, have you tried to reproduce with a different radio, if possible?
Unfortunately, I don't have here another Ettus radio (different model).
* Please list Center Frequency, bandwidth and any other applicable RF settings that are pertinent to the issue.
I tried 23 cm and 70 cm bands with the SBX daughterboard, but the problem doesn't seem to be related to this settings.

### Log Files
* Please read any pertinant log files!
* Attach any relevant information you find here.

Log:
```
2022-01-27 23:07:16.442 (I) SDRangel 6.18.1 Qt 5.15.2 64b x86_64 Fedora 34 (Workstation Edition) DSP Rx:24b Tx:16b PID 33437
2022-01-27 23:07:16.443 (I) MainSettings::MainSettings: settings file: format: 0 location: /home/yarda/.config/f4exb/SDRangel.conf
2022-01-27 23:07:17.112 (W) QMetaObject::connectSlotsByName: No matching signal for on_presetTree_currentItemChanged(QTreeWidgetItem*,QTreeWidgetItem*)
2022-01-27 23:07:17.112 (W) QMetaObject::connectSlotsByName: No matching signal for on_presetTree_itemActivated(QTreeWidgetItem*,int)
2022-01-27 23:07:18.666 (I) PluginManager::loadPluginsDir: loaded plugin libantennatools.so
2022-01-27 23:07:18.683 (I) PluginManager::loadPluginsDir: loaded plugin libaprs.so
2022-01-27 23:07:18.685 (I) PluginManager::loadPluginsDir: loaded plugin libbeamsteeringcwmod.so
2022-01-27 23:07:18.687 (I) PluginManager::loadPluginsDir: loaded plugin libchanalyzer.so
2022-01-27 23:07:18.731 (I) PluginManager::loadPluginsDir: loaded plugin libdemodadsb.so
2022-01-27 23:07:18.733 (I) PluginManager::loadPluginsDir: loaded plugin libdemodais.so
2022-01-27 23:07:18.735 (I) PluginManager::loadPluginsDir: loaded plugin libdemodam.so
2022-01-27 23:07:18.737 (I) PluginManager::loadPluginsDir: loaded plugin libdemodatv.so
2022-01-27 23:07:18.739 (I) PluginManager::loadPluginsDir: loaded plugin libdemodbfm.so
2022-01-27 23:07:18.741 (I) PluginManager::loadPluginsDir: loaded plugin libdemodchirpchat.so
2022-01-27 23:07:18.747 (I) PluginManager::loadPluginsDir: loaded plugin libdemodfreedv.so
2022-01-27 23:07:18.749 (I) PluginManager::loadPluginsDir: loaded plugin libdemodnfm.so
[... 96 lines cut ...]
[WARNING] [MULTI_USRP] AGC is not available on this device.
[WARNING] [MULTI_USRP] AGC is not available on this device.
2022-01-27 23:08:38.323 (W) QOpenGLTexturePrivate::destroy() called but texture context 0x55781d71bde0 is not shared with current context 0x55781c1bf750.
Texture has not been destroyed
2022-01-27 23:08:38.323 (W) QOpenGLTexturePrivate::destroy() called but texture context 0x55781d71bde0 is not shared with current context 0x55781c1bf750.
Texture has not been destroyed
2022-01-27 23:08:38.323 (W) QOpenGLTexturePrivate::destroy() called but texture context 0x55781d71bde0 is not shared with current context 0x55781c1bf750.
Texture has not been destroyed
2022-01-27 23:08:38.323 (W) QOpenGLTexturePrivate::destroy() called but texture context 0x55781d71bde0 is not shared with current context 0x55781c1bf750.
Texture has not been destroyed
2022-01-27 23:08:38.323 (W) QOpenGLTexturePrivate::destroy() called but texture context 0x55781d71bde0 is not shared with current context 0x55781c1bf750.
Texture has not been destroyed
2022-01-27 23:08:50.317 (C) AudioInput::openDevice: could not open audio source
2022-01-27 23:08:50.551 (C) AudioInput::applySettings: failed to open device  ""
2022-01-27 23:08:50.637 (C) AudioInput::applySettings: failed to open device  ""
[INFO] [USRP1] Opening a USRP1 device...
[INFO] [USRP1] Using FPGA clock rate of 64.000000MHz...
2022-01-27 23:09:04.869 (C) USRPInput::applySettings: could not set clock  "internal"
2022-01-27 23:09:04.869 (C) USRPInput::applySettings: could not set clock  "internal"
2022-01-27 23:09:04.952 (C) USRPInput::applySettings: could not set clock  "internal"
[WARNING] [MULTI_USRP] Setting IQ imbalance compensation is not possible on this device.
[WARNING] [MULTI_USRP] AGC is not available on this device.
[WARNING] [MULTI_USRP] AGC is not available on this device.
2022-01-27 23:09:17.639 (C) USRPInput::acquireChannel: Failed to lock LO
```

f4exb, 2022-01-27:
https://kb.ettus.com/USRP1

yarda, 2022-01-27:
> https://kb.ettus.com/USRP1

Unfortunately, it seems there is nothing related to this problem (or at least I wasn't able to find it).

My next steps will be extraction of the minimal reproducer from the sdrangel code and with it I will ask on the Ettus USRP forum, but I am new to sdrangel, so it will take some time.

f4exb, 2022-01-27:
What about the big red text in that page:
**Please note that this product is now End-of-Life (EOL), and is no longer available for sale through Ettus Research, and is not recommended for use in new designs or in new projects.**

and
```
[WARNING] [MULTI_USRP] Setting IQ imbalance compensation is not possible on this device.
[WARNING] [MULTI_USRP] AGC is not available on this device.
[WARNING] [MULTI_USRP] AGC is not available on this device.
2022-01-27 23:09:17.639 (C) USRPInput::acquireChannel: Failed to lock LO
```
Pretty obvious isn't it?

yarda, 2022-01-27:
> What about the big red text in that page: **Please note that this product is now End-of-Life (EOL), and is no longer available for sale through Ettus Research, and is not recommended for use in new designs or in new projects.**
>
It's open hardware and it is still fully supported by the UHD driver. Anybody can build it. There is no need to buy it from the Ettus.

> and
> 
> ```
> [WARNING] [MULTI_USRP] Setting IQ imbalance compensation is not possible on this device.
1) This just say that the HW doesn't support IQ imbalance.

> [WARNING] [MULTI_USRP] AGC is not available on this device.
2) This just say that the HW doesn't support AGC
> [WARNING] [MULTI_USRP] AGC is not available on this device.
> 2022-01-27 23:09:17.639 (C) USRPInput::acquireChannel: Failed to lock LO

This error is probably not related, I tried different things during the session, and something on my last try failed, see the previous lines from the log.

ad 1) and 2) if the client find out that the HW doesn't support AGC and IQ balance it shouldn't provide controls in the UI (or the controls should be grayed out).

f4exb, 2022-01-28:
If it really "Failed to lock LO" it will not lead you very far...
In short it was never assumed that the USRP plugin would support all USRPs this is mainly for B200' series.

f4exb, 2022-01-28:
I do not have a USRP1 so I am unable to improve or correct things for the USRP1. Feel free to do a PR.

f4exb, 2022-01-28:
If you compile from source I think it enables debug output by default but maybe this has been disabled in the Fedora package (I do not own this part). In the log options you can control the console message level but I think that for debug it should be enabled as a whole at compile time (there is some Qt variable).

yarda, 2022-01-31:
Yes, it seems the debug output was disabled in Fedora intentionally not to spam regular users. It can be re-enabled by (this could be similar in other distros):
```
$ echo -e "[Rules]\n*.debug=true" > ~/.config/QtProject/qtlogging.ini
```
Regarding the original problem it could be related to the sub devices settings and channel mapping. I will experiment with it.

yarda, 2022-04-01:
Sorry, I didn't know the policy about closing inactive issues. I opened the issue to track the problem and let other users know that such problem exists. I am bit slow with the fix. So feel free to close this issue. I will provide PR once I have something usable. I think I can reference even closed issue in the PR.

f4exb, 2022-04-01:
If you have a fix that's great and if you need some more time to turn it into a PR please assign the issue to yourself. The no activity bot will skip issues that are assigned. I think you should be able to assign the issue to yourself if not please let us know here. In the meantime I remove the no issue label.

yarda, 2022-04-01:
I have project "Contributor" label, but it seems I cannot assign myself to the issue. If you can assign me to the issue, please do it.

yarda, 2022-04-27:
The problem doesn't seem to be in the antenna switching, but that the USRP sometimes doesn't stop transmitting. And with the power going to the TX/RX antenna, the rx is routed from the RX2 antenna and the firmware will not switch antennas until the the transmission cease. Which is correct function. Now I am trying to figure out why it sometimes doesn't stop transmitting.