f4exb/sdrangel #1121: Can't compile on M1 Macbook.

ReDetection, 2022-01-30:
I recently switched to Apple M1 MacBook and now I have troubles compiling sdrangel. 

First of all, clang does not support `-march=native` flag when compiling to Apple Silicon target, but this is rather simple: I commented out `ARCH_OPT "native"` line in the CMakeLists.txt. Done.

Next problem: even though I explicitly pass which QT to use via `env Qt5_DIR="/opt/homebrew/Cellar/qt@5/5.15.2_1/" cmake ./`, for some reason different version comes into play, according to the many errors in the log:

```
/opt/homebrew/include/QtCore/qglobal.h:816:31: error: no template named 'Promoted' in namespace 'QTypeTraits'
constexpr inline QTypeTraits::Promoted<T, U> qBound(const T &min, const T &val, const U &max)
```

So this `/opt/homebrew/include/QtCore/` is actually a simlink to `/opt/homebrew/lib/QtCore.framework/` which is definitely different from `/opt/homebrew/Cellar/qt\@5/5.15.2_1/include/QtCore/`.

Can it be, that there is additional Qt auto-discovery which ignores the Qt5_DIR environment variable?

Dev environment:
MacBook Air M1 2020, 16GB RAM
```
$ uname -smr
Darwin 21.2.0 arm64
```

Basically I followed my own guide on how to compile sdrangel from the times I had Intel-based machine https://www.patreon.com/posts/49274096 

Where I get confused: 
* `compile_commands.json` has no mention of `/opt/homebrew/include/QtCore/`, only correct paths from  `/opt/homebrew/Cellar/qt\@5/5.15.2_1/`
* `*.cpp.o.d` is the only files to mention the wrong Qt directory, so far only from sdrbase module (the one that fails compilation), but probably will have more later

How can I troubleshoot where this directory comes from?

ReDetection, 2022-01-30:
The fastest workaround I found so far: to temporary remove qt brew package (keep only qt@5) during sdrangel compilation. Which is kinda fine by me now, so if you don't have a quick idea, then it is probably no sense to invest a lot of efforts

srcejon, 2022-01-30:
On Windows I specify qt dir with -DCMAKE_PREFIX_PATH=path

ReDetection, 2022-01-30:
@srcejon thanks for the suggestion! Unfortunately, it had the same effect as in original message. I think the problem is somewhere outside of cmake – because `compile_commands.json` is nice and clean, no mention of the wrong qt path. It happens somewhere down the compile chain and that is kinda unexpected. 

Example of the command I consider clean:
```json
{
  "directory": "/Users/sbuglakov/src/github/sdrangel",
  "command": "/Applications/Xcode13.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -DAPPLICATION_NAME=\\\"SDRangel\\\" -DARCHITECTURE_ARM64 -DCOMPANY=\\\"f4exb\\\" -DCOPYRIGHT=\"\\\"Copyright (c) 2019 Edouard Griffiths. All rights reserved.\\\"\" -DIDENTIFIER=\\\"com.f4exb.SDRangel\\\" -DQT_CORE_LIB -DQT_GUI_LIB -DQT_MULTIMEDIA_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_NO_DEBUG_OUTPUT -DQT_WEBSOCKETS_LIB -DSDRANGEL_LIB_VERSION=\\\"6.18.0\\\" -DSDRANGEL_VERSION=\\\"6.18.0-32-g85184e2b4\\\" -DSDR_RX_SAMPLE_24BIT -DUSE_NEON -I/Users/sbuglakov/src/github/sdrangel -I/Users/sbuglakov/src/github/sdrangel/sdrangelsrv_autogen/include -I/Users/sbuglakov/src/github/sdrangel/sdrbase -I/Users/sbuglakov/src/github/sdrangel/exports -I/Users/sbuglakov/src/github/sdrangel/sdrgui -I/Users/sbuglakov/src/github/sdrangel/sdrsrv -I/Users/sbuglakov/src/github/sdrangel/sdrbench -I/Users/sbuglakov/src/github/sdrangel/logging -F/Applications/Xcode13.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.1.sdk/System/Library/Frameworks -iframework /opt/homebrew/Cellar/qt@5/5.15.2_1/lib -isystem /opt/homebrew/Cellar/qt@5/5.15.2_1/lib/QtMultimedia.framework/Headers -isystem /opt/homebrew/Cellar/qt@5/5.15.2_1/lib/QtNetwork.framework/Headers -isystem /opt/homebrew/Cellar/qt@5/5.15.2_1/lib/QtCore.framework/Headers -isystem /opt/homebrew/Cellar/qt@5/5.15.2_1/./mkspecs/macx-clang-arm64 -isystem /opt/homebrew/Cellar/qt@5/5.15.2_1/lib/QtGui.framework/Headers -isystem /Applications/Xcode13.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.1.sdk/System/Library/Frameworks/OpenGL.framework/Headers -isystem /opt/homebrew/Cellar/qt@5/5.15.2_1/lib/QtWebSockets.framework/Headers -O3 -DNDEBUG -arch arm64 -isysroot /Applications/Xcode13.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.1.sdk -Wall -Wextra -Wvla -Woverloaded-virtual -ffast-math -ftree-vectorize -fPIC -std=c++14 -o CMakeFiles/sdrangelsrv.dir/sdrangelsrv_autogen/mocs_compilation.cpp.o -c /Users/sbuglakov/src/github/sdrangel/sdrangelsrv_autogen/mocs_compilation.cpp",
  "file": "/Users/sbuglakov/src/github/sdrangel/sdrangelsrv_autogen/mocs_compilation.cpp"
},
```

closed 2022-04-07 by github-actions[bot]