f4exb/sdrangel #1122: CRC error when loading .wav file

sbeukers, 2022-01-30:
Hi all,
What am I doing wrong?
Or is it a bug
I use SDRAngel 6.18.1 WIN10

When I load .wav files with the FileInput i get a CRC error. Cannot Start Sample Source.

### SDR
FileInput 6.17.7
Also reinstalled everything. 
(I also use SDRangel with the HackRF no problem)

### Actual behavior
When I load .wav files with the FileInput i get a CRC error. Cannot Start Sample Source.
Files are freshly saved by audacity. 
I also tried from a different source same issue.
But. some files are working. Like the CRC is stored somewhere.
Other programs play the WAV files just fine
(they are PCM 16b mono)

### Expected behavior
-CRC updated on filechange
-Can CRC check be disabled or bypassed?

### Steps to reproduce
Load .WAV 16b mono in the FileInput filter. then change the file to other a bunch of times. Change/Save the wav while loaded in SDRAngel maybe?

### Log Files

2022-01-30 11:21:59.703 (D) FileInput::handleMessage: MsgStartStop:  stop
2022-01-30 11:21:59.703 (D) DSPDeviceSourceEngine::stopAcquistion
2022-01-30 11:21:59.703 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionStop
2022-01-30 11:21:59.703 (D) DSPDeviceSourceEngine::gotoIdle
2022-01-30 11:22:00.218 (D) FileInput::handleMessage: MsgStartStop:  start
2022-01-30 11:22:00.218 (D) DSPDeviceSourceEngine::initAcquisition
2022-01-30 11:22:00.218 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionInit
2022-01-30 11:22:00.218 (D) DSPDeviceSourceEngine::gotoIdle
2022-01-30 11:22:00.218 (D) DSPDeviceSourceEngine::gotoInit:   m_deviceDescription:    sampleRate:  48000  centerFrequency:  0
2022-01-30 11:22:00.218 (D) DSPDeviceSourceEngine::gotoInit: initializing  SpectrumVis
2022-01-30 11:22:00.218 (D) SpectrumVis::handleMessage: DSPSignalNotification:  centerFrequency:  0  sampleRate:  48000
2022-01-30 11:22:00.218 (D) DSPDeviceSourceEngine::startAcquisition
2022-01-30 11:22:00.218 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionStart
2022-01-30 11:22:00.218 (D) DSPDeviceSourceEngine::gotoRunning
2022-01-30 11:22:00.218 (D) DSPDeviceSourceEngine::gotoRunning:    started
2022-01-30 11:22:00.218 (W) FileInput::start: file not open. not starting
2022-01-30 11:22:00.218 (D) DSPDeviceSourceEngine::gotoError:  "Could not start sample source"
2022-01-30 11:22:00.218 (D) FileInputGUI::handleInputMessages: DSPSignalNotification: SampleRate:48000, CenterFrequency:0
2022-01-30 11:22:00.218 (D) GLSpectrumGUI::handleInputMessages: message: GLSpectrum::MsgReportSampleRate
2022-01-30 11:22:00.357 (D) DSPDeviceSourceEngine::errorMessage
2022-01-30 11:22:00.358 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPGetErrorMessage
2022-01-30 11:22:17.823 (D) FileInputGUI::configureFileName:  C:/Users/sjoerd/Downloads/Lock LEFT.wav
2022-01-30 11:22:17.824 (D) FileInput::openFileStream:  C:/Users/sjoerd/Downloads/Lock LEFT.wav  fileSize:  315798092  bytes  length:  1644781500  microseconds  sample rate:  48000  S/s  center frequency:  0  Hz  sample size:  16  bits
2022-01-30 11:22:17.824 (D) FileInputGUI::handleInputMessages: DSPSignalNotification: SampleRate:48000, CenterFrequency:0
2022-01-30 11:22:17.824 (D) GLSpectrumGUI::handleInputMessages: message: GLSpectrum::MsgReportSampleRate
2022-01-30 11:22:20.513 (D) FileInput::handleMessage: MsgStartStop:  stop
2022-01-30 11:22:20.513 (D) DSPDeviceSourceEngine::stopAcquistion
2022-01-30 11:22:20.513 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionStop
2022-01-30 11:22:20.513 (D) DSPDeviceSourceEngine::gotoIdle
2022-01-30 11:22:20.762 (D) FileInput::handleMessage: MsgStartStop:  start
2022-01-30 11:22:20.762 (D) DSPDeviceSourceEngine::initAcquisition
2022-01-30 11:22:20.762 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionInit
2022-01-30 11:22:20.762 (D) DSPDeviceSourceEngine::gotoIdle
2022-01-30 11:22:20.762 (D) DSPDeviceSourceEngine::gotoInit:   m_deviceDescription:    sampleRate:  48000  centerFrequency:  0
2022-01-30 11:22:20.762 (D) DSPDeviceSourceEngine::gotoInit: initializing  SpectrumVis
2022-01-30 11:22:20.762 (D) SpectrumVis::handleMessage: DSPSignalNotification:  centerFrequency:  0  sampleRate:  48000
2022-01-30 11:22:20.762 (D) DSPDeviceSourceEngine::startAcquisition
2022-01-30 11:22:20.762 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionStart
2022-01-30 11:22:20.762 (D) DSPDeviceSourceEngine::gotoRunning
2022-01-30 11:22:20.762 (D) DSPDeviceSourceEngine::gotoRunning:    started
2022-01-30 11:22:20.762 (D) FileInput::start
[... 47 lines cut ...]
2022-01-30 11:22:32.388 (D) FileInput::handleMessage: MsgStartStop:  stop
2022-01-30 11:22:32.388 (D) DSPDeviceSourceEngine::stopAcquistion
2022-01-30 11:22:32.388 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionStop
2022-01-30 11:22:32.388 (D) DSPDeviceSourceEngine::gotoIdle
2022-01-30 11:22:34.788 (D) MainWindow::closeEvent
2022-01-30 11:22:34.789 (D) MainWindow::savePresetSettings: preset [default | no name]
2022-01-30 11:22:34.789 (D) DeviceAPI::saveSamplingDeviceSettings: serializing source sdrangel.samplesource.fileinput[0]: 
2022-01-30 11:22:34.789 (D) MainWindow::saveFeatureSetPresetSettings: preset [default | no name]
2022-01-30 11:22:34.789 (D) AudioDeviceManager::serialize
2022-01-30 11:22:34.789 (D) AudioDeviceManager::debugAudioOutputInfos:  name:  "System default device"  sampleRate:  48000  udpAddress:  "127.0.0.1"  udpPort:  9998  copyToUDP:  false  udpUseRTP:  false  udpChannelMode:  0  udpChannelCodec:  0  decimationFactor:  1
2022-01-30 11:22:34.789 (D) DSPDeviceSourceEngine::stopAcquistion
2022-01-30 11:22:34.789 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionStop
2022-01-30 11:22:34.790 (D) DSPDeviceSourceEngine::gotoIdle
2022-01-30 11:22:34.790 (D) DSPDeviceSourceEngine::removeSink:  SpectrumVis
2022-01-30 11:22:34.790 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPRemoveBasebandSampleSink
2022-01-30 11:22:34.794 (D) FileInput::stop
2022-01-30 11:22:34.866 (D) DSPDeviceSourceEngine::stop
2022-01-30 11:22:34.866 (D) DSPDeviceSourceEngine::gotoIdle
2022-01-30 11:22:34.866 (D) DSPDeviceSourceEngine::stop
2022-01-30 11:22:34.866 (D) DSPDeviceSourceEngine::gotoIdle
2022-01-30 11:22:34.902 (D) HttpListener: closed
2022-01-30 11:22:34.902 (D) HttpConnectionHandlerPool (0x273e298fd10): destroyed
2022-01-30 11:22:34.902 (D) HttpListener: destroyed
2022-01-30 11:22:34.902 (I) WebAPIServer::stop: stopped web API server at http://:8091
2022-01-30 11:22:34.902 (D) MainWindow::~MainWindow: end

sbeukers, 2022-01-30:
I also tried adding Everybody full access to the SDRAngel folder

srcejon, 2022-01-30:
Mono files wont work. It only supports stereo for IQ samples.

sbeukers, 2022-02-01:
That did the trick! thanks

closed 2022-02-01 by sbeukers