f4exb/sdrangel #1124: APT demodulator plugin not installed macos

99wombats, 2022-02-02:
I have just installed sdrangel using macports, I was hoping to use the APT demodulator plugin but it doesn't seem to have been installed - it's not available in the available channels. Is it not compatible with a macos environment? When I run `port contents sdrangel` I get:

```
Port SDRangel contains:
  /Applications/MacPorts/SDRangel.app/Contents/Info.plist
  /Applications/MacPorts/SDRangel.app/Contents/MacOS/SDRangel
  /Applications/MacPorts/SDRangel.app/Contents/PkgInfo
  /Applications/MacPorts/SDRangel.app/Contents/Resources/SDRangel.icns
  /opt/local/bin/sdrangel
  /opt/local/bin/sdrangelbench
  /opt/local/bin/sdrangelsrv
  /opt/local/lib/sdrangel/libfcdhid.dylib
  /opt/local/lib/sdrangel/libfcdlib.dylib
  /opt/local/lib/sdrangel/libhttpserver.dylib
  /opt/local/lib/sdrangel/liblimesdrdevice.dylib
  /opt/local/lib/sdrangel/liblogging.dylib
  /opt/local/lib/sdrangel/libmetisdevice.dylib
  /opt/local/lib/sdrangel/libqrtplib.dylib
  /opt/local/lib/sdrangel/libsdrbase.dylib
  /opt/local/lib/sdrangel/libsdrbench.dylib
  /opt/local/lib/sdrangel/libsdrgui.dylib
  /opt/local/lib/sdrangel/libsdrsrv.dylib
  /opt/local/lib/sdrangel/libsoapysdrdevice.dylib
  /opt/local/lib/sdrangel/libswagger.dylib
  /opt/local/lib/sdrangel/plugins/libantennatools.dylib
  /opt/local/lib/sdrangel/plugins/libaprs.dylib
  /opt/local/lib/sdrangel/plugins/libbeamsteeringcwmod.dylib
  /opt/local/lib/sdrangel/plugins/libchanalyzer.dylib
  /opt/local/lib/sdrangel/plugins/libdemodadsb.dylib
  /opt/local/lib/sdrangel/plugins/libdemodais.dylib
  /opt/local/lib/sdrangel/plugins/libdemodam.dylib
  /opt/local/lib/sdrangel/plugins/libdemodatv.dylib
  /opt/local/lib/sdrangel/plugins/libdemodbfm.dylib
  /opt/local/lib/sdrangel/plugins/libdemodchirpchat.dylib
  /opt/local/lib/sdrangel/plugins/libdemoddatv.dylib
  /opt/local/lib/sdrangel/plugins/libdemoddsd.dylib
  /opt/local/lib/sdrangel/plugins/libdemodfreedv.dylib
  /opt/local/lib/sdrangel/plugins/libdemodnfm.dylib
  /opt/local/lib/sdrangel/plugins/libdemodpacket.dylib
  /opt/local/lib/sdrangel/plugins/libdemodpager.dylib
  /opt/local/lib/sdrangel/plugins/libdemodssb.dylib
  /opt/local/lib/sdrangel/plugins/libdemodvor.dylib
  /opt/local/lib/sdrangel/plugins/libdemodvorsc.dylib
  /opt/local/lib/sdrangel/plugins/libdemodwfm.dylib
  /opt/local/lib/sdrangel/plugins/libfeatureafc.dylib
  /opt/local/lib/sdrangel/plugins/libfeatureais.dylib
  /opt/local/lib/sdrangel/plugins/libfeaturedemodanalyzer.dylib
  /opt/local/lib/sdrangel/plugins/libfeaturegs232controller.dylib
  /opt/local/lib/sdrangel/plugins/libfeaturepertester.dylib
  /opt/local/lib/sdrangel/plugins/libfeaturerigctlserver.dylib
  /opt/local/lib/sdrangel/plugins/libfeaturesimpleptt.dylib
  /opt/local/lib/sdrangel/plugins/libfeaturestartracker.dylib
  /opt/local/lib/sdrangel/plugins/libfilesink.dylib
  /opt/local/lib/sdrangel/plugins/libfilesource.dylib
  /opt/local/lib/sdrangel/plugins/libfreqtracker.dylib
  /opt/local/lib/sdrangel/plugins/libinputairspy.dylib
  /opt/local/lib/sdrangel/plugins/libinputairspyhf.dylib
  /opt/local/lib/sdrangel/plugins/libinputaudio.dylib
  /opt/local/lib/sdrangel/plugins/libinputfcdpro.dylib
  /opt/local/lib/sdrangel/plugins/libinputfcdproplus.dylib
  /opt/local/lib/sdrangel/plugins/libinputfileinput.dylib
  /opt/local/lib/sdrangel/plugins/libinputkiwisdr.dylib
  /opt/local/lib/sdrangel/plugins/libinputlimesdr.dylib
  /opt/local/lib/sdrangel/plugins/libinputlocal.dylib
  /opt/local/lib/sdrangel/plugins/libinputremote.dylib
  /opt/local/lib/sdrangel/plugins/libinputrtlsdr.dylib
  /opt/local/lib/sdrangel/plugins/libinputsoapysdr.dylib
  /opt/local/lib/sdrangel/plugins/libinputtestsource.dylib
  /opt/local/lib/sdrangel/plugins/libinterferometer.dylib
  /opt/local/lib/sdrangel/plugins/libjogdialcontroller.dylib
[... 85 lines cut ...]
  /opt/local/lib/sdrangel/pluginssrv/libmodieee_802_15_4srv.dylib
  /opt/local/lib/sdrangel/pluginssrv/libmodnfmsrv.dylib
  /opt/local/lib/sdrangel/pluginssrv/libmodpacketsrv.dylib
  /opt/local/lib/sdrangel/pluginssrv/libmodssbsrv.dylib
  /opt/local/lib/sdrangel/pluginssrv/libmodwfmsrv.dylib
  /opt/local/lib/sdrangel/pluginssrv/libnoisefiguresrv.dylib
  /opt/local/lib/sdrangel/pluginssrv/liboutputaudiosrv.dylib
  /opt/local/lib/sdrangel/pluginssrv/liboutputfileoutputsrv.dylib
  /opt/local/lib/sdrangel/pluginssrv/liboutputlimesdrsrv.dylib
  /opt/local/lib/sdrangel/pluginssrv/liboutputlocalsrv.dylib
  /opt/local/lib/sdrangel/pluginssrv/liboutputremotesrv.dylib
  /opt/local/lib/sdrangel/pluginssrv/liboutputsoapysdrsrv.dylib
  /opt/local/lib/sdrangel/pluginssrv/liboutputtestmosyncsrv.dylib
  /opt/local/lib/sdrangel/pluginssrv/liboutputtestsinksrv.dylib
  /opt/local/lib/sdrangel/pluginssrv/libradioastronomysrv.dylib
  /opt/local/lib/sdrangel/pluginssrv/libradioclocksrv.dylib
  /opt/local/lib/sdrangel/pluginssrv/libremotesinksrv.dylib
  /opt/local/lib/sdrangel/pluginssrv/libudpsinksrv.dylib
  /opt/local/lib/sdrangel/pluginssrv/libudpsourcesrv.dylib
  /opt/local/lib/sdrangel/pluginssrv/libvorlocalizersrv.dylib
  /opt/local/share/sdrangel/CHANGELOG
  /opt/local/share/sdrangel/LICENSE
  /opt/local/share/sdrangel/Readme.md

```

99wombats, 2022-02-02:
Satellite tracker also not installed.

I did have this error during the build, could it be relevant? I just forced the activation..

`Activating dbus @1.12.20_0
Error: Failed to activate dbus: Image error: /Library/LaunchAgents/org.freedesktop.dbus-session.plist already exists and does not belong to a registered port.  Unable to activate port dbus. Use 'port -f activate dbus' to force the activation.
Error: See /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_devel_dbus/dbus/main.log for details.
Error: Follow https://guide.macports.org/#project.tickets if you believe there
is a bug.
Error: Processing of port sdrangel failed`

srcejon, 2022-02-02:
To use the APT demod, aptdec needs to be installed. Likewise, SGP4 is needed for the Satellite Tracker, as detailed here:

https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux

You might want to contact the macports maintainers to see if they can be built

99wombats, 2022-02-02:
Thank you, I have now contacted the macports maintainers as you suggested

closed 2022-02-03 by f4exb