f4exb/sdrangel #1125: SDRPlay RSPDx Broadcast FM Demodulator Sound Clipping

neilharvey94044, 2022-02-03:
Context
I've just started using SDRAngel, but the behavior I'm seeing with sound using the RSPDx is not present for the RTL-SDR dongle.  I figure since I can successfully use the RTL-SDR dongle I know enough to know something is wrong, perhaps with the build.
The RSPDx and API I am using are working fine with multiple other SDR software, including SDRUno.

Problem
Sound is "clipped" (e.g. extreme chipmunk sounding human voice) when using the Broadcast FM Demodulator.  I know I'm on the station and am getting some portion of the sound, but it's distorted, like it's being clipped.  This also occurs when trying to use the AM Demodulator for AM broadcast stations.  I notice that the bluish vertical bar which I assume shows the bandwidth being sampled appears too narrow, it doesn't cover the entire station signal.  Adjusting the RF BW setting does not widen this blue vertical band nor does it fix the sound problem.  The blue vertical band covers the full station signal when using an RTL-SDR dongle.

Environment
RSPDx, using API 3.07, running on Windows 10.
SDRAngel Version 6.18.1, Build Info: Qt 5.15.2 64 bits

I have tried adjusting every parameter to no avail.

f4exb, 2022-02-04:
This does not make sense to me. Can you please add some screenshots including the device and broadcast FM settings?

neilharvey94044, 2022-02-04:
[image]

f4exb, 2022-02-05:
You have obviously a bad reception. It looks like the peak is ~20dB above noise. Broadcast stations are fairly loud and you should get something better. There are also sidelobes larger than the 250 kHz bandwidth this is odd. 250 kHz is a generous bandwidth often 180 kHz already gives good reception (maybe not for good RDS decoding). Conversely this is what I get myseld with the RSP2 with a peak ~35dB above noise:

[image]

Note that I am using zero IF. If I turn the IF at 450 kHz then I experience the same disorders. It seems that it also breaks the sample rate since the pitch is much higher than normal.

There seems to be an issue with the way IF shift is handled to recover some good signal I have to slightly increase the sample rate from the 2 MS/s minimum boundary but then it seems the frequency conversion is not accounted for at all:

[image]

For now I suggest to stay at zero IF.

f4exb, 2022-02-05:
I knew SDRPlay v3 was a bit special... Indeed if you use a non-zero IF you have to respect special constraints described [here](https://github.com/f4exb/sdrangel/tree/master/plugins/samplesource/sdrplayv3#15-if-frequency)

f4exb, 2022-02-05:
Well it will not work for 450 kHz because the frequency input lower bound is 2 MS/s. For higher IFs it appears the pitch is right but the rest is weird...

neilharvey94044, 2022-02-05:
After seeing your note above I changed it to zero IF and it now works.  Must have been the one setting I didn't try.  Thanks!

f4exb, 2022-02-05:
OK... Non zero IF is unpredictable anyway

f4exb, 2022-02-06:
Reopening. Non zero IF is workable in some conditions and implementing transverter mode can help with the frequency shift.

f4exb, 2022-02-08:
Deployed in v6.19.0

closed 2022-02-08 by f4exb