f4exb/sdrangel #1126: Settings for MIMO Device stream and spectrum in the API

oorrwullie, 2022-02-04:
Settings for MIMO devices do not save with the system settings, so I'm writing a script that will add the device and configure it using the REST API. I'm trying to configure a MetisMISO type. I have managed to get all of the settings configured through the API. but I cannot change the Stream and Spectrum settings from Rx0 to Tx.

**Describe the solution you'd like**
The settings for Spectrum and Stream are available for a MetisMISO device in the API.

f4exb, 2022-02-06:
Originally these settings were thought to be volatile. In order to have them handled in the API they also have to be part of the settings and thus persistent. I don; t think the lock should be made persistent however.

oorrwullie, 2022-02-07:
That makes perfect sense. I have to say, thank you for making an API. It makes an already great application really powerful. 👏

f4exb, 2022-02-09:
A few other things:

> Settings for MIMO devices do not save with the system settings

That's for any device set other than R0 and is the goal of this ticket: https://github.com/f4exb/sdrangel/issues/1050

> so I'm writing a script that will add the device and configure it using the REST API

It exists already:
  - [dump.py](https://github.com/f4exb/sdrangel/tree/master/scriptsapi#dumppy) saves the whole configuration to JSON
  - [config.py](https://github.com/f4exb/sdrangel/tree/master/scriptsapi#configpy) restores a configuration from JSON

commit a99e4589 (2022-02-09): Metis MISO: save stream index and spectrum stream index in settings. Make it available via API. Fixes #1126
closed 2022-02-09 by f4exb