f4exb/sdrangel #1129: Sat Tracker does not always stop APT demod automatically at LOS

choupette-change, 2022-02-06:
Hello, are you sure it is working correctly? I see problems again. https://github.com/f4exb/sdrangel/issues/893
Thank you for your help

srcejon, 2022-02-06:
I made some additional changes for this yesterday. Were you using the build with the 3D map when you saw this, or an older version?

choupette-change, 2022-02-06:
[image]
I used this version (see screenshot), should I load the latest version?

srcejon, 2022-02-06:
Ok, thanks. A "better" fix should be in the next release of SDRangel. So let us know if you still have problems after 6.19 is released.

choupette-change, 2022-02-06:
Thanks for your quick return. Another improvement, if possible, add "My position" on the map produced by the satellite. In cloudy weather, it is difficult to identify geographically.

srcejon, 2022-02-06:
You get that with the new 3D Map - one of the Red dots in this image is "my position" and you can customise what is drawn for it and the transparency of the clouds etc.

[image]

srcejon, 2022-03-08:
Have you seen this problem with the most recent version?

choupette-change, 2022-03-08:
Now it's ok with new versions Thanks

closed 2022-03-08 by srcejon