f4exb/sdrangel #1134: FFTW regenerating on every start

Jcwscience, 2022-02-09:
I can give more info about my setup, but I was hoping it might just be a simple settings issue. (My setup is a mess)

srcejon, 2022-02-10:
Please give us more info about your setup, and the log file, where you should see some lines like this:

2022-02-10 11:23:46.130 (D) FFTEngine::create: using FFTW engine
2022-02-10 11:23:46.131 (D) FFTWEngine::configure: successfully imported from FFTW wisdom file: 'C:/Users/jon/AppData/Roaming/f4exb/SDRangel\fftw-wisdom'
2022-02-10 11:23:46.131 (D) FFT: creating FFTW plan (n=128,forward) took 1 ms
2022-02-10 11:23:46.131 (D) FFTEngine::create: using FFTW engine
2022-02-10 11:23:46.132 (D) FFTWEngine::configure: successfully imported from FFTW wisdom file: 'C:/Users/jon/AppData/Roaming/f4exb/SDRangel\fftw-wisdom'
2022-02-10 11:23:46.132 (D) FFT: creating FFTW plan (n=256,forward) took 0 ms
2022-02-10 11:23:46.132 (D) FFTEngine::create: using FFTW engine
2022-02-10 11:23:46.132 (D) FFTWEngine::configure: successfully imported from FFTW wisdom file: 'C:/Users/jon/AppData/Roaming/f4exb/SDRangel\fftw-wisdom'
2022-02-10 11:23:46.132 (D) FFT: creating FFTW plan (n=512,forward) took 0 ms
2022-02-10 11:23:46.132 (D) FFTEngine::create: using FFTW engine
2022-02-10 11:23:46.134 (D) FFTWEngine::configure: successfully imported from FFTW wisdom file: 'C:/Users/jon/AppData/Roaming/f4exb/SDRangel\fftw-wisdom'
2022-02-10 11:23:46.134 (D) FFT: creating FFTW plan (n=1024,forward) took 0 ms
2022-02-10 11:23:46.134 (D) MainWindow::MainWindow: load settings...

Jcwscience, 2022-02-12:
Here it is.

```
2022-02-12 12:02:45.098 (D) FFTEngine::create: using FFTW engine
2022-02-12 12:02:45.098 (D) FFTWEngine::configure: no FFTW wisdom file
2022-02-12 12:03:06.130 (D) FFT: creating FFTW plan (n=128,forward) took 21031 ms
```
It looks like it isn't saving the wisdom file after generation

srcejon, 2022-02-12:
Still need some more info. What OS are you on for a start? The more info you provide, the more chance there is someone can help.

Jcwscience, 2022-02-19:
@srcejon Sorry for the late response, I've had a lot going on. I'm running ubuntu 21.1 aarch64 on an Nvidia AGX Xavier. I compiled everything from source as per the instructions and everything seems to work except for the weird start issue. where are the wisdom files stored? Maybe it's a permissions issue?

Thanks!

srcejon, 2022-02-21:
You can create the Wisdom file using the menu Preferences > FFT

On linux the default location is /home/yourname/.local/share/f4exb/SDRangel/fftw-wisdom - That should be filled in automatically when you select the menu item above.

Jcwscience, 2022-02-24:
@srcejon That worked! Maybe y’all should put that as a step in the build instructions (unless it already is and I’m just blind). I thought it was generating the files at start, I didn’t realize it needed to be done separately. Thanks!

closed 2022-02-24 by Jcwscience

nox3000, 2023-10-02:
because the directory may not exist,
1. mkdir -p  /home/yourname/.local/share/f4exb/SDRangel/
2. /usr/bin/fftwf-wisdom -v -n -o /home/yourname/.local/share/f4exb/SDRangel/fftw-wisdom 128 256 512 1024