f4exb/sdrangel #1140: Spectrum marker cur/max doesn't appear at correct level when zoomed in

srcejon, 2022-02-10:
Hi,

I'm trying to use the Spectrum Markers with the Cur/Max setting. It's doesn't appear to work properly when zoomed in.

When zoomed out, the marker is at -9.5dB as expected:

[image]

When zooming in it drops to -72db:

[image]

f4exb, 2022-02-18:
Possibly the correspondence between screen position and FFT bin does not follow the zooming for the current/max being watched.

f4exb, 2022-03-06:
The problem is that the fft bin index remains based on the full spectrum while the spectrum used is only a part of it.

commit 74497a68 (2022-03-06): GLSpectrum: take zooming into account when computing FFT bin number. Fixes #1140
closed 2022-03-06 by f4exb