f4exb/sdrangel #1148: 3D map does not appears (unbuntu 20.04)

parlotto, 2022-02-12:
Thank you for this amazing software.

### The 3D map zone is white
* I use the ads-b module to display planes on the map. 2D map is fine but 3D map zone is white.
* I have tried my own API key from maptiler but the result is the same.
[image]

### Log Files
Here is my log file, if it could help
[sdrangel.log](https://github.com/f4exb/sdrangel/files/8053423/sdrangel.log)

srcejon, 2022-02-12:
This should have been fixed with PR #1135 but isn't in the latest linux release yet.

parlotto, 2022-02-12:
Thank you for the answer. I will wait for the next release.

srcejon, 2022-03-03:
This should be fixed. Can you let us know if you still have a problem? Thanks.

parlotto, 2022-03-03:
Yes !!! 3D view is awesome. Thanks for this great software.

closed 2022-03-03 by srcejon