f4exb/sdrangel #1149: (W) qUncompress: Input data is corrupted

Valstee, 2022-02-14:
### Subject or type of issue
I don't see any error when compiling and everything works on other machines with ubuntu but when I run sdrangel server on an RPI with ubuntu server I get this message :

2022-02-04 18:18:46.298 (D) MainServer::loadSettings
**2022-02-04 18:18:46.299 (W) qUncompress: Input data is corrupted
2022-02-04 18:18:46.299 (W) qUncompress: Input data is corrupted
2022-02-04 18:18:46.300 (W) qUncompress: Input data is corrupted
2022-02-04 18:18:46.300 (W) qUncompress: Input data is corrupted**
2022-02-04 18:18:46.300 (D) AudioDeviceManager::deserialize
2022-02-04 18:18:46.300 (W) qUncompress: Input data is corrupted
2022-02-04 18:18:46.300 (D) AMBEEngine::deserialize: invalid or no data
**2022-02-04 18:18:46.301 (W) qUncompress: Input data is corrupted
2022-02-04 18:18:46.301 (W) qUncompress: Input data is corrupted**
2022-02-04 18:18:46.301 (D) MainServer::MainServer: finishing...
2022-02-04 18:18:46.302 (D) StaticFileController: docroot=:/webapi, encoding=UTF-8, maxAge=60000
2022-02-04 18:18:46.302 (D) StaticFileController: cache timeout=60000, size=1000000
2022-02-04 18:18:46.303 (D) HttpListener: Listening on port 8091
2022-02-04 18:18:46.303 (I) WebAPIServer::start: starting web API server at http://172.17.0.2:8091
2022-02-04 18:18:46.303 (D) MainServer::MainServer: end

I tried with all the releases until 16.6 with Debian (Bullseye armf) and ubunutu (20.04.3 LTS arm64)  but I always get the same message.

I think it's from my installation but I don't understand what these messages in the logs refer to...

Thanks for your help !

f4exb, 2022-02-14:
It looks like these messages come from Qt and thus give no information of the component at the origin. However it seems to be related to some compressed file like the ones that are downloaded from the application e.g. the list of beacons. Normally they would be stored in $HOME/.config/f4exb/SDRangel and thus you could inspect possible zipped files there.

srcejon, 2022-02-15:
The beacons aren't compressed. Of the plugins I've written, I think it's just the StarTracker and 3D Map that use qUncompress directly. I wouldn't expect either of those to be used by the server.

mainsettings.cpp does call qUncompress though - is it possible your settings have been corrupted somehow?

Valstee, 2022-02-21:
It's strange because it's a new raspberry with a brand new distribution... I keep looking for solutions on forums

f4exb, 2022-02-21:
SD cards are not born equal it can be worth the little extra money to get industrial grade ones.

Valstee, 2022-02-21:
Okay, I didn't think it could come from there. Thanks a lot.
Indeed, I had to install / reinstall about 150 different ditributions on this SD for testing... and I had trouble installing ubuntu server the last time.
I will buy a new one this week and close the issue.

closed 2022-04-28 by github-actions[bot]