f4exb/sdrangel #1153: QSocketNotifier: Socket notifiers cannot be enabled or disabled from another thread

Valstee, 2022-02-17:
I just installed version 6.19.0-392-g8e79cd7ff with Ubuntu 20.04.3 LTS and qtVersion 5.12.8.
I have this problem on almost every AOS.
I use satellite tracker to trigger the antenna control:

2022-02-17 13:07:24.534 (D) GS232Controller::handleMessage: MsgStartStop: start: true
2022-02-17 13:07:24.534 (D) GS232Controller::start
2022-02-17 13:07:24.534 (D) GS232ControllerWorker::openSerialPort:  "ttyUSB1"
2022-02-17 13:07:24.534 (W) QSocketNotifier: Socket notifiers cannot be enabled or disabled from another thread
2022-02-17 13:07:24.535 (D) HttpConnectionHandler (0x564ca253c4e0): disconnected
2022-02-17 13:07:24.535 (W) QObject: Cannot create children for a parent that is in a different thread.
(Parent is QSerialPort(0x564ca19692b0), parent's thread is QThread(0x564ca1969440), current thread is QThread(0x564ca088aa10)
2022-02-17 13:07:24.535 (D) GS232ControllerWorker::applySettings:  m_azimuth:  282  m_elevation:  0  m_azimuthOffset:  0  m_elevationOffset:  2  m_azimuthMin:  0  m_azimuthMax:  450  m_elevationMin:  0  m_elevationMax:  180  m_tolerance:  1  m_protocol:  0  m_connection:  0  m_serialPort:  "ttyUSB1"  m_baudRate:  9600  m_host:  "127.0.0.1"  m_port:  4533  force:  true
2022-02-17 13:07:24.535 (D) GS232ControllerWorker::openSerialPort:  "ttyUSB1"
2022-02-17 13:07:24.535 (W) QSocketNotifier: Socket notifiers cannot be enabled or disabled from another thread
Erreur de segmentation (core dumped)

Valstee, 2022-02-21:
I tried to reinstall everything but I still have this error... I tried on another PC and I don't have this issue but it's frustrating not to be able to fix this problem on my PC... 

2022-02-21 10:39:57.095 (D) GS232Controller::handleMessage: MsgConfigureGS232Controller
2022-02-21 10:39:57.095 (D) GS232Controller::applySettings:  m_azimuth:  0  m_elevation:  0  m_azimuthOffset:  0  m_elevationOffset:  2  m_azimuthMin:  0  m_azimuthMax:  450  m_elevationMin:  0  m_elevationMax:  180  m_tolerance:  1  m_protocol:  0  m_serialPort:  "ttyUSB1"  m_baudRate:  9600  m_host:  "127.0.0.1"  m_port:  4533  m_track:  true  m_source:  "F0:1 SatelliteTracker"  m_title:  "GS-232 Rotator Controller"  m_rgbColor:  4292942179  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIFeatureSetIndex:  0  m_reverseAPIFeatureIndex:  0  force:  false
**2022-02-21 10:39:57.095 (W) QObject::killTimer: Timers cannot be stopped from another thread
2022-02-21 10:39:57.095 (W) QObject::~QObject: Timers cannot be stopped from another thread**
2022-02-21 10:40:10.655 (D) MainWindow::featureAddClicked: tab: 0 index: 9

Valstee, 2022-02-21:
I saw this on the beacon, i don't know if it's related to this issue :
2022-02-21 11:07:21.262 (W) QMetaObject::connectSlotsByName: No matching signal for on_presetTree_currentItemChanged(QTreeWidgetItem*,QTreeWidgetItem*)
2022-02-21 11:07:21.263 (W) QMetaObject::connectSlotsByName: No matching signal for on_presetTree_itemActivated(QTreeWidgetItem*,int)

srcejon, 2022-02-21:
Presumably this is Linux specific as seems fine on Windows.

Probably related to https://github.com/f4exb/sdrangel/pull/1059 implemented in 6.17.4-1. What was the last version you were using?

Valstee, 2022-02-21:
You are too good! by installing version 6.17.3 the problem is solved

f4exb, 2022-03-01:
Unlike in other plugins  I do not see the `started()` and `finished()` slots in `GS232ControllerWorker`. This could be an indication that #1059 related changes have not been implemented in the `GS232ControllerWorker`. Indeed the serial interface related stuff should be managed from the thread and it does not appear that this is the case here.

srcejon, 2022-03-03:
RE #1059 - It didn't seem to be necessary on Windows - but as it is on Linux, I'll make the change.

f4exb, 2022-03-05:
implemented in v6.20.0

closed 2022-03-05 by f4exb