f4exb/sdrangel #1158: DATV-Demodulator

DL9UFB, 2022-02-21:
DATV-Demodulator bandwith adjust, the 100kc scrolling not correct working (10kc insted 100kc)
# SDRangel Issue Template
Please let us know how you use the software, and what you enjoy, or don't enjoy the most.  
We are interested to find out how other people make use of this software.

#### From the Project Wiki:  
   *I ask you (as a User or Collaborator) to take the role of first level support.   
   It means you have to help me and not the opposite: you have to help me  
   understand the problem or what you think is a problem*

### Subject or type of issue
* Describe your issue here.
* Is this a runtime or compilation issue?
* If compilation, where in the instructions are you stuck?

### SDR
* We always want to first rule out that there is no issue with your hardware
* Which SDR are you using?
* Which version of Driver do you have loaded for the SDR?
* If applicable, Can you prove SDR functionality with a loopback test?
* Have you tried power cycling the SDR or the machine connecting to the SDR? Sometimes a reset is all you need.
 * Perhaps this is the issue, that the issue causes you to need to reset.

### Development environment
* Compilation instructions are given in terms of Ubuntu 21.04, have you tried to recreate the issue using this environment?
* CPU and RAM of your machine if applicable

### Actual behavior
* Tell us what happens.
* Be specific here, It is important that we be able to recreate the issue
* Please list any relevant information leading up to the issue

### Expected behavior
* Tell us what should happen
* Helps us ensure that you understand what you are trying to do so that we can rule out any user error.

### Steps to reproduce
* Tell us how to reproduce this problem. Include screenshots, where applicable
* If radio related, have you tried to reproduce with a different radio, if possible?
* Please list Center Frequency, bandwidth and any other applicable RF settings that are pertinent to the issue.

### Log Files
* Please read any pertinant log files!
* Attach any relevant information you find here.

DL9UFB, 2022-02-21:
DATV-Demodulator bandwith adjust, the 100kc scrolling not correct working (10kc insted 100kc)

f4exb, 2022-02-22:
The bandwidth control of the DATV demodulator is a frequency dial therefore there could not be a confusion between decades. If you had followed the guidelines instead of just copying the canvas may be we would know more about the context.

DL9UFB, 2022-02-22:
Hello Edouard, I think we  have us misunderstanding .
If the mouse pointer itself is at the 100kHz position
and I move the scroll wheel, the 100kHz position does not change,
but the 10kHz position does, and even if the mouse pointer is on the 10kHz position,
the 1kHz position changes and not the 10kHz position.
This was from version 6.19.0, before it was ok.

73, Wolfgang

PS.

f4exb, 2022-02-23:
OK that's clearer now thanks.

f4exb, 2022-02-23:
Since https://github.com/f4exb/sdrangel/pull/1142 the number of digits has increased to 8. There seems to be an issue in `ValueDial` when the number of digits is > 7. `ValueDialZ` does not seem to be affected with a similar issue.

srcejon, 2022-03-03:
The BW widget is a ValueDialZ with positiveOnly set. It seems there's a problem even when digits=7 (some work properly, some don't).

Problem appears to be in ValueDialZ::findExponent, in that wrong exponent is returned for a given digit. Looks like it was introduced when support for decimal points was added.

Changing:

        if ((i%4 == 0) || (m_positiveOnly && (i == d+s-1))) { // non digit positions

To

        if (i%4 == 0) { // non digit positions

Fixes it in DATVDemod - but I presume there's a case with m_decimalPos != 0 that will then break. Not sure where such a widget is though.

f4exb, 2022-03-03:
IT is in the SpectrumCalibrationPointsDialog here: https://github.com/f4exb/sdrangel/blob/master/sdrgui/gui/spectrumcalibrationpointsdialog.cpp#L51

srcejon, 2022-03-03:
The one you just linked to isn't positive only, and the change will only effect those where that is true.

f4exb, 2022-03-03:
in `if ((i%4 == 0) || (m_positiveOnly && (i == d+s-1)))` if `m_positiveOnly` the second member of the or condition is always false and thus is equivalent to `if ((i%4 == 0)` so in both cases the (fixed) condition is effectively the same

srcejon, 2022-03-03:
Yep, that's what I meant.

commit 22381c5d (2022-03-03): ValueDialZ: fix find exponent. Fixes #1158