f4exb/sdrangel #1159: Support for NI USRP network devices

w6bvb, 2022-02-23:
Hi,
I was able to build sdrangel, but there is no luck to make it work with NI USRP-2922 or similar. 
I try to use SoapyUHD, USRP. No luck.

srcejon, 2022-02-23:
What OS are you on? Did you build it with USRP/UHD driver (https://github.com/f4exb/sdrangel/blob/master/plugins/samplesource/usrpinput/readme.md)?

What do:

uhd_usrp_probe
uhd_find_devices

Output?

Can you run the UHD benchmark program?

./benchmark_rate --rx_rate 10e6 --tx_rate 10e6

w6bvb, 2022-02-23:
It is Ubuntu 21.10.
Yes, I build it with USRP/UHD driver. 
Now strange part: uhd_usrp_probe or uhd_find_devices can not see my NI-2922 but they detected RTL unit when it connected.
At same time I can run my NI-2922 using GNURadio without problem.
Will run ./benchmark_rate ... later and post output.
Note: I changed IP of NI USRP from default 192.168.10.2 to 192.168.1.131 and host PC used dynamic IP but it is not changing in my network. Ettus recommended to use default settings for USRP and for the host computer.

srcejon, 2022-02-23:
I think you need to find out why the uhd_* tools can't see it first.

w6bvb, 2022-02-23:
Will do, but why GNURadio can see it?

srcejon, 2022-02-23:
I have no idea - I don't use it. Do you tell it what the IP address is?

w6bvb, 2022-02-24:
uhd_find_devices:
*******@********:/opt/install/uhd/lib/uhd/utils$ uhd_find_devices --args="addr=192.168.1.131"
[INFO] [UHD] linux; GNU C++ version 10.3.0; Boost_107400; UHD_4.2.0.git-290-g2c7ce2db
--------------------------------------------------
-- UHD Device 0
--------------------------------------------------
Device Address:
    serial: 3075767
    addr: 192.168.1.131
    name: 
    type: usrp2

uhd_uspr_probe

*******@********:/opt/install/uhd/lib/uhd/utils$ uhd_usrp_probe --args="addr=192.168.1.131"
[INFO] [UHD] linux; GNU C++ version 10.3.0; Boost_107400; UHD_4.2.0.git-290-g2c7ce2db
[INFO] [USRP2] Opening a USRP2/N-Series device...
[INFO] [USRP2] Current recv frame size: 1472 bytes
[INFO] [USRP2] Current send frame size: 1472 bytes
[WARNING] [UDP] The send buffer could not be resized sufficiently.
Target sock buff size: 2500000 bytes.
Actual sock buff size: 1048576 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.wmem_max=2500000
[WARNING] [UDP] The send buffer could not be resized sufficiently.
Target sock buff size: 2500000 bytes.
Actual sock buff size: 1048576 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.wmem_max=2500000
[WARNING] [UDP] The send buffer could not be resized sufficiently.
Target sock buff size: 2500000 bytes.
Actual sock buff size: 1048576 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.wmem_max=2500000
[INFO] [USRP2] Detecting internal GPSDO.... 
[INFO] [GPS] No GPSDO found
[WARNING] [UHD] Unable to set the thread priority. Performance may be negatively affected.
Please see the general application notes in the manual for instructions.
EnvironmentError: OSError: error in pthread_setschedparam
  _____________________________________________________
 /
|       Device: USRP2 / N-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: N210r4
|   |   hardware: 2577
|   |   product: 30261
|   |   mac-addr: 00:80:2f:19:72:f1
|   |   ip-addr: 192.168.1.131
|   |   subnet: 255.255.255.0
|   |   gateway: 255.255.255.255
|   |   gpsdo: none
|   |   serial: 3075767
|   |   FW Version: 12.4
|   |   FPGA Version: 11.1
|   |   
|   |   Time sources:  none, external, _external_, mimo
|   |   Clock sources: internal, external, mimo
|   |   Sensors: mimo_locked, ref_locked
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 0
|   |   |   
|   |   |   Freq range: -50.000 to 50.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 1
|   |   |   
|   |   |   Freq range: -50.000 to 50.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX Dboard: A
|   |   |   ID: SBX (0x0054)
|   |   |   Serial: 30727BD
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: 0
|   |   |   |   Name: SBXv3 RX
|   |   |   |   Antennas: TX/RX, RX2, CAL
|   |   |   |   Sensors: lo_locked
|   |   |   |   Freq range: 400.000 to 4400.000 MHz
|   |   |   |   Gain range PGA0: 0.0 to 31.5 step 0.5 dB
|   |   |   |   Bandwidth range: 40000000.0 to 40000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Codec: A
|   |   |   |   Name: ads62p44
|   |   |   |   Gain range digital: 0.0 to 6.0 step 0.5 dB
|   |   |   |   Gain range fine: 0.0 to 0.5 step 0.1 dB
|   |     _____________________________________________________
|   |    /
|   |   |       TX DSP: 0
|   |   |   
|   |   |   Freq range: -200.000 to 200.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       TX Dboard: A
|   |   |   ID: SBX (0x0055)
|   |   |   Serial: 30727BD
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: 0
|   |   |   |   Name: SBXv3 TX
|   |   |   |   Antennas: TX/RX, CAL
|   |   |   |   Sensors: lo_locked
|   |   |   |   Freq range: 400.000 to 4400.000 MHz
|   |   |   |   Gain range PGA0: 0.0 to 31.5 step 0.5 dB
|   |   |   |   Bandwidth range: 40000000.0 to 40000000.0 step 0.0 Hz
|   |   |   |   Connection Type: QI
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Codec: A
|   |   |   |   Name: ad9777
|   |   |   |   Gain Elements: None

srcejon, 2022-02-24:
So it looks like you needed to specify the IP address.

SDRangel has the menu Preferences > Devices > User arguments for specifying things like this - but the USRP driver doesn't support it.

Probably needs to be added to DeviceUSRP::enumOriginDevices uhd::device::find() somehow.

srcejon, 2022-06-01:
@f4exb 

Looking at the pluto code, I see m_deviceAPI->getHardwareUserArguments() are used in PlutoSDRInput::openDevice() - rather than DevicePlutoSDRScan::enumOriginDevices, so I presume if a device needs an IP address to be specified, it should be handled as a non discoverable device. How should non-discoverable devices work?

I've created a non discoverable device in the Device user arguments dialog:

[image]

I can then select this as the sampling device:

[image]

However, when I select OK, I end up with a File Input device, and in the log, I have:

    2022-06-01 12:39:22.455 (D) MainWindow::sampleSourceCreate: deviceSetIndex: 1 deviceIndex: 6 hardwareId: "USRP" sequence: 0 id: "" serial: "USRP-0" displayedName: "USRP[0:0] user defined"
    2022-06-01 12:39:22.455 (D) MainWindow::sampleSourceCreate: non existent device replaced by File Input

It doesn't seem to create a USRPInput::USRPInput instance, so USRPInput::openDevice() is never called.

I don't have a pluto to debug, so can't follow how that works. Any suggestions?

f4exb, 2022-06-01:
Released in v7.3.1

closed 2022-06-01 by f4exb