f4exb/sdrangel #1160: Please unbundle the hidapi or allow compilation with the system hidapi

yarda, 2022-02-23:
### Subject or type of issue
The fcd/pp code uses some code from the qthid which bundles the hidapi. It seems it doesn't change the hidapi and it should work OK with the system hidapi (in Fedora we provide gr-funcube built with the system hidapi for a while and it works OK). Please unbundle the hidapi or allow compilation with the system hidapi.

We use the following patch in the Fedora downstream to unbundle the hidapi:
https://bugzilla.redhat.com/bugzilla/show_bug.cgi?id=2045924#c10

Unfortunately, I cannot test it on other arches, so it may need more work. Or maybe the cmake could check for the system hidapi and if not found it could fallback to the bundled one.

### SDR
Not applicable.

### Development environmentne if applicable
Fedora Rawhide (f37), x86_64

### Actual behavior
There is bundled hidapi and it is not possible to compile with the system hidapi.

### Expected behavior
There should be a way to compile with the system hidapi.

### Steps to reproduce
Check the code/cmake or build the projects and check for the linked libraries.

### Log Files
Not applicable.

f4exb, 2022-02-24:
> There should be a way to compile with the system hidapi.

I don't see the point in doing this sorry

yarda, 2022-02-24:
It could help with the distro downstream packaging, e.g.:
https://fedoraproject.org/wiki/Bundled_Libraries?rd=Packaging:Bundled_Libraries

Other distros have similar policies about bundling.

You can either bundle everything, bundle something or fully support system libs. As sdrangel supports building with the system libs for 99% of the libs I think system hidapi should be also covered (e.g. with the cmake switch).

I could try updating the patch to have such switch.

srcejon, 2022-02-24:
> I don't see the point in doing this sorry

From the link - it looks like resolving this is one of the "must" items in order to get SDRangel included with Fedora / RedHat distributions - which seems like a good thing.

@yarda - can you perhaps create a pull request here with the proposed patch? I don't have a FCD, but can test if it builds.

yarda, 2022-02-24:
I have both fcd/fcdpp and I can do the functional testing. 

Are you OK to unbundle or do you prefer a switch (e.g. --with-hidapi=PATH) and keep the bundled lib?

f4exb, 2022-02-24:
No in this case the hidapi code inside SDRangel should be simply removed. `libhidapi-dev` package in Ubuntu that can be added to the dependencies. 

If it breaks the FCD support well... I don't care.

hiperiondev, 2022-03-04:
@yarda Many Sdrangel dependencies are missing from Fedora. For my needs I keep a repo in COPR with all the missing packages. I started the process to add them to the official repos but the bureaucracy is a lot (I don't think it's necessarily a bad thing) and the acceptance times are too long for the times I have available to follow the project.
Maybe what I already have working can help you: https://copr.fedorainfracloud.org/coprs/lu3vea/sdrangel/

yarda, 2022-03-04:
@hiperiondev we are progressing step-by-step. The Fedora review is nearly finalized and it will bring sdrangel with the basic functionality from the distro repo. It's pretty usable. We can add the additional deps steb-by-step once they reached Fedora. I can add you as a comaintainer if you want. Also we could build plugins which are not allowed by Fedora policy out-of-the-tree and ship them through the rpmfusion to extend the functionality of the sdrangel from the Fedora distro repo.

f4exb, 2022-03-05:
I think hidapi has been moved inside libusb so there should not be a need for an extra dependency.

closed 2022-03-05 by f4exb

srcejon, 2022-09-20:
@yarda - I see sdrangel can now easily be installed in Fedora - good job!

What's the process to add in some of the additional libraries? E.g. sgp4, libdab, aptdec, libfaad

yarda, 2022-09-20:
Hi @srcejon, it depends on licensing and patents. Non patented libraries under good license can go directly to Fedora. Just somebody is needed to package it/review it. I can help with this. Then I will enable the support in sdrangel. Patented content could go to rpmfusion. It's good design of the sdrangel which probably allows delivering the support for patented content as plugins from the rpmfusion repo. Unfortunately, I haven't time to sort it out. I am mostly using sdrangel for SSB/FM communication, Es'hail, etc., and it's doing good job for me even in the current state.

yarda, 2022-09-21:
@srcejon maybe you could start with a list of libraries that are missing from Fedora and open bug in Fedora downstream bugzilla (e.g. against sdrangel component) and we could continue discussion there which will be more appropriate.

commit fb306288 (2022-03-05): Removed embedded hidapi. Fixes #1160
commit 3dc0c26f (2022-03-06): Repair FCD support with better FindHIDAPI cmake module. Fixes #1160