f4exb/sdrangel #1165: ADB-S after starting sdr doesn't work

Cherenkov11, 2022-02-25:
### Subject or type of issue
I noticed that if I add the ADS-B Demodulator after I start the acquisition of the SDR the plugins doesn't detect any signal, no dBs. And if you try to close the plugin or stop the acquisition sdrangel freezes and it crash. The only way to use ADS-B is adding it BEFORE you start the acquisition. If you do it like this you can use the plugin normally.

### SDR
- RTL-SDR V3

### Development environment
- sdrangel 6.19.1
- Arch Linux
- CPU AMD Ryzen 4800h

### Actual behavior
ADB-S Demodulator doesn't get any signal if you add it after you start the acquisition and if you try to close it or stop the acquisition the program crashes/freezes,

### Expected behavior
Adding the ADS-B plugins in any condition without the program crashing like any other plugin.

### Steps to reproduce (I tried it on two different computers, both with Arch Linux, same issue)
- Open sdrangel
- Start the acquisition of the SDR
- Add the ADS-B Demodulator (the dB meter doesn't move at all [I attached a screenshot])
- Try to remove the ADS-S Demodulator or stopping the acquisition
- sdrangel crashes/freezes 

[image]

(The screenshot shows that even the acquisition is started the dB meter doesn't detect any signal and if I try to close the plugins the program will crash)

f4exb, 2022-03-07:
This is only when you add the ADS-B demodulator while the acquisition is running.

Cherenkov11, 2022-03-07:
Exactly

f4exb, 2022-03-07:
if `void ADSBDemodSink::applyChannelSettings(int channelSampleRate, int channelFrequencyOffset, bool force)` is called with `channelSampleRate = 0` which is the case in this case then it creates a blocking situation and further processing of `ADSBDemodBaseband` input messages that would instead process the correct sample rate.

A solution is to bail out from `ADSBDemodSink::applyChannelSettings` if channel sample rate is zero.

commit 82ebab8a (2022-03-07): ADSB demod: do not apply channel sample rate if it is 0. Fixes #1165
closed 2022-03-07 by f4exb