f4exb/sdrangel #1167: Macports Arm new arm build fails to build

M4ttjabz, 2022-02-27:
New Macport for M1 Macs will not build.  Several failures today.  

Just thought I'd let you know.

Thanks!

srcejon, 2022-02-27:
Probably better off reporting on the macports site.

No M1 mac here - so unless you can post some errors logs - no way to help.

M4ttjabz, 2022-02-27:
here's the bottom of the log:

 :info:build [ 46%] Built target sdrgui
:info:build make[1]: Leaving directory `/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_SDRangel/SDRangel/work/build'
:info:build make: *** [all] Error 2
:info:build make: Leaving directory `/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_SDRangel/SDRangel/work/build'
:info:build Command failed:  cd "/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_SDRangel/SDRangel/work/build" && /usr/bin/make -j8 -w all VERBOSE=ON
:info:build Exit code: 2
:error:build Failed to build SDRangel: command execution failed
:debug:build Error code: CHILDSTATUS 19734 2
:debug:build Backtrace: command execution failed
:debug:build     while executing
:debug:build "system {*}$notty {*}$callback {*}$nice $fullcmdstring"
:debug:build     invoked from within
:debug:build "command_exec -callback portprogress::target_progress_callback build"
:debug:build     (procedure "portbuild::build_main" line 8)

and this was repeated about every other line:

:info:build clang: warning: argument unused during compilation: '-msse4.1' [-Wunused-command-line-argument]

^G Get Help

srcejon, 2022-02-27:
Thanks, but that's not quite enough of the log, as it doesn't show the error that causes make to fail.

M4ttjabz, 2022-02-27:
[log.txt](https://github.com/f4exb/sdrangel/files/8149485/log.txt)

srcejon, 2022-02-27:
I don't think you've attached the correct file.

nuclearace, 2022-05-04:
I'm getting these in my log

```
d In file included from /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_SDRangel/SDRangel/work/sdrangel-6.20.3/devices/usrp/deviceusrpparam.cpp:19:
:info:build In file included from /opt/local/libexec/qt5/lib/QtCore.framework/Headers/QDebug:1:
:info:build /opt/local/libexec/qt5/lib/QtCore.framework/Headers/qdebug.h:162:37: error: unknown type name 'QStringRef'; did you mean 'CFStringRef'?
:info:build     inline QDebug &operator<<(const QStringRef & t) { putString(t.constData(), uint(t.length())); return maybeSpace(); }
:info:build                                     ^
:info:build /opt/homebrew/include/QtCore/qstring.h:69:1: note: 'CFStringRef' declared here
:info:build Q_FORWARD_DECLARE_CF_TYPE(CFString);
:info:build ^
:info:build /opt/homebrew/include/QtCore/qglobal.h:830:77: note: expanded from macro 'Q_FORWARD_DECLARE_CF_TYPE'
:info:build #  define Q_FORWARD_DECLARE_CF_TYPE(type) typedef const struct __ ## type * type ## Ref
:info:build                                                                             ^
:info:build <scratch space>:244:1: note: expanded from here
:info:build CFStringRef
:info:build ^
:info:build In file included from /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_SDRangel/SDRangel/work/sdrangel-6.20.3/devices/usrp/deviceusrpparam.cpp:19:
:info:build In file included from /opt/local/libexec/qt5/lib/QtCore.framework/Headers/QDebug:1:
:info:build /opt/local/libexec/qt5/lib/QtCore.framework/Headers/qdebug.h:75:46: error: incomplete type 'QIODevice' named in nested name specifier
:info:build         Stream(QString *string) : ts(string, QIODevice::WriteOnly), ref(1), type(QtDebugMsg),
:info:build                                              ^~~~~~~~~~~
:info:build /opt/homebrew/include/QtCore/qtextstream.h:57:7: note: forward declaration of 'QIODevice'
:info:build class QIODevice;
:info:build       ^
:info:build In file included from /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_SDRangel/SDRangel/work/sdrangel-6.20.3/devices/usrp/deviceusrpparam.cpp:19:
:info:build In file included from /opt/local/libexec/qt5/lib/QtCore.framework/Headers/QDebug:1:
:info:build /opt/local/libexec/qt5/lib/QtCore.framework/Headers/qdebug.h:77:43: error: incomplete type 'QIODevice' named in nested name specifier
:info:build         Stream(QtMsgType t) : ts(&buffer, QIODevice::WriteOnly), ref(1), type(t),
:info:build                                           ^~~~~~~~~~~
:info:build /opt/homebrew/include/QtCore/qtextstream.h:57:7: note: forward declaration of 'QIODevice'
:info:build class QIODevice;
:info:build       ^
:info:build [ 39%] Building CXX object pluginssrv/channelrx/demodchirpchat/CMakeFiles/demodchirpchatsrv.dir/chirpchatdemodsettings.cpp.o
:info:build In file included from /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_SDRangel/SDRangel/work/sdrangel-6.20.3/devices/usrp/deviceusrpparam.cpp:19:
:info:build In file included from /opt/local/libexec/qt5/lib/QtCore.framework/Headers/QDebug:1:
:info:build /opt/local/libexec/qt5/lib/QtCore.framework/Headers/qdebug.h:162:66: error: member reference type 'const CFStringRef' (aka 'const __CFString *const') is a pointer; did you mean to use '->'?
:info:build     inline QDebug &operator<<(const QStringRef & t) { putString(t.constData(), uint(t.length())); return maybeSpace(); }
:info:build                                                                 ~^
:info:build /opt/local/libexec/qt5/lib/QtCore.framework/Headers/qdebug.h:162:66: error: incomplete definition of type '__CFString'
:info:build     inline QDebug &operator<<(const QStringRef & t) { putString(t.constData(), uint(t.length())); return maybeSpace(); }
:info:build                                                                 ~^
:info:build /opt/homebrew/include/QtCore/qstringview.h:65:1: note: forward declaration of '__CFString'
:info:build Q_FORWARD_DECLARE_CF_TYPE(CFString);
:info:build ^
:info:build /opt/homebrew/include/QtCore/qglobal.h:830:64: note: expanded from macro 'Q_FORWARD_DECLARE_CF_TYPE'
:info:build #  define Q_FORWARD_DECLARE_CF_TYPE(type) typedef const struct __ ## type * type ## Ref
:info:build                                                                ^
:info:build <scratch space>:213:1: note: expanded from here
:info:build __CFString
:info:build ^
:info:build In file included from /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_SDRangel/SDRangel/work/sdrangel-6.20.3/devices/usrp/deviceusrpparam.cpp:19:
:info:build In file included from /opt/local/libexec/qt5/lib/QtCore.framework/Headers/QDebug:1:
:info:build /opt/local/libexec/qt5/lib/QtCore.framework/Headers/qdebug.h:162:86: error: member reference type 'const CFStringRef' (aka 'const __CFString *const') is a pointer; did you mean to use '->'?
:info:build     inline QDebug &operator<<(const QStringRef & t) { putString(t.constData(), uint(t.length())); return maybeSpace(); }
:info:build                                                                                     ~^
:info:build /opt/local/libexec/qt5/lib/QtCore.framework/Headers/qdebug.h:162:86: error: incomplete definition of type '__CFString'
[... cut ...]  uint(t.length())); return maybeSpace(); }
:info:build                                                                                     ~^
:info:build /opt/homebrew/include/QtCore/qstringview.h:65:1: note: forward declaration of '__CFString'
:info:build Q_FORWARD_DECLARE_CF_TYPE(CFString);
:info:build ^
:info:build /opt/homebrew/include/QtCore/qglobal.h:830:64: note: expanded from macro 'Q_FORWARD_DECLARE_CF_TY
```
[main.log](https://github.com/f4exb/sdrangel/files/8624664/main.log)

@srcejon

nuclearace, 2022-05-04:
I think the issue is related to having homebrew installed components when trying to build using macports. They generally don't coexist well together... 

Trying to figure out how to force macports to ignore the homebrew headers/libs.

nuclearace, 2022-05-04:
`brew uninstall qt` solved it for me. Don't know if just swapping between installed versions is the best long term solution, but works for now.

closed 2022-07-10 by github-actions[bot]