f4exb/sdrangel #1169: Total GUI freeze when loading the Map (in Alpine Linux `edge`)

Nurrl, 2022-02-28:
Hello :)

### Context

I'm trying to provide a package to pkgs.alpinelinux.org for `sdrangel`, because I think this would be cool for this project to be available out of the box in Alpine.

### My specs (from `neofetch`):
```
OS: Alpine Linux v3.15 x86_64
Host: Latitude 5414
Kernel: 5.15.24-2-lts
CPU: Intel i5-6300U (4) @ 3.000GHz
GPU: Intel Skylake GT2 [HD Graphics 520]
Memory: 5257MiB / 7851MiB
```
Also, this issue happens with or without a SDR connected to the computer, for reference I own a HackRF.

### Actual behavior
Note: The project is build from sources using the MUSL (default in Alpine Linux).

When launching the program, and adding anything related to a Map (either **Channel -> ADS-B** or **Feature -> Map**), the program instantly freezes, nothing responds, and 1 CPU core has a 100% usage for an undefined time (I waited for 10mn and it didn't unfreeze).

After some investigations, using `gdb`, when adding the Map element, the code blocks in a recursive call to:
```
QSGBatchRenderer::Renderer::buildRenderLists(QSGNode*) () from /usr/lib/libQt5Quick.so.5
```

Also, below are the full logs when launching the project and adding the **Map Feature** until it freezes.

### Expected behavior
This shouldn't freeze the GUI.

### Steps to reproduce
Build, launch the project, and add the **Map feature** in the right panel.

### Log Files
```
2022-02-28 08:16:29.082 (I) SDRangel 6.19.1 Qt 5.15.3 64b x86_64 Alpine Linux v3.15 DSP Rx:24b Tx:16b PID 22488
2022-02-28 08:16:29.082 (I) MainSettings::MainSettings: settings file: format: 0 location: /home/unix/.config/f4exb/SDRangel.conf
2022-02-28 08:16:29.082 (D) MessagePipes::startGC
2022-02-28 08:16:29.082 (D) DataPipes::startGC
2022-02-28 08:16:29.082 (D) AudioDeviceManager::AudioDeviceManager: scan input devices
2022-02-28 08:16:29.140 (D) AudioDeviceManager::AudioDeviceManager: input device #0: pipewire
2022-02-28 08:16:29.140 (D) AudioDeviceManager::AudioDeviceManager: input device #1: default
2022-02-28 08:16:29.140 (D) AudioDeviceManager::AudioDeviceManager: input device #2: sysdefault:CARD=PCH
2022-02-28 08:16:29.140 (D) AudioDeviceManager::AudioDeviceManager: input device #3: front:CARD=PCH,DEV=0
2022-02-28 08:16:29.140 (D) AudioDeviceManager::AudioDeviceManager: scan output devices
2022-02-28 08:16:29.307 (D) AudioDeviceManager::AudioDeviceManager: output device #0: pipewire
2022-02-28 08:16:29.307 (D) AudioDeviceManager::AudioDeviceManager: output device #1: default
2022-02-28 08:16:29.307 (D) AudioDeviceManager::AudioDeviceManager: output device #2: sysdefault:CARD=PCH
2022-02-28 08:16:29.307 (D) AudioDeviceManager::AudioDeviceManager: output device #3: front:CARD=PCH,DEV=0
2022-02-28 08:16:29.307 (D) AudioDeviceManager::AudioDeviceManager: output device #4: surround21:CARD=PCH,DEV=0
2022-02-28 08:16:29.307 (D) AudioDeviceManager::AudioDeviceManager: output device #5: surround40:CARD=PCH,DEV=0
2022-02-28 08:16:29.307 (D) AudioDeviceManager::AudioDeviceManager: output device #6: surround41:CARD=PCH,DEV=0
2022-02-28 08:16:29.307 (D) AudioDeviceManager::AudioDeviceManager: output device #7: surround50:CARD=PCH,DEV=0
2022-02-28 08:16:29.307 (D) AudioDeviceManager::AudioDeviceManager: output device #8: surround51:CARD=PCH,DEV=0
2022-02-28 08:16:29.307 (D) AudioDeviceManager::AudioDeviceManager: output device #9: surround71:CARD=PCH,DEV=0
2022-02-28 08:16:29.307 (D) AudioDeviceManager::AudioDeviceManager: output device #10: hdmi:CARD=PCH,DEV=0
2022-02-28 08:16:29.307 (D) AudioDeviceManager::AudioDeviceManager: output device #11: hdmi:CARD=PCH,DEV=1
2022-02-28 08:16:29.307 (D) AudioDeviceManager::AudioDeviceManager: output device #12: hdmi:CARD=PCH,DEV=2
2022-02-28 08:16:29.307 (D) AudioDeviceManager::AudioDeviceManager: output device #13: hdmi:CARD=PCH,DEV=3
2022-02-28 08:16:29.307 (D) AudioDeviceManager::AudioDeviceManager: output device #14: hdmi:CARD=PCH,DEV=4
2022-02-28 08:16:29.307 (D) MainWindow::MainWindow: start
2022-02-28 08:16:29.387 (W) QMetaObject::connectSlotsByName: No matching signal for on_presetTree_currentItemChanged(QTreeWidgetItem*,QTreeWidgetItem*)
2022-02-28 08:16:29.387 (W) QMetaObject::connectSlotsByName: No matching signal for on_presetTree_itemActivated(QTreeWidgetItem*,int)
2022-02-28 08:16:29.394 (D) FFTEngine::create: using FFTW engine
2022-02-28 08:16:29.394 (D) FFTWEngine::configure: no FFTW wisdom file
2022-02-28 08:16:29.503 (D) FFT: creating FFTW plan (n=128,forward) took 108 ms
2022-02-28 08:16:29.503 (D) FFTEngine::create: using FFTW engine
2022-02-28 08:16:29.503 (D) FFTWEngine::configure: no FFTW wisdom file
[... 282 lines cut ...]

---------------------------------------------------------------------------------------------------------------------------------

2022-02-28 08:16:39.756 (D) MainWindow::featureAddClicked: tab: 0 index: 6
2022-02-28 08:16:39.756 (D) MainWindow::featureAddClicked: m_apiAdapter: 0x7f48db428080
2022-02-28 08:16:39.756 (D) Map::Map: webAPIAdapterInterface: 0x7f48db428080
2022-02-28 08:16:39.766 (D) WebServer on port  45389
2022-02-28 08:16:39.803 (D) Mapping  "3d"  to  "/home/unix/.local/share/f4exb/SDRangel/3d"
2022-02-28 08:16:40.115 (D) Beacon::readIARUCSV: Failed to open  "/home/unix/.local/share/f4exb/SDRangel/iaru_beacons.csv"
2022-02-28 08:16:40.115 (D) Beacon::readIARUCSV: Read  0  beacons
2022-02-28 08:16:40.250 (D) Map::handleMessage: MsgConfigureMap
2022-02-28 08:16:40.250 (D) Map::applySettings:  m_displayNames:  true  m_title:  "Map"  m_rgbColor:  4292942179  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIFeatureSetIndex:  0  m_reverseAPIFeatureIndex:  0  force:  true
2022-02-28 08:16:40.304 (D) HTTP Request:  "GET /street HTTP/1.1\r\n"
2022-02-28 08:16:40.304 (D) HTTP Request:  "GET /cycle HTTP/1.1\r\n"
2022-02-28 08:16:40.305 (D) HTTP Request:  "GET /night-transit HTTP/1.1\r\n"
2022-02-28 08:16:40.305 (D) HTTP Request:  "GET /satellite HTTP/1.1\r\n"
2022-02-28 08:16:40.305 (D) HTTP Request:  "GET /terrain HTTP/1.1\r\n"
2022-02-28 08:16:40.332 (D) HTTP Request:  "GET /transit HTTP/1.1\r\n"
2022-02-28 08:16:40.333 (D) HTTP Request:  "GET /hiking HTTP/1.1\r\n"
2022-02-28 08:16:40.397 (W) QGeoTileFetcherOsm: all providers resolved
```

Note: I added a dash separator for when I clicked the **Apply** button in the **Add feature** dialog.

Thanks !

Nurrl, 2022-02-28:
My main question is: Do you have any idea on what could be happening and how could I fix it ?

srcejon, 2022-02-28:
No idea, I'm afriad - QSGBatchRenderer::Renderer::buildRenderLists is within the Qt library.

The only unusual thing I can see, is you are using Qt 5.15.3, which I think is the commercial version of Qt - so probably hasn't been used previously. Might be worth trying Qt 5.15.2 if possible.

Nurrl, 2022-02-28:
I'll try, I don't know if I can easily downgrade a package in Alpine without breaking everything ^^'

Anyways, this is really weird, because it seems like it blocks the main event loop, because even the API on port `8091` does not respond after adding the Map component (while it does before).

I don't have a lot of experience with Qt, but couldn't it be due to a missing dependency of some sort ? Maybe a QML dependency ?

Nurrl, 2022-02-28:
Actually, the only time I got `ADS-B` working with my setup was when I forgot to install `qt5-qtquickcontrols2`, it showed an error in the logs, didn't render the map, so it didn't block the event loop ¯\_(ツ)_/¯

Nurrl, 2022-02-28:
After some investigation, I found this not-too-old SO post: https://stackoverflow.com/questions/52663856/qwidget-ui-freezes-when-using-a-qquickwidget, it says:
> Note: Using QQuickWidget disables the threaded render loop on all platforms. This means that some of the benefits of threaded rendering, for example Animator classes and vsync driven animations, will not be available.
> I changed my QQuickWidget to a QQuickView and everything is working.
What do you think ?

srcejon, 2022-03-02:
What specific changes did you make? Do you have a diff?

Nurrl, 2022-03-02:
I made no change whatsoever to the source code, I took the latest available version tag from the releases, I was just saying that removing the QML dependency `qt5-qtquickcontrols2` from my system made the map crash (with a QML error in the console), and thus making the ADS-B module available, but with a blank map.
Unfortunately this is not suitable for distribution use.

janczeresnia, 2022-03-30:
I have the same problem on Gentoo. I tested qt-5.15.2 and qt-5.15.3 it doesn't change anything. The only map that works is mapboxgl, but you can't run it in sdrangel without changing the source code. I wrote a patch for version 6.17.3 that changes the default map from osm to mapboxgl. But unfortunately my patches don't work with the new sdrangel version. The problem is the qt library.

srcejon, 2022-03-31:
Thanks, but we can't apply that patch, as mapboxgl isn't supported on Windows.

Nurrl, 2022-03-31:
Hello @janczeresnia, thank you for this patch, I'll try it locally with Alpine !

> Thanks, but we can't apply that patch, as mapboxgl isn't supported on Windows (but of course map be useful to others with this issue) — @srcejon 
Couldn't we branch in the `.qml` file to use `mapboxgl` in Linux and `osm` in Windows ? Alternatively a compilation flag would be fine with me !

janczeresnia, 2022-03-31:
> Couldn't we branch in the `.qml` file to use `mapboxgl` in Linux and `osm` in Windows ? Alternatively a compilation flag would be fine with me !

You can probably do it. I need to add a preprocessor flag to the source code like this:

```
#if !((defined _WIN32_) || (defined _MSC_VER))
 "Change for linux to mapboxgl"
#endif
```

Nurrl, 2022-03-31:
Thanks, I'll look into it :)
For now, the patch kinda works, but there is still some freezes, I'll try to investigate a bit more.

janczeresnia, 2022-03-31:
There is only a small problem that QML files do not support preprocessor macros. But it can be done.

janczeresnia, 2022-03-31:
I wrote a test application in qt-creator to check if the problem is really in the qt library. It looks like there is a bug in qt.

srcejon, 2022-03-31:
It works fine in 5.15.2 in Ubuntu for me.

I haven't tried 5.15.3, although I see it is now available as open-source, so maybe I can when builds are available.

janczeresnia, 2022-03-31:
I am not a QT specialist but in Gentoo the osm maps do not work with qt-5.15.2 and qt-5.15.3.
Only these two versions of QT are available in Gentoo. Maybe it's some other bug not related to QT I don't know.

Nurrl, 2022-03-31:
@srcejon: Would it be manageable for you if I provided an option (via a PR) in the `CMakeFile.txt` called `DISABLE_OSM_MAPPROVIDER` (or any name you'd prefer) that'll disable OpenStreetMap in `sdrangel` and replace it by MapBoxGL ?

for now I did this, and it seems to work:
```diff
diff --git a/plugins/channelrx/demodadsb/adsbdemodgui.cpp b/plugins/channelrx/demodadsb/adsbdemodgui.cpp
index 37e8084de..8a979cbe8 100644
--- a/plugins/channelrx/demodadsb/adsbdemodgui.cpp
+++ b/plugins/channelrx/demodadsb/adsbdemodgui.cpp
@@ -3590,7 +3590,7 @@ void ADSBDemodGUI::applyMapSettings()
     }
 
     // Create the map using the specified provider
-    QQmlProperty::write(item, "mapProvider", "osm");
+    QQmlProperty::write(item, "mapProvider", "mapboxgl");
     QVariantMap parameters;
     // Use our repo, so we can append API key and redefine transmit maps
     parameters["osm.mapping.providersrepository.address"] = QString("http://127.0.0.1:%1/").arg(m_osmPort);
diff --git a/plugins/feature/map/mapsettings.cpp b/plugins/feature/map/mapsettings.cpp
index 732e58502..24e250bcf 100644
--- a/plugins/feature/map/mapsettings.cpp
+++ b/plugins/feature/map/mapsettings.cpp
@@ -47,7 +47,6 @@ const QStringList MapSettings::m_pipeURIs = {
 
 // GUI combo box should match ordering in this list
 const QStringList MapSettings::m_mapProviders = {
-    QStringLiteral("osm"),
     QStringLiteral("esri"),
     QStringLiteral("mapbox"),
     QStringLiteral("mapboxgl"),
@@ -83,7 +82,7 @@ MapSettings::~MapSettings()
 void MapSettings::resetToDefaults()
 {
     m_displayNames = true;
-    m_mapProvider = "osm";
+    m_mapProvider = "mapboxgl";
     m_thunderforestAPIKey = "";
     m_maptilerAPIKey = "";
     m_mapBoxAPIKey = "";
@@ -166,7 +165,7 @@ bool MapSettings::deserialize(const QByteArray& data)
         QByteArray blob;
 
         d.readBool(1, &m_displayNames, true);
-        d.readString(2, &m_mapProvider, "osm");
+        d.readString(2, &m_mapProvider, "mapboxgl");
         d.readString(3, &m_mapBoxAPIKey, "");
         d.readString(4, &m_mapBoxStyles, "");
         d.readString(8, &m_title, "Map");
diff --git a/plugins/feature/map/mapsettingsdialog.ui b/plugins/feature/map/mapsettingsdialog.ui
index 30cc95f37..963a5e289 100644
--- a/plugins/feature/map/mapsettingsdialog.ui
+++ b/plugins/feature/map/mapsettingsdialog.ui
@@ -135,11 +135,6 @@
                <property name="toolTip">
                 <string>Select map provider</string>
                </property>
-               <item>
-                <property name="text">
-                 <string>OpenStreetMap</string>
-                </property>
-               </item>
```

janczeresnia, 2022-03-31:
> @srcejon: Would it be manageable for you if I provided an option (via a PR) in the `CMakeFile.txt` called `DISABLE_OSM_MAPPROVIDER` (or any name you'd prefer) that'll disable OpenStreetMap in `sdrangel` and replace it by MapBoxGL ?
> That would save me from maintaining a patch file throughout versions.

It would be very nice to add this functionality to the next version of sdrangel :)

srcejon, 2022-05-18:
Created a Qt bug for this: https://bugreports.qt.io/browse/QTBUG-103592 (as even the Qt Map Viewer example program doesn't work on Qt 5.15.3)

srcejon, 2022-06-15:
Workaround implemented in 7.3.2.

closed 2022-06-15 by srcejon