f4exb/sdrangel #1173: Airspy R2

diodelass, 2022-03-04:
What is the current status of support for the Airspy R2? 

I just got one, and was hoping to use it with SDRangel, but I've been having a lot of trouble getting it working. Notably, the version of the Airspy library that the build instructions specify is not compatible with it, and causes SDRangel to segfault shortly after startup if the device is plugged in at the time. Moving to a newer version of the library partially works; the input controls all appear functional and the spectrum plot looks fine, but I can't really demodulate anything consistently because it constantly drops samples (logging frequent messages to the console similar to `SampleSinkFifo::write: overflow - dropping 65536 samples`). It's almost usable in the low (2.5 MHz) sample rate mode, but seems to drop more samples than it catches in the high (10 MHz) mode.

I have also tried using the SoapySDR backend, which gives very slightly fewer dropped samples, but brings in many other issues (the amplifier controls make no sense, and the sample rate keeps changing itself back to 10 MHz when I try to change it to 2.5 MHz). This appears to also affect the RTL-SDR when the SoapySDR backend is selected, however, so I don't think this is specific to the Airspy. Other SDR programs using SoapySDR (GQRX and CubicSDR) do not seem to have any of these problems, though, so I'm not really sure where they're coming from.

My current best guess in general is just that one or more buffers in the chain are too short, and are underflowing because they're not being filled often enough, but I'm not sure how to test this because I don't know how to change the buffer size.

It seems like the Airspy R2 is very close to working well in SDRangel (via newer versions of libairspy), but it seems to need some kind of tweaking before it can really work. I'd be happy to provide more information as needed, but I'm not quite sure where to find it at the moment.

f4exb, 2022-03-04:
libairspy v1.0.10 as recommended is the latest release: https://github.com/airspy/airspyone_host/releases

diodelass, 2022-03-04:
Okay, I guess I'm not sure exactly what's going on here, then. What I meant to say is that when I do issue the `git reset --hard v1.0.10` command before the libairspy build process according to the instructions, then I get the consistent segfault when I start SDRangel if the device is connected, and when I skip that step and proceed directly to building, it doesn't segfault. I admit I'm still pretty new to git and I'm not quite sure I understand what that step is doing, but I had been assuming it was pinning it to one specific version.

closed 2022-05-09 by github-actions[bot]