f4exb/sdrangel #1177: release 6.20.1 freeze during loading process

IU4FIR, 2022-03-10:
### Subject or type of issue
* the program during the loading phase stops at the point "LOAD PLUGINS" and remains freezed...
* sdrangel-6.20.0-win64.exe and 6.20.1-win64.exe have the same issue.
* Install and/or run like USER or ADMIN have the same issue

### SDR
* Nooelec NESDR SMArTee XTR SDR - Premium RTL-SDR w/ Extended Tuning Range, Aluminum Enclosure, Bias Tee, 0.5PPM TCXO, SMA Input

### HW
* CPU 11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz
* RAM 16,0 GB
* SW Windows 11 Pro (V 22H2)

[image]

srcejon, 2022-03-10:
Works OK for me.  Does 6.19 or earlier work for you? Does it help if you unplug any SDRs?

Did you happen to enable logging previously? If you can run an older version - enable the log file - then try the new version and send the log file it creates before hanging.

IU4FIR, 2022-03-11:
Even the older version does not work, I suspect that the issue it's generate from "Cumulative Update for Windows 11, version 22H2 (10.0.22567.200) (KB5012432)" installed two days ago. 
The same behaviour using AIRSPY HF+.
Any suggestion or idea ???

f4exb, 2022-03-12:
How do you connect the devices to the USB port on the machine? Do you use an extension cable? Cables are not born equal and  I have encountered issues sometimes particularly when the cable is long (> 3m).

IU4FIR, 2022-03-14:
after a WIN11 update, now the sw remain idle (same point LOAD PLUGIN) for 3/4 minutes and start!

srcejon, 2022-03-15:
If it is now starting, enable the log file (Preferences > logging menu) and then we can see why it is taking so long to start.

Another user recently had a long startup time due to SoapySDR scanning lots of audio devices.

IU4FIR, 2022-03-15:
now after several startup the idle time decrease...

[IU4FIR.txt](https://github.com/f4exb/sdrangel/files/8253070/IU4FIR.txt)

srcejon, 2022-03-16:
15 seconds seems about normal.

IU4FIR, 2022-03-19:
After another win11 update, seems ok!

closed 2022-03-19 by IU4FIR

PD5DJ, 2022-10-29:
I suddenly have the same with the latest download of SDRAngle.. worked for days.. and suddenly freezes upon start.. with load plugins and thats it.. I have to restart the PC, I can not kill the process in W10

srcejon, 2022-10-29:
Does other SDR s/w run OK?